mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #2: Installation instructions are incomplete

johschmitz, 2021-12-21:
After installation of the file what is the user supposed to do to get the dashboard? Could you please add that information to the readme?

mkaiser, 2021-12-21:
The Dashboard shown in the screenshot in the readme.md was manually added. 

1. Create a new dashboard (or use the standard one)
2. Click "Add new Card" and  select "Entities" 
3. Add the sensor values you want to display. I (more or less randomly) divided them into momentary power, energy, battery and other. 

Another approach is to copy my dashboard using the "raw configuration editor" (top right, the 3 dots)

insert 

> views:
>   - title: Home
>     cards:
>       - type: entities
>         entities:
>           - entity: sensor.mppt1_voltage
>           - entity: sensor.mppt1_current
>           - entity: sensor.mppt2_voltage
>           - entity: sensor.mppt2_current
>           - entity: sensor.total_dc_power
>         title: PV Gen
>       - type: entities
>         entities:
>           - entity: binary_sensor.pv_generating
>           - entity: binary_sensor.battery_charging
>           - entity: binary_sensor.battery_discharging
>           - entity: binary_sensor.importing_power
>           - entity: binary_sensor.exporting_power
>           - entity: sensor.total_dc_power
>           - entity: sensor.export_power
>           - entity: sensor.load_power
>           - entity: sensor.battery_power
>           - entity: sensor.total_active_power
>           - entity: sensor.reactive_power
>         title: Power
>       - type: entities
>         entities:
>           - entity: sensor.daily_pv_generation_battery_discharge
>           - entity: sensor.total_pv_generation_battery_discharge
>           - entity: sensor.daily_pv_generation
>           - entity: sensor.total_pv_generation
>           - entity: sensor.daily_exported_energy_from_pv
>           - entity: sensor.total_exported_energy_from_pv
>           - entity: sensor.daily_battery_charge_from_pv
>           - entity: sensor.total_battery_charge_from_pv
>           - entity: sensor.daily_direct_energy_consumption
>           - entity: sensor.total_direct_energy_consumption
>           - entity: sensor.daily_battery_discharge
>           - entity: sensor.total_battery_discharge
>           - entity: sensor.daily_imported_energy
>           - entity: sensor.total_imported_energy
>           - entity: sensor.daily_battery_charge
>           - entity: sensor.total_battery_charge
>           - entity: sensor.daily_exported_energy
>           - entity: sensor.total_exported_energy
>         title: Energy
>       - type: entities
>         entities:
>           - entity: sensor.daily_battery_charge_from_pv
>           - entity: sensor.total_battery_charge_from_pv
>           - entity: sensor.battery_voltage
>           - entity: sensor.battery_current
>           - entity: sensor.battery_power
>           - entity: sensor.battery_level
>           - entity: sensor.battery_state_of_health
>           - entity: sensor.battery_temperature
>           - entity: sensor.daily_battery_discharge
>           - entity: sensor.total_battery_discharge
>           - entity: sensor.daily_battery_charge
>           - entity: sensor.total_battery_charge
>         title: Battery
>       - type: entities
>         entities:
>           - entity: sensor.grid_state
>           - entity: sensor.running_state
>           - entity: sensor.system_state
>         title: Inverter states
>       - type: gauge
>         entity: sensor.battery_level
>         min: 0
>         max: 100
>         severity:
>           green: 70
>           yellow: 40
>           red: 20
> title: Energy Dashboard

johschmitz, 2021-12-21:
Cool, thanks for the answer. Could you please add this file to the repository and the instructions to the readme for other people to see in the future?

johschmitz, 2021-12-23:
When trying to copy your dashboard code I am getting:

`YAMLException: duplicated mapping key (7:7) 4 | type: entities 5 | entities: 6 | entity: sensor.mppt1_voltage 7 | entity: sensor.mppt1_current -----------^ 8 | entity: sensor.mppt2_voltage 9 | entity: sensor.mppt2_current`

Which version of HA are you using? When you copy the code from GitHub and use the "raw configuration editor" to paste it does it work for you?

mkaiser, 2022-01-06:
hi @johschmitz,

sorry for my late reply. 

It was my fault. When i copy my github snippet from above the identation gets lost. 

Please try this file. Is it working? Wenn it works, I'll add it to the git-code

[Sungrow_Dashboard.txt](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/files/7821138/Sungrow_Dashboard.txt)

johschmitz, 2022-01-06:
I recommend to call it sungrow_dashboard.yaml not .txt. 
I can try it later but I think if it works for you, you should already go ahead and add it.

johschmitz, 2022-01-08:
Adding the content of your txt file to my dashboard yaml (starting with - title) worked for me, so you could add it as a .yaml file to the repository.