mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #3: How to map the sensors to energy dashboard

johschmitz, 2021-12-21:
Is there a way to map the sensors to the energy dashboard like with this add-on:
https://github.com/TenySmart/ModbusTCP2MQTT
Then it could look like this:
https://github.com/TenySmart/ModbusTCP2MQTT/blob/main/images/diagram.gif

I find your way of implementing it is easier but I just couldn't get it to work with the nice dashboard..

mkaiser, 2022-01-07:
totally missed this Issue... :/

do you mean the HA integrated Energy Dashboard?

[image]

for this you need to set the highlighted values in "Configuration" --> "Energy"
[image]

johschmitz, 2022-01-07:
Okay but how to display temhe current PV generation in W instead of the total?

mkaiser, 2022-01-07:
It seems that in Home Assistant the "Energy" dashboard is only meant to be showing the Energy (including the energy flow) in kWh.

For me, I summarized the momentary / current electrical power (in W) in an extra dashboard, to know if I should start heating / washing / whatever :)

johschmitz, 2022-01-07:
Then could you explain how to do that?

mkaiser, 2022-01-07:
Use the dashboard config file I provided in #2 

When this works, I will add that to the README

johschmitz, 2022-01-07:
So which sensor in there is the current PV production in W? I couldn't find a good documentation.

mkaiser, 2022-01-07:
look for "total_dc_power"
[image]

johschmitz, 2022-01-09:
Could you add the screenshot of the energy configuration with the different sensors also to the readme? For us everything works now. I hope with the extra information about the dashboard setup and the energy setup it will also become easier for other people

mkaiser, 2022-01-11:
Updated the readme. Thanks!

closed 2022-01-11 by mkaiser