mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #4: Not working with SH6.0RS

MatteoSalviTech, 2022-01-05:
I'm trying to make your integration working with my Sungrow SH6.0RS Hybrid Inverter but all the sensors return a "Not available" warning and I can't get the read values. I checked all the parameters (IP, port, slave) which are correct. How can I solve the problem?

mkaiser, 2022-01-06:
hi, 

if you are using Windows you can easily check, if the modbus connection generally works by [QModMaster ](https://sourceforge.net/projects/qmodmaster/)

I tested it with the values shown in the screenshot. First set the TCP Settings (Options --> Modbus TCP) then adapt all highlighted parameters
[image]

MatteoSalviTech, 2022-01-06:
I get the “Illegal Data Read” error but I can access the WiNet dongle on web browser…

mkaiser, 2022-01-07:
Can you give more information? 

The Webinterface does not have anything to do with the modbus connection.  How is your inverter exactly connected?

MatteoSalviTech, 2022-01-07:
The inverter is connected to WiFi (of course with static IP assigned) via WiNet dongle. I checked on the web interface (logging-in as admin) and the modbus server is correctly enabled but I can't access it via QModMaster or the HA integration...

MatteoSalviTech, 2022-01-07:
I also changed the Unit ID in QModMaster to "1" since this is the slave address of the inverter (also putting "100" it doesn't work properly).

mkaiser, 2022-01-07:
hmm I cannot find detailled information about the WiNet dongle. 

On my SH10 RT inverter I have LAN directly attached and use this for modbus. 

In addition I had a Wifi Dongle attached to a separate port (I don't know the exact model number). 

In the inverter settings I set the modbus slave adress to 100. standard is 1. 

Do you have a LAN cord attached or just Wifi?  Maybe it works only with LAN...

MatteoSalviTech, 2022-01-07:
So you have directly connected the inverter via RJ45 port without any additional device (such as the WiNet dongle)?

mkaiser, 2022-01-07:
my SH10 model has two different ports: One to directly attach Ethernet, one for a Wifi dongle

I use the Ethernet cable for Modbus and disconnected the Wifi (there is no need to send my information anywhere else than my home).

Can you check if it works with a Ethernet cable attached?

johschmitz, 2022-01-07:
I also saw this multiple times in other discussions that the Modbus connection only works through the LAN port. Additionally I saw some mentions of encryption. So wondering which Inverter firmware version you guys are running?

johschmitz, 2022-01-07:
I wonder if removing the Wi-Fi dongle while using the LAN port makes a difference. I was also having trouble so maybe I should try that combination as well.

MatteoSalviTech, 2022-01-07:
My inverter runs SUNSTONE-H_01011.02.19 firmware. I'll try without WiNet dongle and let you know.

mkaiser, 2022-01-07:
SH10.RT
MSDP: SAPPHIRE-H_03011.01.14 
LCD: SAPPHIRE-H_01011.01.17

MatteoSalviTech, 2022-01-07:
> I wonder if removing the Wi-Fi dongle while using the LAN port makes a difference. I was also having trouble so maybe I should try that combination as well.

I'm not completely sure I can connect the inverter directly to Ethernet without WiNet dongle...

johschmitz, 2022-01-08:
Okay we removed the Wi-Fi dongle and the isolarcloud connection seems to be still working. So I guess the dongle is only needed for firmware updates. Ours is not WiNet but "WiFi V313".

MatteoSalviTech, 2022-01-08:
Have you connected the Ethernet cable directly into the dongle port?

johschmitz, 2022-01-08:
I think what I just wrote is wrong, there seems to be a large delay until the isolarcloud notices a problem. Now the device seems to be offline, which I would have expected in the first place. But maybe the home assistant modbus connection will work properly now.
The cable is connected to the "LAN" Ethernet port not the one for the WiFi dongle. I never tried connecting the cable there.

MatteoSalviTech, 2022-01-08:
I think there is no Ethernet port in SH6.0RS model. On the manual it says that the Ethernet cable should be connected to the WiNet dongle port. The other RJ45 ports are intended for BUS and COM communication…

johschmitz, 2022-01-08:
Okay it seems the isolarcloud Android app just had an issue. Working again after restarting the app. So it does indeed work without the Wi-Fi dongle on the SH10RT and just Ethernet plugged to the "LAN" port.

johschmitz, 2022-01-08:
Btw: It shows the LAN network module which I guess is build into the SH10 as an "iE-Net".

johschmitz, 2022-01-08:
Okay so if your inverter has only one Ethernet port then I guess you have to go with that one 😊

johschmitz, 2022-01-08:
I got it to work now on our SH10, I found that I had the wrong slave ID set (100 instead of 1), fixed that and now it is working. Created a pull request with my simplified configuration and setup instructions.

MatteoSalviTech, 2022-01-08:
> I got it to work now on our SH10, I found that I had the wrong slave ID set (100 instead of 1), fixed that and now it is working. Created a pull request with my simplified configuration and setup instructions.

Did you get it working with or without the WiFi dongle?

johschmitz, 2022-01-08:
Without WiFi dongle. Late we can try if it also works when we replug the Wifi dongle or if that disturbs the home assistant modbus. But you should definitely try WITHOUT the Wifi dongle. Also see the home assistant log and check for any error messages regarding the modbus connection.
You can find my simplified yaml here: https://github.com/johschmitz/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml
Then you don't need the secrets.yaml anymore.

MatteoSalviTech, 2022-01-08:
The fact is that on thge SH6.0RS model there aren't any other LAN/Ethernet ports so I must connect the Ethernet cable to the WiNet dongle LAN port...

MatteoSalviTech, 2022-01-08:
Nothing to do: connecting the inverter via the WiNet dongle Ethernet port still doesn't give me modbus access.

johschmitz, 2022-01-08:
Why can't you unplug the dongle and plug the Ethernet cable directly into your inverter?

MatteoSalviTech, 2022-01-08:
Tried it: the port is designed to work with the dongle. If I unplug it, I can’t receive any data (the Ethernet switch can’t see the inverter at all).

johschmitz, 2022-01-08:
There's also different types of dongles from Sungrow. Maybe you have to get another one with Ethernet only to be able to use the Modbus connection?

Also try to disable the Wi-Fi connection with your home network in the web interface of the dongle:

[image]

MatteoSalviTech, 2022-01-08:
The strange thing is that modbus server is enabled:
[image]

mkaiser, 2022-01-11:
Saw this in a german forum: https://forum.iobroker.net/topic/38441/sungrow-wr-sgh10rt-erfolgreich-mit-modbus-eingebunden/115?_=1641918692873&lang=de 

It looks like it is a know problem and Sungrow is currently working on enabling the modbus for the WiNet-S adaptors for the S**G**x-series (S**H**x is working)

MatteoSalviTech, 2022-01-11:
Maybe I need to update the firmware. I'll check and let you know!

MatteoSalviTech, 2022-01-18:
Hello. After several tests I saw that with @mkaiser QModMaster configuration I can read Holding Registers data but not the input one (I always get "Illegal data" error). Any suggestions?

MatteoSalviTech, 2022-01-18:
Basically, this is the data I read on address 5002 by reading the input registers (with error):
[image]

And this is the data I can read on address 5002 by reading the holding registers (no error):
[image]

Can't figure out what's wrong...

mkaiser, 2022-02-28:
Any updates on this? 

I saw in a german forum, that Sungrow is actively working on fixing the WiNet-S stuff. But no good news, yet

MatteoSalviTech, 2022-02-28:
Still not working on my inverter.

eragooon, 2022-03-29:
I'm having the same problems with the SH8.0RT, I just updated all of the firmware, but nothing worked.
The firmware seemed to be from Dec. 14th 2021. And the QModMaster also allowed holding register read, but no input register read. 
Does anyone have the link to the sungrow team saying that they will fix this?

mkaiser, 2022-04-01:
I have my intel from here (german)
https://www.photovoltaikforum.com/thread/158136-sungrow-sammelthread-produktmanagement/?postID=2557804#post2557804

```
Die LAN-Buchse des Sungrow Hybrid ist bevorzugt für Modbus TCP zu verwenden, iSolarCloud Monitoring hat daran nur 5 Minuten Aktualisierungsrate.

Der WiNet-S Ethernetanschluss ist bevorzugt für iSolarCloud Monitoring zu verwenden, die Aktualisierungsrate liegt bei 10s wenn aktiviert und gute Internetverbindung vorausgesetzt. Modbus-TCP ist damit leider derzeit nur eingeschränkt bis gar nicht möglich bis die Firmware upgedated ist, dies steht kurz vor dem Abschluss.
```

Short translated version:

```
Use the LAN port of the Sungrow Hybrid to use ModBus TCP. Using this port the iSolarcloud will have a refresh rate of 5 minutes.

The WiNet-S Port should be used for iSolarCloud Monitoring with a refresh rate of 10seconds.  Modbus TCP is very limited to unusable on this port. A firmware upgrade for this is quiet near to a release. 
```

What port did you use? Can you send us a picture?

MatteoSalviTech, 2022-04-01:
Personally, I only have the ethernet port of the WiNet dongle.

eragooon, 2022-04-05:
I do have both, but the I do not have a LAN cable anywhere near the device.
So I'm hoping they will fix the Modbus-TCP for the WiNet-S dongle soon.

mkaiser, 2022-05-20:
hi, 

Just read a recent post from a german forum from a (official) Sungrow product management team:

`Beim Sungrow SGxxRT (normaler WR) sollte seit dem Firmwareupdate Anfang diesen Monats Modbus TCP über den WiNet-S möglich sein.`

rough translation:
`The newest firmware, released early this month, should add support MODBUS TCP via WiNet-S for the Sungrow SGxxRT (including non-hybrid inverter) `

https://www.photovoltaikforum.com/thread/166134-daten-lesen-vom-sungrow-wechselrichtern-modbus/?postID=2622220#post2622220

@eragoon and @MatteoSalviTech 
Can you check this?

MatteoSalviTech, 2022-05-20:
I continue to get the "Illegal Data Read" on QModMaster. Maybe because I own a SHx series inverter. My actual firmware version is: SUNSTONE-H_01011.02.19.

mkaiser, 2022-07-05:
Modbus TCP via WinetS works for a collegue of mine with a SH8RT now. 

- plug LAN cable into Inverter (not WinetS)
- Look in router for PC*** device and assign a static IP in the range of .20 to 240. (the lower range seems to not work...)

Note
PC** is the inverter 
Espressif*** is the WinetS Dongle 

maybe this helps?

mkaiser, 2022-07-06:
picture of SGH8.RT with working modbus from my collegue. LAN is connected directly to the inverter, not via WiNetS

[image]

MatteoSalviTech, 2022-07-06:
The problem is that on my SH6.0RS there isn’t any additional Ethernet port…

bjeanes, 2022-07-12:
> The problem is that on my SH6.0RS there isn’t any additional Ethernet port…

Same with my SH5.0RS

> I get the “Illegal Data Read” error but I can access the WiNet dongle on web browser…

Make sure you update the WiNet-S firmware. Mine came with with `WINET-SV200.001.00.P008` and I was able to upgrade to `WINET-SV200.001.00.P012`. This enabled Modbus access -- however, many registers are inaccessible, such as `13001` (running state). Other registers, such as `5036` (grid frequency) are off by a factor of 10 (showing me 500.9 Hz right now)!

Hopefully it improves, but I've had better getting info using `SunGather` with the `http` connection type (which essentially gets values via the web interface).

Bullischorsch, 2022-08-14:
> Anyone knows if it is possible to connect two Ethernet cables simultaneously to use the cloud and local Modbus TCP together?

Yes, for me this works on my SH10RT. LAN-Port for Modus, Ethernet-Port on Wifi-Dongle for WiNet-Communication (iSolarCloud).

mkaiser, 2022-09-05:
@MatteoSalviTech see #18  - can you check for newer versions?

@bjeanes is there an update to your current firmware from 12.7. Does the newest version maybe fix the running state register? Or were you already on the current version?

bjeanes, 2022-09-05:
Interesting. I don't see a 12.7. I see `WINET_B012.V000.P001` in iSolarCloud app but that's the same as I've had for a couple of months. They probably just mean the latest firmware (vs the one it ships with) includes Modbus support. It does, it's just buggy...

schoofseggl, 2022-09-13:
> Yes, for me this works on my SH10RT. LAN-Port for Modus, Ethernet-Port on Wifi-Dongle for WiNet-Communication (iSolarCloud).
The same applies to SH8RT. The dongle provides two interfaces: LAN and WIFI both operate with different MAC addresses and get different IP addresses. In parallel I have connected ETH0 (the internal one) to LAN and I operate the WiNET-S through WIFI (because I have only one LAN cable available currently)

Maxify23, 2022-11-07:
Please Update the Firmware. 

Most Modbus Registers now work with WII Net Dongel, but not all.... 

I send a bug report to sungrow for fixing

schoofseggl, 2022-11-09:
Update the firmware of the WII Net Dongle what? Do you know any instructions or firmware images? I found nothing to update it...

johschmitz, 2022-11-09:
You need to use the iSolarCloud app for the firmware updates.

MatteoSalviTech, 2022-11-09:
Just updated WiNetS to WINET_B012.V000.P001 but nothing changed...

bjeanes, 2022-11-09:
Restart Home Assistant too. https://github.com/home-assistant/core/issues/68161

MatteoSalviTech, 2022-11-09:
Restarted it of course but I continue to get so many unavailable sensors...

mkaiser, 2023-01-16:
There were no updates any more.

The LAN port of the WiNet is not supported. Will update the documentation accordingly

I guess it is okay to close the issue now.

closed 2023-01-16 by mkaiser