mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #9: Add support to set the min / max SoC

mkaiser, 2022-02-28:
ToDo for me :)

mkaiser, 2022-06-06:
will release soon. Here for the impatient...

just copy the content of the file at the end of your modbus_sungrow.yaml

This creates two new input sliders. 

open issue: After writing a new min / max value, the current min / max SoC values are only updated in the next modbus update cycle (aka scan interval), which is currently set to 60seconds

[setSoc.yaml.txt](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/files/8848729/setSoc.yaml.txt)

lindehoff, 2022-06-07:
Hi,
Shouldn't the holding register address for max Soc be `13058` and `13059` for min Soc?
`13057` seams to be Battery capacity.

lindehoff, 2022-06-07:
Ok, nice "feature" I did not know that 🙇

lindehoff, 2022-06-07:
We could also add sensors for these, like so 
```yaml

      - name: Battery Min. SOC
        slave: !secret sungrow_modbus_slave
        address: 13058
        input_type: holding
        count: 1
        scan_interval: 600
        data_type: uint16
        swap: word
        precision: 0
        unit_of_measurement: "%"
        state_class: measurement
        scale: 0.1

      - name: Battery Max. SOC
        slave: !secret sungrow_modbus_slave
        address: 13057
        input_type: holding
        count: 1
        scan_interval: 600
        data_type: uint16
        swap: word
        precision: 0
        unit_of_measurement: "%"
        state_class: measurement
        scale: 0.1
```
right?

mkaiser, 2022-06-07:
oh, I kind of already have this lines in my version, but left it out copying to the file above.... I hope I did not cause you much debugging

attached is my current state, which includes the holding registers you meantioned. 
I will work on your other requests for setting the EMS. If that worls I will do a proper commit with all modifications merged

[modbus_sungrow.yaml.txt](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/files/8856112/modbus_sungrow.yaml.txt)

lindehoff, 2022-06-07:
Great, thanks a lot. 👍