mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #13: Limit max charge/discharge of battery

Bullischorsch, 2022-08-24:
I integrated a sensor for register 33046 "Max charge power"
```
      - name: Battery Max Charge Power
        slave: !secret sungrow_modbus_slave
        address: 33046 # 33047
        input_type: holding
        count: 1
        data_type: uint16
        swap: word
        precision: 0
        unit_of_measurement: W
        device_class: power
        state_class: measurement
        scan_interval: 10
```
and a realting input_number
```
  set_sg_charge_discharge_power_percentage:
    name: max charge discharge power in W
    #initial: 40
    min: 0
    max: 5000
    step: 50
```

Idea is to limit charging to 1W until 12am if forecast for the day is more than 20 kWh for example. So the 70% cut can be used to charge the battery .
Unfortunately this doesn't work in EMS "Self consumption mode", I have to change to forced mode.
Does anybody know a solution? I don't want to change EMS mode.

mkaiser, 2022-08-25:
hi, 

I can't find anything related to the address 33046 # 33047 in the datasheet (v1.0.23)

where did you find that register?

mkaiser, 2022-08-26:
what about setting the maximum SOC to adjust (or basically start/stop) charging the battery?

set it to a low value in the mornings and increase max Soc when the 70% limit would be triggered

Hopefully the 70% limit will be obsolete in the near future. Right now there is always the question for me how anybody could possibly supervise it without coming to your house and actually seeing the inverters parameter...

link is in german - sorry -, but that questionable law is also german....
https://www.energie-experten.org/news/jetzt-auch-fuer-bestandsanlagen-habeck-kassiert-70-prozent-abregelung-komplett

Bullischorsch, 2022-08-26:
> hi,
> 
> I can't find anything related to the address 33046 # 33047 in the datasheet (v1.0.23)
> 
> where did you find that register?

Hi
Found it here:
https://www.photovoltaikforum.com/thread/171744-sungrow-wechselrichter-zeiten-zum-akkuladen-definieren/

Bullischorsch, 2022-08-26:
> > hi,
> > I can't find anything related to the address 33046 # 33047 in the datasheet (v1.0.23)
> > where did you find that register?
> 
> Hi Found it here: https://www.photovoltaikforum.com/thread/171744-sungrow-wechselrichter-zeiten-zum-akkuladen-definieren/

> what about setting the maximum SOC to adjust (or basically start/stop) charging the battery?
>
> set it to a low value in the mornings and increase max Soc when the 70% limit would be triggered [...]

Yes, that would be great

mkaiser, 2022-08-26:
thanks for the links!

I just pushed a new yaml to the testing branch. Setting a limit for charging / discharging works for me now :)
(tested by setting via Home Assistant and reading back via local iSolarcloud)

Can you try it and give me a feedback?

In your linked threads, there was also something about battery capacity ("# h,33049,battery_capacity,"Batterie Kapazität","kWh",U16,,,0.01")

But I could not get that to work :/
Does it work for you? (I left three " TEST TEST Battery Capacity 330** " sensors in the yaml)

Getting the battery capacity is basically the only thing I am missing right now. Somehow I think that might be related to my semi-supported Pylontech Force H1 battery....

Bullischorsch, 2022-08-27:
> thanks for the links!
>
> I just pushed a new yaml to the testing branch. Setting a limit for charging / discharging works for me now :) (tested by setting via Home Assistant and reading back via local [...]

Hi

for my with a BYD HVS Battery (10 kWh) at the sh10rt the capacity-registers show:

TEST TEST Battery Capacity 33049 shows 6553.5
TEST TEST Battery Capacity 33050 shows 6554
TEST TEST Battery Capacity 33051 shows 6554

Seems not to work also.

regards 
Georg

mkaiser, 2022-08-27:
Found the culprit by using QModMaster and reading the register manually....

I forgot to declare the sensor with the correct "input_type: holding"

This works for me: 
      - name: Battery Capacity 
        slave: !secret sungrow_modbus_slave
        address: 33048  #33049 
        input_type: holding
        count: 1
        data_type: uint16
        swap: word
        unit_of_measurement: kWh
        device_class: energy
        scale: 10
        scan_interval: 600
		
		
Can you double check? 

If that works - are there any more features missing?

bjeanes, 2022-08-27:
On SH5.0RS that register just returns `65535`, likewise for `33046`. I have heard that it does work on the SH#RT inverters though

mkaiser, 2022-09-23:
original request (limit battery charge/discharge) is integrated into main. Closing this issue

closed 2022-09-23 by mkaiser

OhneNik, 2023-05-17:
Hey ! Just ran into an issue were I set a limit and afterwards I can’t remove it any longer. Also the range is too small, e.g. Sungrow SBR192 is rated for 13,4 kW sustained. Writing a 0 to that register is an invalid Value (https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/58). As a Workaround I manually set the range up to 15000 as that’s my peak for that battery.   

So what’s the correct way to remove the limit ?

Louis712, 2023-05-17:
You can set the limit to 15000 in the sungrow_modbus.yaml. There are two input_numbers used for the battery charging/discharging power limit. You can change the max to 15000.