mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #15: homeassistant zeigt nicht alles an, was mache ich falsch?

AH727AH, 2022-08-28:
Hallo Herr Kaiser , erst mal vielen Dank für die Integation des Sungrow WR in Homeassistant.
Leider wird nicht alles angezeigt siehe Anhang.
Auch werden die Binäeren Werte nur mit off angezeigt. Habe heute das neue Update für den SH10RT gemacht, hat keine Änderung gebracht.
Haben Sie eine Idee was da nicht stimmt?

Vielen Dank im Vorraus

André

[image]

mkaiser, 2022-08-28:
Hi André, 

das liegt bestimmt noch an mir, weil ich viel im Testing branch herumprobiert habe, aber das Dashboard noch nicht wieder richtig zusammengeklickt habe....

Kommt bald, muss nur mal wieder einen Abend Zeit haben :)

Das ist aber schon merkwürdig, dass Inverter State, Running State und System State alle '0' sind. 

Aus diesen Statusregistern werden dann die binären Werte abgeleitet, weshalb diese alle ebenfalls 'Aus' sind. 

Ist es denn ein SH10RT, wie im Screenshot angezeigt?

Grüße aus Bielefeld, 
Martin

AH727AH, 2022-08-29:
Hallo Martin,
Vielen Dank für die schnelle Antwort.
Es ist ein SH10RT

Mit freundlichen Grüßen 
André Hertwig

kaptenpeter, 2022-09-01:
I have the exact same problem.

I also get these error messages in the log.

[image]

I have an SH8.0RT inverter.

mkaiser, 2022-09-01:
Does this fix it?

https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/commit/346c2e9da04615b9cf22e9f09dc64dd86efca04d

splitted the one file into two: One for Status (readonly) and one for control. I hope this will prevent myself on doing unwanted stuff on my mobile

f-zappa, 2022-09-19:
Hi, for my inverter (SH10RT) my state readings are also zero. Can I provide anything helpful for debugging this?
Regards, Uli

[image]

f-zappa, 2022-09-20:
Checked the system state register via command line today and the value is as expected. So this seems to be not caused by the inverter but somewhere inside home assistant.

```
ulrich@hp:~$ mbpoll -a 1 -r 12999 -t3:hex -0 -c1 sungrow-modbus
mbpoll 1.0-0 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright © 2015-2019 Pascal JEAN, https://github.com/epsilonrt/mbpoll
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: Modbus TCP
Slave configuration...: address = [1]
                        start reference = 12999, count = 1
Communication.........: sungrow-modbus, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, input register table

-- Polling slave 1... Ctrl-C to stop)
[12999]:        0x0040
```

sniffing the transmission gives me

```
T 10.0.3.7:53340 -> 10.3.7.3:502 [AP] #18
  00 3a 00 00 00 06 01 04    32 c7 00 01                .:......2...    
#
T 10.3.7.3:502 -> 10.0.3.7:53340 [AP] #19
  00 3a 00 00 00 05 01 04    02 00 40                   .:........@     

``` 
"tell me contents of 1 input register, starting with 0x32c7"
- "sure .. the value is 0x0040"
- 
now I sniff the communication between hass and the inverter
```
T 10.10.7.3:57641 -> 10.3.7.2:502 [AP] #18
  15 1a 00 00 00 06 01 04    32 c7 00 01                ........2...
#
T 10.3.7.2:502 -> 10.10.7.3:57641 [AP] #19
  15 1a 00 00 00 05 01 04    02 00 00                   ...........
```
**-> same question, but another response(0x0)! what is going on here?**

f-zappa, 2022-09-21:
TLDR; WinetS adapter is buggy, use the other LAN port for modbus.

It seems that yesterday evening I was to tired to see the difference. With "mbpoll" connection was made to the internal LAN-Port of the inverter, while home assistant connected to the LAN cable running into the WinetS adapter. Changed this now and all the states (and derived readings) are there. 
[image]
Seems like WinetS has:
PORT     STATE SERVICE
80/tcp   open  http (internal webserver)
443/tcp  open  https (?)
502/tcp  open  mbap (buggy modbus instance)
8082/tcp open  blackice-alerts (?)

The other LAN port only has 502/tcp for modbus, but here all data registers are available.
You better don't portscan this port, my inverter didn't like this and I had to physically re-connect the port before I could make a modbus connection again.

mkaiser, 2022-09-23:
good thing that you are capable of debugging like this :)

I guess this issue is solved then?

Also there is a major update including the dashboard in the main branch now...  Please open a new issue if anything does not work!

closed 2022-09-23 by mkaiser