mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #16: Two inverters

bergh-io, 2022-09-01:
First off, great work! 

I have two SH10RT, is it possible to add both? I only have battery on one of them, the master, and that works like a charm but for information like current solar power, daily yield and such its not enough to add the master as it doesn't show the stats from the slave.

mkaiser, 2022-09-01:
hi, 

should be do-able somehow:

First guess (there might be other options):

- Keep the yaml for the fist (master?) inverter like provided in the git. 

Copy and adapt the three secrets/ variables to match the second inverter:

```
secrets.yaml:
#first inverter
sungrow_modbus_host_ip: 192.168.178.20 # TODO update with the IP of your inverter. No default. Check your router.
sungrow_modbus_port: 502 # TODO update with the Modbus port of your inverter. Default is '502'
sungrow_modbus_slave: 100 #TODO update with the slave address of your inverter. Default is '1'
#second inverter
sungrow2_modbus_host_ip: 192.168.178.21 # TODO update with the IP of your inverter. No default. Check your router.
sungrow2_modbus_port: 502 # TODO update with the Modbus port of your inverter. Default is '502'
sungrow2_modbus_slave: 100 #TODO update with the slave address of your inverter. Default is '1'
```

Copy modbus_sungrow.yaml and rename it to "modbus_sungrow2.yaml" and adapt sungrow2.yaml:
- find and replace all three secrets 
- give identical names to each sensors, automations and inputs (e.g. change "Total PV generation & battery discharge" to " SG2 Total PV generation & battery discharge" (or similar)

Hope that helps :)

bergh-io, 2022-09-02:
Thanks! I did try that quickly but did not got it to work, but I'm sure its because I did not rename the sensors etc unique names, I'm guessing you mean unique instead of identical in you last section? But I'll give it another go :)

Not related to this question and maybe I should create a new topic but I dont seem to get the import/export figures in HA. 
Right now I'm draining my battery because of hing spot prices but its not registering and the imports have not registered either, in HA, I see the numbers in iSolarCloud.
[image]
[image]

mkaiser, 2022-09-05:
cool!

Would you share your changes for other users?

bergh-io, 2022-09-05:
Yes, I will but right now I’m getting completely different numbers from modbus and iSolarCloud. so I’m going to hold of until I get it to work correctly. But it shouldn’t be the code for this though.
I do think that modbus is wrong but I can’t figure out why so I have asked Sungrow for help. It probably has something to do with the communication between the inverters or that it is different modbus address when you have two but I’m not sure why that would be.

Matze5593, 2023-08-15:
@mkaiser 

hi, i have problems with the power mppt 1/2. I don't understand how the code works and my mistake lies. I just don't get any values ​​for the power. Could you please help me

`
      - name: MPPT1 voltage inv 2
        unique_id: sg_mppt1_voltage_inv2
        slave: !secret sungrow_modbus_slave_inv2
        address: 5010 # reg 5011
        input_type: input
        count: 1
        data_type: uint16
        swap: word
        precision: 1
        unit_of_measurement: V
        device_class: Voltage
        scale: 0.1
        scan_interval: 10

      - name: MPPT1 current inv 2
        unique_id: sg_mppt1_current_inv2
        slave: !secret sungrow_modbus_slave_inv2
        address: 5011 # reg 5012
        input_type: input
        count: 1
        data_type: uint16
        swap: word
        precision: 2
        unit_of_measurement: A
        device_class: Current
        scale: 0.1
        scan_interval: 10

      - name: MPPT2 voltage inv 2
        unique_id: sg_mppt2_voltage_inv2
        slave: !secret sungrow_modbus_slave_inv2
        address: 5012 # reg 5013
        input_type: input
        count: 1
        data_type: uint16
        swap: word
        precision: 1
        unit_of_measurement: V
        device_class: Voltage
        scale: 0.1
        scan_interval: 10

      - name: MPPT2 current inv 2
        unique_id: sg_mppt2_current_inv2
        slave: !secret sungrow_modbus_slave_inv2
        address: 5013 # reg 5014
        input_type: input
        count: 1
        data_type: uint16
        swap: word
        precision: 2
        unit_of_measurement: A
        device_class: Current
        scale: 0.1
        scan_interval: 10

      - name: Total DC power inv 2
        unique_id: sg_total_dc_power_inv2
        slave: !secret sungrow_modbus_slave_inv2
        address: 5016 # reg 5017
        input_type: input
        count: 2
        data_type: uint32
        swap: word
        precision: 0
        unit_of_measurement: W
        device_class: power
        state_class: measurement
        scale: 1
        scan_interval: 10

  - sensor:
      - name: MPPT1_inv2 power
        unique_id: sg_mppt1_power_inv2
        unit_of_measurement: W
        device_class: power
        availability: "{{states('sensor.mppt1_voltage_inv2')|is_number and states('sensor.mppt1_current_inv2')|is_number }}"
        state: "{{ (states('sensor.mppt1_voltage_inv2') | float * states('sensor.mppt1_current_inv2') | float) |int }}"

      - name: MPPT2_inv2 power
        unique_id: sg_mppt2_power_inv2
        unit_of_measurement: W
        device_class: power
        availability: "{{states('sensor.mppt2_voltage_inv2')|is_number and states('sensor.mppt2_current_inv2')|is_number }}"
        state: "{{ (states('sensor.mppt2_voltage_inv2') | float * states('sensor.mppt2_current_inv2') | float) |int }}"

reesaroo74, 2023-11-12:
FYI: https://github.com/reesaroo74/Sungrow-SHx-Inverter-Modbus-Home-Assistant

I've managed to get my SH10RT and SH10RS (independent systems) up an running.  Few wrinkles to work out with the installer because they've clamped the wrong lines in the main board for one of my systems.

I'm pretty new to yaml and only found out that the unique id in hass is generated from the name ... and the unique_id that Martin has wired up is only used to allow Ui interaction - https://community.home-assistant.io/t/template-sensor-unique-id-is-ignored/485206.

We have three phase, but for the system that is on the three phase line to the house, we had to install an RS on a single phase because the RT has an 18A per phase limit due to the EMS breaker (we'd might have kicked that with a spike if we had an oven on and the A/C kicked in).

I'm not sure if anyone has an SH10RS out there, but I had to connect my RJ45 hard-wired LAN connection line into the end of the wifi-dongle.  There are two RJ45 ports (under a large cover with the EMS connector - so not on its own like in the RT).  I tried connecting directly to those, but wasn't able to connect to the modbus.  Not sure if they are for something else?

Gnarfoz, 2024-05-23:
The user manual for the SH10RS describes the connectors within the COM2 terminal in section 6.2. Neither of the RJ45 ports are Ethernet ports, I'm afraid. So Modbus TCP directly to the inverter (like on the SH10RT) is not available.
There's a RS485 connection (A1/B1) but I think you're probably using that already (should be connected to your SH10RT).
There's also the connector for the energy meter (A2/B2). I'm not sure if that uses Modbus.

If you're desperate, you could check if it's possible to loop in some kind of Modbus RTU serial converter between the inverter and the WiNet-S dongle. But be aware, since the dongle is a Modbus master, and there can only be *one* Modbus master active at a time, you'd have to be very careful that the converter never sends anything when the dongle does.
Maybe passively reading everything would be good enough. Either way, that's probably a lot of effort.