mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #17: Differences SH*RT and SH*RS - battery capacity

mkaiser, 2022-09-01:
I can retrieve the battery capacity for my SH10.RT with this register

```
 - name: Battery Capacity 
        slave: !secret sungrow_modbus_slave
        address: 33048  #33049 
        input_type: holding
        count: 1
        data_type: uint16
        swap: word
        unit_of_measurement: kWh
        device_class: energy
        scale: 10
        scan_interval: 600
```

bjeanes reported, that it does not work on an SH5.0RS  https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/13#issuecomment-1229277162 

It would be nice to have more support for the **RS** (not RT)

Are there any more differences?

bjeanes, 2022-09-23:
I think that's a better question for Sungrow as a company. Their Modbus implementation is all over the place and registers can work differently depending on the battery pack and firmware versions, as far as I've seen. Most notably for me, when my SH5.0RS arrived, `battery_power` was _signed_ and was negative when charging, positive when discharging (despite docs showing it was _un_signed. A firmware upgrade changed that, which is kind of annoying actually, because having it signed was far more useful!

mkaiser, 2023-07-07:
No progress here, closing....

closed 2023-07-07 by mkaiser