mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #18: Winet-S

nygana, 2022-09-05:
I just got this message from Sungrow

The Modbus TCP is added to the latest FW version of the WiNet dongle.

Med vänliga hälsningar | Best Regards 

William Strijks
Service Manager Sweden

Does it work with Winet-S, has anyone tested

philkman, 2022-09-05:
Yes, works for me with the latest firmware

schoofseggl, 2022-09-13:
Just to confirm: you have connected it through WIFI / LAN (the WInet-S dongle) and NOT through the integrated eth0 and you can read in all relevant registers? 
I just tried today (SH8RT) and the only way it worked for me is via eth0 (the integrated one).

nygana, 2022-09-13:
Hi

I have not had time to test it after the message from sungrow support yet. Gonna try it, and wanted to relay the info to the project

 
Mvh 
Anders Nygren

076 107 29 04

philkman, 2022-09-14:
The LAN-cable is connected directly to the Winet-S, no WiFi connection and no LAN-cable to the integrated LAN-Port. 

Unfortunately, I do not know how to check the firmware of the Winet-S. So if anybody describes how to do it, I will post the exact firmware.

schoofseggl, 2022-09-14:
And with this setup, were you able to read out ALL registers that are in the yaml? Or just some registers? With my WinNet-S I could connect to modbus and read _some_ holdings at around register 5000 to 5100, but e.g. not in the register area 13000 where according to the register PDF (or the modbus yaml) the interesting stuff should be. Many registers just returned "-1" for or generated connection issues. 

Regarding the Firmware of the Winet-S: It would be interesting to know if you can update it. My inverter has version M_WiNet-S_V01_V01_A. (Looked  up throught the Web interface) But I have no clue to check if this is up to date or if I can update it.

philkman, 2022-09-14:
Everything except the states could be read out (also registers in the 13'000-area).

This is how my HA-dashboard for the inverter looks like:
[image]

I know, that once I saw the WiNet-S-Version somewhere but I have no clue, where to get it. Which Web-Interface are you talking about?

SW | Version
--|--
ARM-Softwareversion | SAPPHIRE-H_01011.01.19 
MDSP-Softwareversion | SAPPHIRE-H_03011.01.17
SDSP Softwareversion | SUBCTL-S_04011.01.01

Did you enable Modbus under http://{WinetS-IP}/#/system/agreements/ProtocolConfigure
[image]

schoofseggl, 2022-09-14:
Yes, Modbus is enabled - but as I mentioned I can just get some values via Modbus, but not all. Which version do you see in the Web Interface of the WinetS? I have mine from http://yourWinetSIP/#/about/about

andi-blafasl, 2022-11-29:
Just added your configuration to my Home Assistant and connected via WiNet-S to my SH6.0RT with an BYD HVS 10.2 attached. As far as I can see, everything is working and all entities are showing some values. Hopefully tomorrow will have some sun to see if battery is charging. Right now it is empty and I cant tell anything about it 😁

mkaiser, 2023-01-16:
Modbus via WiNet is still not working. Added that to the documentation.

closed 2023-01-16 by mkaiser

andi-blafasl, 2023-01-16:
With the latest firmware it is working nearly complete. Only the binary sensors are not working with my connection through the winet-s.