mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #20: Battery SoC Reserve don´t seem to work

NBD99, 2022-10-01:
I have a SH10RT and a SBR98 battery when i try to set the Battery SoC Reserve value nothing happens it remains at 0%, all the other set parameters seem to work fine.

mkaiser, 2022-10-10:
yep, I can confirm this. 

Somehow while testing some time ago, it was somehow working...

I took just a quick peak at the sungrow mobile app, but could not find the reserve values there. 

Or did you set/ get the values via the installer's webinterface admin account?

NBD99, 2022-10-11:
I am not using the mobile app but i can´t find the values in the local webinterface either. The only SOC values i can find is under Energy management parameters and they are related to forced mode.

mkaiser, 2023-01-24:
fixed in c2fd05352331ac73f3556ea5b6cd19756f527aed

closed 2023-01-24 by mkaiser