mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #21: Request: Be able to set Energy Management Parameter

bergh-io, 2022-10-04:
I'd love to be able to set the parameters for the energy management if possible some how. 
This would allow to automatically charge the battery from the sun or the grid depending on which is cheaper that particular day.
I know that i can just change EMS Mode if i don´t want to charge the battery from the sun but if i still would like to use the battery instead of the grid if the sun goes away in a cloud or something I'd have to change EMS mode quite often depending on the actual solar production and price instead of just the price. Id suspect that it´s not that god for the system to manually set EMS mode efery few minutes in worst case.

Example: 
Between 07:00-20:00 I'd like to use self-consumption mode to use the energy in the battery instead of buying from the grid. I'd also like to charge the battery from the sun only if the price is lower than X money (0.95 SEK in my case). 
Therefore I'd like to check every full hour if the price is over X money and then set Forced charge percentage to 0 if true, otherwise to 100%. I also have a "backup" that charger the battery at night if not already full.

[image]

Battlier, 2022-10-10:
I would also very much like this to be possible.

mkaiser, 2022-10-10:
hmm thats a really reasonable feature.

The yaml integration already provides all required parameters / configs for this. 

Holding registers
- Forced mode instead of the auto EMS mode "addr 13049, EMS Mode Selection raw"
- select between charge and discharge: "addr 13050, Battery Forced Charge Discharge Cmd raw"
- setting the power for charging / discharging  "addr 13051, Battery Forced Charge Discharge Power"

What is missing is a control logic to determine the states and the power / energy to charge/ discharge. I never checked: are there some EMS (Energy management solutions) available. Maybe something similar to EVCC, which primary targets electric vehicle energy management?

From where do you get the information about the pricing? If it is already available in Home assistant you can start writing automations based on the lower part of my modbus_sungrow.yaml file (Last section about Automation)

mkaiser, 2022-10-10:
maybe have a look here, bjeanes made something similar

https://github.com/bjeanes/ha-config/blob/main/config/packages/sungrow.yaml

purcell-lab, 2022-10-12:
> What is missing is a control logic to determine the states and the power / energy to charge/ discharge. I never checked: are there some EMS (Energy management solutions) available.

The excellent EMHASS add on can provide this desired control logic. It takes into account your buy/sell price as well as forecasts for PV production and household loads and determines the optimal schedule for battery and other household loads, which it exposes as HA sensors that you could feed into this integration to set battery charging/ discharging profiles.

https://emhass.readthedocs.io/en/latest/intro.html

[image]

Battlier, 2022-10-12:
Thanks all for the info, lots of possibilities. A question I guess that “force discharge” is not equal to to “discharge”? The discharge option only uses the battery to supply the house with necessary power but does not export to grid, is there any way to set the system into discharge mode?

Battlier, 2022-11-13:
Hi again
It seems like there is something I am missing here. Has anyone found a way to update the time of allowed discharging (as in post by @bergh-io, parameter 1-4) or should you update the discharging energy based on the power needed momentanously?

mkaiser, 2022-11-13:
Hi, 

sorry, I missed your post earlier. 

While writing this answer and making some screenshots to describe it better, I found out, that I made a small mistake in the "PV Control" YAML page. In the original version the "forced charge disscharge cmd" was as readonly status. The drop down menu, I created was missing :/

I fixed it in the git. Just update the "PV_Control.yaml".  I guess that was the important missing part :)

> A question I guess that “force discharge” is not equal to to “discharge”? The discharge option only uses the battery to supply the house with necessary power but does not export to grid, is there any way to set the system into discharge mode?

The mode, that only supplies the load within the house is called "Self-consumption mode" and is set by default. The register EMS mode controls it's behaviour. 

If you want to charge  or discharge the battery from the grid you need to set the EMS mode to "Forced mode" (see screenshot). 
In addition you need to tell the inverter if it should charge or discharge the battery (see second screenshot) and set the forced charge discharge power in W (3rd screenshot). Do not mixup the "forced charge discharge power" with the two separate parameters "max charge power" and "max discharge power". These two control the battery behaviour in the EMS self consumption mode. 

For creating a customized energy management system you need set all 3 parameters and tune the last parameter at run-time. 

[image]
[image]
[image]

closed 2022-12-01 by mkaiser

GrandAmmor, 2023-01-21:
> https://emhass.readthedocs.io/en/latest/intro.html
> 
> [image]

How did you get that beautiful UI?

elektrinis, 2023-01-21:
> How did you get that beautiful UI?

Would really love it if he could share the config :)

GrandAmmor, 2023-01-21:
> > How did you get that beautiful UI?
> 
> Would really love it if he could share the config :)

This is what I wanted to say :p That would be really nice

MonsterD1, 2024-01-15:
> The excellent EMHASS add on can provide this desired control logic. It takes into account your buy/sell price as well as forecasts for PV production and household loads and determines the optimal [...]

very nice UI ... would you please share yure code/config?

commit 30237d24 (2022-11-13): fixed #21: Input select for forced battery charge / discharge was broken