mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #23: Feature Request to add Grid Voltage & Frequency Parameters

drschlaumeier, 2022-10-06:
Hallo,
I have added the following ModBus paramters to also monitor Grid Voltage & Frequency. Maybe you also add this to modbus_sungrow.yaml.

      - name: Phase A Voltage
        slave: !secret sungrow_modbus_slave
        address: 5018
        input_type: input
        count: 1
        data_type: int16
        swap: word
        precision: 1
        unit_of_measurement: V
        device_class: Voltage
        state_class: measurement
        scale: 0.1
        scan_interval: 10

      - name: Phase B Voltage
        slave: !secret sungrow_modbus_slave
        address: 5019
        input_type: input
        count: 1
        data_type: int16
        swap: word
        precision: 1
        unit_of_measurement: V
        device_class: Voltage
        state_class: measurement
        scale: 0.1
        scan_interval: 10

      - name: Phase C Voltage
        slave: !secret sungrow_modbus_slave
        address: 5020
        input_type: input
        count: 1
        data_type: int16
        swap: word
        precision: 1
        unit_of_measurement: V
        device_class: Voltage
        state_class: measurement
        scale: 0.1
        scan_interval: 10

      - name: Grid Frequency
        slave: !secret sungrow_modbus_slave
        address: 5035
        input_type: input
        count: 1
        data_type: int16
        swap: word
        precision: 1
        unit_of_measurement: "Hz"
        device_class: frequency
        state_class: measurement
        scale: 0.1
        scan_interval: 10

mkaiser, 2022-10-10:
thanks, will put that in the next release after some more testing. 

Do you want to make a pull request for this? Otherwise I will copy your code ( I guess the data_type should be uint, doesn't it?)

drschlaumeier, 2022-10-13:
Hallo mkaiser,

thanks, yes, uint16 ... sorry, copy/paste error.
Please put into next release.

BTW:
I'm only using voltage monitor ... but not current monitor: 13030 - Phase A Current, 13031 - Phase B Current, 13032 - Phase C Current ... all sint16 ... maybe we should also add for completeness for 3P4L and 3P3L inverters? 

Thanks

mkaiser, 2022-11-04:
just committed. Can you test and give me a quick feedback?

drschlaumeier, 2022-11-18:
Sorry for the delay...I was unavailable. Yes, its working. Thanks.
I have two more register to map:

      - name: Bus Voltage
        slave: !secret sungrow_modbus_slave
        address: 5147
        input_type: input
        count: 1
        data_type: uint16
        swap: word
        precision: 1
        unit_of_measurement: "V"
        device_class: voltage
        state_class: measurement
        scale: 0.1
        scan_interval: 10

      - name: Array Insulation Ressistance
        slave: !secret sungrow_modbus_slave
        address: 5071
        input_type: input
        count: 1
        data_type: uint16
        swap: word
        unit_of_measurement: "kOhm"
        state_class: measurement
        scale: 1
        scan_interval: 10

I checked it and its working fine! Please also add this two signals.

Thanks & BR

drschlaumeier, 2022-11-26:
Edit: Typo again :-) 5070 instead 5071

  - name: Bus Voltage
    slave: !secret sungrow_modbus_slave
    address: 5147
    input_type: input
    count: 1
    data_type: uint16
    swap: word
    precision: 1
    unit_of_measurement: "V"
    device_class: voltage
    state_class: measurement
    scale: 0.1
    scan_interval: 10

  - name: Array Insulation Ressistance
    slave: !secret sungrow_modbus_slave
    address: 5070
    input_type: input
    count: 1
    data_type: uint16
    swap: word
    unit_of_measurement: "kOhm"
    state_class: measurement
    scale: 1
    scan_interval: 10

drschlaumeier, 2022-11-28:
> just committed. Can you test and give me a quick feedback?

I just noticed that you wrote V instead A for the Phase A/B/C Current ...

andi-blafasl, 2022-11-29:
[image]
Phase Voltage is working, Current and Power do not work.
Frequency Scale seems to be a bit off 😉

elektrinis, 2022-11-30:
Hi,
Is it possible to get data from smart meter?
I'd like to see actual power per phase from smart meter and from inverter (when in bypass backup mode). Currently I only get the overall power, but not individual phases.
Thanks.

drschlaumeier, 2022-12-01:
> [image] ~Phase Voltage is working, Current and Power do not work.~ Sorry, maybe there is no Current and no Power at the Moment 🤷 Battery empty and no Sun 😁 Frequency Scale seems to be a bit off 😉

Frequency: Interesting, its working for me. Which device, which FW ? Its working for me!
[image]

andi-blafasl, 2022-12-01:
I have a SH6.0RT with WiNet-S Dongle connected by it's LAN-Port. Don't know how to get the firmware version, but I made an update a few week ago.
It seems the binary sensors are not working as expected:
[image]
[image]
Battery is empty and no sun at 20:49 in Germany 😉 Maybe I should open a new Issue for this Problem.

elektrinis, 2022-12-01:
My SH10RT has a native LAN port (extra to one on the dongle) and this works well with it.
It does not use that to send tada to cloud though, so if you need that, you still have to have your dongle installed and connected. But I have just ditched it, as I am allergic to cloud services.

mkaiser, 2022-12-01:
hey, sorry I was quiet busy the last weeks. Here my comments

@ [drschlaumeier](https://github.com/drschlaumeier)
> - name: Bus Voltage
>     slave: !secret sungrow_modbus_slave
>     address: 5147
>     input_type: input
>     count: 1
>     data_type: uint16
>     swap: word
>     precision: 1
>     unit_of_measurement: [...]

Which voltage is measured here? I can't really guess, what the "bus voltage" is. Is it the CAN bus between battery / inverter or the RS485 connection between inverter and power meter in the electric cabinet?

Why do you need the insulation? I could integrate it, but I guess a larger poll interval (e.g. 10 minutes) would be sufficient, right?

Will provide an update including the typo (V vs. A) for the current, soon!

@andi-blafasl: 
the phase 'X' current is the output current of the inverter to the house loads (or maybe exporting, if the EMS mode is set accordingly). So without any sun your behaviour is as expected. 

[elektrinis](https://github.com/elektrinis) commented [yesterday](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/23#issuecomment-1332685259)
```
Hi,
Is it possible to get data from smart meter?
I'd like to see actual power per phase from smart meter and from inverter (when in bypass backup mode). Currently I only get the overall power, but not individual phases.
Thanks.
```

One note about that, which made me think for a couple of minutes earlier: 
In case of a 3 phase inverter (e.g. SH10.RT) the current is always balanced by the inverter, so every phase has more or less the same current. 
This does not have any disadvantages, because (I guess most of the ) power meters are based on "balancing counters" (german: saldierender Zähler). The meter justs counts /  adds up the incoming/ outgoing power of all three phases. 

The grid frequency is working well for me. Maybe there are (again) differences between your SH6RT and my SH10.RT

elektrinis, 2022-12-01:
> 

Here in Lithuania they are measuring each phase separately, that is, if I am feeding energy on two phases, they count it as energy fed, and if I take energy from third phase, they will count it as energy being bought, at a fee. Sneaky little bastards. Due to this reason I am looking in to ways to balance it out.

Sungrow states it balances the phases, but I don't see it, there are huge disbalances, according to what my house is consuming.

mkaiser, 2022-12-19:
that is quiet interesting, elektrinis, I always thought that all energy counters are balanced. 

I don't thing that there is a way to get the sungrow inverters to to a non-balanced power output. At least there is nothing in the datasheet / modbus register descriptions mentioning that. 

The original issue (grid and frequency) should be processed now. I am starting to get confused about all the open issues here, so lets close this issue. Please make new ones, if there are any more questions :)

closed 2022-12-19 by mkaiser

elektrinis, 2022-12-20:
Well in one of marketing materials sungrow advertise these inverters as capable of working with non-balanced loads, and they provide an example of various home appliance devices running on different phases, creating different loads. But that does not really make it more clear wat it actually does :) I was hoping for it to run all three internal inverters separately to keep all grid inputs at same power level.
Here's an interesting screenshot from their "isolarcloud" website.
It shows the same 7.9A current on each phase, but then different power on each "backup" phase. So I am not sure where those numbers are coming from, as those nubers don't really add up, of I just can't figure out the pattern/logic behind it.

[image]