mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #27: Install method via HACS?

elektrinis, 2022-12-01:
This integration is so cool, best I've found so far.
However I would like it to have a "standard" install method. It's not tooo difficult as it is now, but HACS would take care of updates.
Would do it myself, but it's a little too much.

mkaiser, 2022-12-01:
hey, 

thank you for your feedback!

I am still quiet new to Home Assistant and installed HACS a long time ago, but never actually used it. 

Having a plugin would indeed be very nice. I am really unhappy with the current installation and update procedure. Especially for non-professionals it is too much....

One thing that is bugging me is that sungrow's inverters behave differently (e.g. grid frequency, some battery stuff...). Therefore the HACS stuff could be quiet beneficial.

Unfortunately I have too many open projects at the moment and won't have any more spare time in the next half year (or more). If anyone would start that I could try to help, but my time is too limited to go for it from scratch :/

elektrinis, 2022-12-01:
I don't have the skills to do it as well (much less than you), so let's hope someone jumps in to help out on this.
Again, really cool project.
As for inverters... yeah.. I'm noticing weird stuff in the data. Will create a separate issue.

karelv, 2022-12-26:
It would be nice if you could give some instruction how to use the yaml files here on github... No idea how to install, even in a manual way.

karelv, 2022-12-26:
Sorry for my above remark, I see now... the readme is basically the manual install guide.
However, I followed it, but it does not display any values. Using `SH6.0RT`. I saw the web page at my local ip address, however the values do not show up in HA.

karelv, 2022-12-26:
At the `WiNet-S` web interface, I saw the ip address of my HA (RPi4), so I guess this is good!
[image]
... Maybe I have to be patient until tomorrow when sun is shining again!

Louis712, 2022-12-26:
Do the entitiys in HA say that it's unavailable or just that the value ist 0?

karelv, 2022-12-26:
It's unavailable... see here:
[image]

karelv, 2022-12-26:
Some of the values it seems readable:
[image]

karelv, 2022-12-26:
Also in the control, HA let change values...

Louis712, 2022-12-26:
That's some very weird behavior. Is your WiNet-S up-to-date?

karelv, 2022-12-26:
Thanks for quick response! How do I check this? And do you mean firmware wise or hardware wise? (the installation is done last september)

Louis712, 2022-12-27:
I meant hardware wise. You can check it by going into the iSolarCloud and there should be an option called Firmware Update.

Louis712, 2022-12-27:
Sorry, wasn't concentrated enough when I wrote the comment. Of course I meant firmware wise.

karelv, 2023-01-05:
Software is up to date. (no update available)
But I'm afraid I cannot further debug anymore on a regular basis.
Thanks for your support.

mkaiser, 2023-01-18:
This project is implemented as a HA addon for sungrow inverters: 
https://github.com/bohdan-s/SunGather

It looks like the project is well advanced, so I will close this issue now. 

I will continue to focus on providing an simple and lightweight YAML integration

closed 2023-01-18 by mkaiser