mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #28: Weird data reported by SH10RT

elektrinis, 2022-12-01:
It's not an issue, just a question if you see this too. I'm noticing that numbers don't add up (with other integration I used before was the same).

First of all, when inverter is active (battery charge/discharge or solar conversion), I get nice numbers that align perfectly, I mean input power = output power. This is impossible however, as there are conversion losses. So I'm certain there is some fakery going on.

Second, I noticed that battery power is wayyyy off of what I get by multiplying battery current * voltage.

Third, I'm seeing this weird "active power" of around -95W, which does not add up with other readings. In my example below, solar power is zero, battery power is zero, and input/output power is equal. So the sum zeroes out, right? However, during the day there is this -95W. I have checked and it almost exactly matches sunrise and sunset time where I live. Weird right, it's almost as it has a schedule and does some extra fakery?
[image]

mkaiser, 2022-12-19:
some interesing findings. 

I guess only the DC-side of the inverter is measured thoroughly. The conversion to the AC-side could be just based on some estimations. 

In some german PV forum I read about the relatively high standby power of around 80-100W. 

We did not have any noticable sunlight the last days. Will check with the battery power, when the sun is back :)

elektrinis, 2022-12-20:
I have updated inverter firmware to latest and things changed a little. Now I am getting input-output power ("total active power") mismatch by ~36W, but the whole 24 hours, not just when sun is above horizon, as before.
Previously it was around 95W, which is not realistic to be used by inverter, as it would run very hot. However in my case the inverter is roughly room temperature, so even the 36W seem bit high.
Will investigate further.

elektrinis, 2023-02-14:
Continuing with this. Just a reminder that I have my house connected to the Backup port on the inverter, and Input port is connected to the smart meter, and that is connected to grid.
On top of this, I have another inverter, connected between input and smart meter, as suggested in sungrow material.

The problem is.... Currently there is snow on my main PV (where sungrow is connected) and it produces no power. And the other inverter is producing power.
However something somewhere interprets this very weirdly.
First issue is that sungrow reports Load power as zero, while definitelly the house is consuming something, and there is no generation on backup port. And even if it counts in the generation on input port, it should be negative.

Second issue is that it reports DC power, when you can clearly see PV/MPPT sitting at zero W. Not sure where this number is coming from, but my other inverter is somehow ending up in this.
[image] [image]

closed 2025-02-24 by github-actions[bot]