mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #29: Unknows device code

Svanmark, 2022-12-04:
I've previously had a working installation of this integration bit from some time ago it can't connect to the inverter (SH10RT). 

I have the latest version of the integration installed. 
Firmware in inverter was updated automatically on the 22nd of November. Can't figure out what version was installed. 

Any help appreciated. Thanks.

mkaiser, 2022-12-19:
mh, a bit clueless here... standard procedure restart all Home-assistant related stuff? After some minutes there should be values. 

At least I can confirm, that there were no larger upgrades / changes in this integration. The yaml files rely on the python modbus module, which has not changed. 

Can you give me a quick feedback, if the problem still exists?

dzarth, 2022-12-19:
Hi, I got the same problem. Could solve it by fixing the slave config from "100" (default in the secrets.yaml) to "1", like it is mentioned in the comment. 
Thanks a lot for your work mkaiser!

mkaiser, 2022-12-19:
hey Dirk, 

thank you for the feedback. I just set the modbus slave address default to 1 in the secrets.yaml. Hopefully that limits confusing in the future :)

@Svanmark: there is a good chance that the new version fixes that for you. Can you double check?

BR
Martin

closed 2023-01-13 by mkaiser