mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #31: Inverter temperature reported as unknown

elektrinis, 2022-12-04:
Unknown temperature.
I was getting temperature with SunGather integration, however it was not correct, +15 degrees of actual. Not sure where's the issue.
[image]

mkaiser, 2022-12-19:
works here. Can you restart, wait and re-check?

elektrinis, 2022-12-20:
I have restarted both HA and inverter, also updated inverter firmware. Still the same :|
How can I help to figure this out?

mkaiser, 2023-01-16:
are you still getting no temperature?`

Which LAN port are you using?

elektrinis, 2023-01-16:
Still Unknown.
I'm using the built-in port, not via dongle.

mkaiser, 2023-01-16:
are all other values working?

I am using the SH10.RT, too. Therefore we should have a very similar setup

elektrinis, 2023-01-16:
Yes everything else seems fine, only temperature is missing.
How can I help?

mkaiser, 2023-07-07:
Any updates with these temperature-issues? Maybe Sungrow updated some firmware?

closed 2023-07-19 by mkaiser