mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #33: incorrect battery voltage, stuck SOC

elektrinis, 2022-12-04:
I was playing with inverter modes, tried force charge and force discharge. I see there are two issues. 
1. SOC is stuck at 100%, even after several hours of discharge. Weirdly, physical battery SOC bar shows 100%. Is this bug in firmware of battery/inverter? Can anyone test and report back? I see this in HA, but also in iSolarClour app?!
[image]

Incorrect battery voltage. Again not sure if this comes from battery/inverter itself. Basically, the batery is of LFP chemistry, which means cell voltage at 0% SOC is 2.0V, and 3.65V at 100% SOC. I see 204V at 100% SOC, so it should be around 111V at 0%. But I see 187V. This is wayy off. I see the same thing in the app. 
[image]

So it's clearly not an issue with this integration, but what is happening? Can anyone check on their setup?

mkaiser, 2022-12-19:
I have a Pylontech Force H1 Battery with 14,4kWh (4 modules)
[datasheet](https://www.klimaworld.com/mwdownloads/download/link/id/45894/)

The datasheet states, that each module has a default voltage of 48VDC. 4* 48V = 192 V. 
In my case this matches the sensor values pretty good (0%: 192V to 100%: 202V). min SoC is at 10% and max Soc at 95%

[image]

The battery power (voltage * current) looks okay for me, too:

[image]

elektrinis, 2022-12-20:
Hm.. Sungrow battery uses LiFePO4 cells with voltage range of 2.5-3.65V, single box is 20s and in my case 3 modules were used, so 60s. In this case the full range of voltage should be 150-219V, which is actually as declared in the datasheet of SBR096.
So it looks like there are some extra issues with inverter or battery/BMS, as not full range is being used. I am also noticing that actual capacity is lower than declared by 20%, even though min/max SOC is set to full range and reserve is set to 0%.

I have also observed a sudden drop of SOC from 100% to 8.8% (after sitting fully charged for a week).

I understand this is an issue with inverter/battery itself (not the integration), just wanted to let you know, and maybe others. I will keep my eye on it.

[image]

elektrinis, 2023-02-20:
I'm closing this as it's likely a battery issue.

closed 2023-02-20 by elektrinis