mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #35: Incorrect value ranges

elektrinis, 2022-12-18:
Correct value ranges for SH10RT are:
Max SOC: 50-100%
Min SOC: 0-50%
Battery SOC reserve: 0-10% (won't accept more than 10%)
Max charge/discharge power: 10.6 kW
Max forced charge/discharge is probably 10.6 kW as well.
(I have a 22 kWh battery)

[image]
[image]

mkaiser, 2022-12-19:
these values (min / max SoC, max charge dis/charge power are currently hard coded in the yaml file. 

For me that was a good compromise preventing myself to accidentaly set too aggressive values ( For the sake of the batteries life I don't want to go below 5% or higher than 95% SoC. Also limiting the maximum charge / discharge power has a benefit on the components' lifetime. 

I can't say anything about the SoC reserve. I still need to do some rewiring to route the inverters backup power to some lights and outlets. So it is still untested (maybe the first blackout in germany will change priorities here ;) )

For a wider range of use it would be good to have some personalized values here. 

We could outsource these values to the secrets.yaml, but It will get more difficult for non-experts to get this integration running ....

I kind of dislike this idea, because most of the current users will just copy and paste upgraded .yaml files from the git and end up with several warnings, until they update their secrets.yaml

```
set_sg_min_soc:
  name: Set Min SoC
  min: !secret sungrow_modbus_min_soc_min 
  max: !secret sungrow_modbus_min_soc_max 
  step: 1
```

``` 
sungrow_modbus_min_soc_min 5
sungrow_modbus_min_soc_max 95
``` 

What do you think? Are there any good ways to solve this?

elektrinis, 2022-12-20:
I think the inverter/battery will simply not accept a number that is not safe or out of range, same as with "reserved SOC" that I observed now: if I set 10%, it will accept this value. But if I set 11%, nothing happens.
I mean it will not start charging the battery ar higher power than it is designed, as the battery/BMS actually informs the inverter of its capabilities. So I believe the inverter will not accept 10.6kW setting if it sees the battery does not support it.
For example, I had a 9.6kW sungrow battery, which is capable of max 30A at 200V, so 6kW, and the inverter was aware of it. Then I have added some more modules and now it is capable of 30A at 470V, or max 10.6kW as limited by inverter itself.

That said I think that it is safe/OK to offer the full range and then the user can decide and test what works.

In my case I needed to set min/max SOC to 0%/100% to properly balance the battery before adding extra modules (as their SOC need to be very close).

On another note about connection of the inverter: in my current setup, I have the whole house connected on "backup" port like this: Grid -> inverter -> house. In this way the whole house would remain powered from battery in case of blackout. I chose sungrow inverter due to exactly this capability.

closed 2025-02-15 by github-actions[bot]