mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #37: Battery Charge / Discharge

Louis712, 2022-12-19:
Would it be possible to have separate entitys for charging and discharging of the battery? Or one entity that's negative when the battery is discharging and positive when the battery is charging.
I don't really know much about templating in Home Assistant, but figured it could be done using the Binary Sensors relying on Running State.

Louis712, 2022-12-19:
Already got it.

mkaiser, 2022-12-19:
hi Louis, 

good timing, I was just about to code something. Can you share and spare us some work? 

Martin

Louis712, 2022-12-19:
This template uses the binary sensors uses the binary sensors for Battery Charging and Battery Discharging and the sensor Battery Power. 
`    - name: "Battery Charging and Discharging Power" # Positive if charging and negative if discharging
      unit_of_measurement: W
      device_class: power
      state: >
        {% if is_state('binary_sensor.battery_charging', 'on') %}
          {{ (states('sensor.battery_power') | float * 1 | float) |int }}
        {% elif is_state('binary_sensor.battery_discharging', 'on') %}
          {{ (states('sensor.battery_power') | float * -1 | float) |int }}
        {% else %}
          {{ (states('sensor.battery_power') | float * 1 | float) |int }}
        {% endif %}`

Louis712, 2022-12-19:
This is my first time creating a template in Home Assistant. I hope it's reliable and works with every inverter.

elektrinis, 2022-12-20:
I would love to have a single entity with battery power, either positive (charging) or negative (discharging). Those flags charging/discharging are not very useful in my case.

Louis712, 2022-12-20:
That‘s exactly what my template does. It‘s positive if the battery is charging and negative if it‘s discharging.

elektrinis, 2022-12-20:
Would love to see this in the integration!

Louis712, 2022-12-20:
If you want to, you can already add it to your configuration.

mkaiser, 2022-12-20:
I just re-discovered a post from bjeanes in this integration's home assistant community thread:

https://community.home-assistant.io/t/sungrow-sh10-rt-modbus-integration/361356/29

There is some more "exception handling" involved, when the sensor did not yet delivered some valid data: 

>       inverter_battery_power:
        unique_id: inverter_battery_power
        friendly_name: Inverter battery power
        value_template: >-
          {% set current = states('sensor.inverter_battery_current') %}
          {% set power = states('sensor.inverter_battery_power_raw') %}
          {% if 'unavailable' in [current, power] %}
            unavailable
          {% elif current|float < 0.0 and power|float > 0.0 %}
            {{ -1 * power | float }}
          {% else %}
            {{ power | float }}
          {% endif %}
        unit_of_measurement: W
        device_class: power

one thing still bothers me about the naming. 

When you look at the new sensor "Battery Charging and Discharging Power", it is a bit ambiguous, in which direction the energy flows. It is very subjective if the battery charging is interpreted as "good / positive value" or "bad / negative value" (or the other way round with discharging)

I did not come up with a better name, yet to describe this template sensor for the battery charge, so thinking loud here....

The grid sensor for "export power", is a bit more distingt: If it is positive, you are exporting power, if negative you can deviate that you are importing power.

Maybe rename the sensor based on your suggestion to 
"Battery Charging Power",     # Positive if charging and negative if discharging

and add a second sensor
"Battery Discharging Power"  # Positive if discharging and negative if charging

The "Export Power" would get the mirrored "Import Power" 

What do you think?

Louis712, 2022-12-21:
That‘s a good idea. I‘m gonna do that today. I‘m gonna post the code in this Issue.

Louis712, 2022-12-21:
Done. Hope it works for you.
`    - name: "Battery Charging Power" # Positive if Charging and negative if Discharging
      unit_of_measurement: W
      device_class: power
      state: >
        {% if is_state('binary_sensor.battery_charging', 'on') %}
          {{ (states('sensor.battery_power') | float * 1 | float) |int }}
        {% elif is_state('binary_sensor.battery_discharging', 'on') %}
          {{ (states('sensor.battery_power') | float * -1 | float) |int }}
        {% else %}
          {{ (states('sensor.battery_power') | float * 1 | float) |int }}
        {% endif %}
    
    - name: "Battery Discharging Power" # Positive if Discharging and negative if Charging
      unit_of_measurement: W
      device_class: power
      state: >
        {% if is_state('binary_sensor.battery_charging', 'on') %}
          {{ (states('sensor.battery_power') | float * -1 | float) |int }}
        {% elif is_state('binary_sensor.battery_discharging', 'on') %}
          {{ (states('sensor.battery_power') | float * 1 | float) |int }}
        {% else %}
          {{ (states('sensor.battery_power') | float * 1 | float) |int }}
        {% endif %} 
`

mkaiser, 2022-12-21:
made a testing branch for that, because there is no sun and I am in a rush

https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/tree/testing

Can you test? I also adapted the PV Info dashboard

Louis712, 2022-12-22:
I'm gonna test it today.

Louis712, 2022-12-22:
[image]
[image]
Schreibe ich mal auf Deutsch weiter.
Wie du auf den Bildern siehst, funktioniert es soweit ganz gut. Deine Ergänzung, dass es als "Unavailable" angezeigt wird, wenn etwas nicht passt, finde ich sehr gut. Auch mit dem Power Flow Card Addon funktioniert es super.

mkaiser, 2023-01-03:
the power flow card addon is really nice - thanks for the tip. 

My configuration is

>  views:
>   - theme: Backend-selected
>     title: PV Power
>     icon: mdi:solar-power
>     badges: []
>     cards:
>       - type: custom:power-flow-card
>         kw_decimals: 2
>         entities:
>           grid: sensor.import_power
>           battery: sensor.battery_discharging_power
>           battery_charge: sensor.battery_level
>           solar: sensor.total_dc_power
>

closed 2023-01-03 by mkaiser

Louis712, 2023-01-03:
[image]
I've also added the Tesla Style Power Flow Card, because I think that it looks quite nice. I had to create some new templates. If you want to check it out, I've put them below.
```
- name: "Inverter loss"
      unit_of_measurement: W
      device_class: power
      state: >
        {{ states('sensor.total_dc_power')|int(default=0) - states('sensor.battery_charging_power')|int(default=0) - states('sensor.load_power')|int(default=0) - states('sensor.export_power')|int(default=0) }}
    
    - name: "Real load" # includes house power and inverter loss
      unit_of_measurement: W
      device_class: power
      state: >
        {{ states('sensor.load_power')|int(default=0) + states('sensor.inverter_loss')|int(default=0) }}
        
    - name: "Import Power" # only positive
      unit_of_measurement: W 
      device_class: power 
      state: > 
        {% if ((states('sensor.export_power')  |int(default=0)) < 0 ) %}
            {{ states('sensor.export_power')|int(default=0)|abs }}
        {% else %}
            0
        {% endif %}
        
    - name: "Grid to House" # only positive
      unit_of_measurement: W
      device_class: power
      state: >
        {% if states('sensor.import_power')|float(default=0) > states('sensor.real_load')|float(default=0) %}
            {{ states('sensor.load_power')|float(default=0) }}
        {% else %}
            {{ states('sensor.import_power')|float(default=0) }}
        {% endif %}
        
    - name: "Grid to Battery"
      unit_of_measurement: W
      device_class: power
      state: >
        {% if states('sensor.import_power')|int(default=0) > states('sensor.real_load')|int(default=0) %}
            {{ states('sensor.import_power')|int(default=0) - states('sensor.real_load')|int(default=0) }}
        {% else %}
            0
        {% endif %}
        
    - name: "Battery to House"
      unit_of_measurement: W
      device_class: power
      state: >
        {% if states('sensor.battery_charging_power')|int(default=0) < 0 %}
            {% if states('sensor.battery_charging_power')|int(default=0)|abs > states('sensor.real_load')|int(default=0) %}
                {{ states('sensor.real_load')|int(default=0) }}
            {% else %}
                {{ states('sensor.battery_charging_power')|int(default=0)|abs }}
            {% endif %}
        {% else %}
            0
        {% endif %}
        
    - name: "Battery to Grid"
      unit_of_measurement: W
      device_class: power
      state: >
        {% if states('sensor.battery_charging_power')|int(default=0) < 0 %}
            {% if states('sensor.battery_charging_power')|int(default=0)|abs > states('sensor.real_load')|int(default=0) %}
                {{ states('sensor.battery_charging_power')|int(default=0)|abs - states('sensor.real_load')|int(default=0) }}
            {% else %}
                0
            {% endif %}
        {% else %}
            0
        {% endif %}
        
    - name: "Solar to Grid"
      unit_of_measurement: W
      device_class: power
      state: >
        {% if states('sensor.export_power')|int(default=0) > states('sensor.battery_to_grid')|int(default=0) %}
            {{ states('sensor.export_power')|int(default=0) - states('sensor.battery_to_grid')|int(default=0) }}
        {% else %}
            0
        {% endif %}
        
    - name: "Solar to House"
      unit_of_measurement: W
      device_class: power
      state: >
        {% if states('sensor.total_dc_power')|int(default=0) > 0 and states('sensor.real_load')|int(default=0) > states('sensor.battery_to_house')|int(default=0) + states('sensor.import_power')|int(default=0) %}
            {% if states('sensor.total_dc_power')|int(default=0) > states('sensor.real_load')|int(default=0) - states('sensor.battery_to_house')|int(default=0) - states('sensor.grid_to_house')|int(default=0) %}
                {{ states('sensor.real_load')|int(default=0) - states('sensor.battery_to_house')|int(default=0) - states('sensor.grid_to_house')|int(default=0) }}
            {% else %}
                {{ states('sensor.total_dc_power')|int(default=0) }}
            {% endif %}
        {% else %}
            0
        {% endif %}
        
    - name: "Solar to Battery"
      unit_of_measurement: W
      device_class: power
      state: >
        {% if states('sensor.total_dc_power')|int(default=0) > 0 and states('sensor.battery_charging_power')|int(default=0) > 0 %}
            {% if states('sensor.battery_charging_power')|int(default=0) > states('sensor.grid_to_battery')|int(default=0) %}
                {% if states('sensor.total_dc_power')|int(default=0) - states('sensor.solar_to_house')|int(default=0) > states('sensor.battery_charging_power')|int(default=0) - states('sensor.grid_to_battery')|int(default=0) %}
                    {{ states('sensor.battery_charging_power')|int(default=0) - states('sensor.grid_to_battery')|int(default=0) }}
                {% else %}
                    {{ states('sensor.total_dc_power')|int(default=0) - states('sensor.solar_to_house')|int(default=0) - states('sensor.solar_to_grid')|int(default=0) }}
                {% endif %}
            {% else %}
                0
            {% endif %}
        {% else %}
            0
        {% endif %}
```