mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #41: Not working on SH6.0RT

florian-hild, 2023-01-05:
Version: Home Assistant 2022.12.8

I get no data from my SH6.0RT inverter.
I can see this errors in the home-assistant.log
```
2023-01-05 22:34:47.674 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{{ (states('sensor.mppt1_voltage') | float * states('sensor.mppt1_current') | float) |int }}")
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1710, in forgiving_float_filter
    return float(value)
ValueError: could not convert string to float: 'unknown'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 423, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1942, in _render_with_context
    return template.render(**kwargs)
  File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1713, in forgiving_float_filter
    raise_no_default("float", value)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1411, in raise_no_default
    raise ValueError(
ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.mppt1_voltage') | float * states('sensor.mppt1_current') | float) |int }}' but no default was specified

The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 540, in async_render_to_info
    render_info._result = self.async_render(variables, strict=strict, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 425, in async_render
    raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.mppt1_voltage') | float * states('sensor.mppt1_current') | float) |int }}' but no default was specified
2023-01-05 22:34:47.686 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.mppt1_voltage') | float * states('sensor.mppt1_current') | float) |int }}' but no default was specified') while processing template 'Template("{{ (states('sensor.mppt1_voltage') | float * states('sensor.mppt1_current') | float) |int }}")' for attribute '_attr_native_value' in entity 'sensor.mppt1_power'
2023-01-05 22:34:47.701 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{{ (states('sensor.mppt2_voltage') | float * states('sensor.mppt2_current') | float) |int}}")
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1710, in forgiving_float_filter
    return float(value)
ValueError: could not convert string to float: 'unknown'

During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 423, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1942, in _render_with_context
    return template.render(**kwargs)
  File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1713, in forgiving_float_filter
    raise_no_default("float", value)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1411, in raise_no_default
    raise ValueError(
ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.mppt2_voltage') | float * states('sensor.mppt2_current') | float) |int}}' but no default was specified

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 540, in async_render_to_info
    render_info._result = self.async_render(variables, strict=strict, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 425, in async_render
    raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.mppt2_voltage') | float * states('sensor.mppt2_current') | float) |int}}' but no default was specified
2023-01-05 22:34:47.705 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.mppt2_voltage') | float * states('sensor.mppt2_current') | float) |int}}' but no default was specified') while processing template 'Template("{{ (states('sensor.mppt2_voltage') | float * states('sensor.mppt2_current') | float) |int}}")' for attribute '_attr_native_value' in entity 'sensor.mppt2_power'
2023-01-05 22:34:47.711 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{{ (states('sensor.phase_a_voltage') | float * states('sensor.phase_a_current') | float) |int}}")
[... 41 lines cut ...]
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1942, in _render_with_context
    return template.render(**kwargs)
  File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1713, in forgiving_float_filter
    raise_no_default("float", value)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1411, in raise_no_default
    raise ValueError(
ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.phase_b_voltage') | float * states('sensor.phase_b_current') | float) |int}}' but no default was specified

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 540, in async_render_to_info
    render_info._result = self.async_render(variables, strict=strict, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 425, in async_render
    raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.phase_b_voltage') | float * states('sensor.phase_b_current') | float) |int}}' but no default was specified
2023-01-05 22:34:47.725 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.phase_b_voltage') | float * states('sensor.phase_b_current') | float) |int}}' but no default was specified') while processing template 'Template("{{ (states('sensor.phase_b_voltage') | float * states('sensor.phase_b_current') | float) |int}}")' for attribute '_attr_native_value' in entity 'sensor.phase_b_power'
2023-01-05 22:34:47.729 ERROR (MainThread)

```

lassej01, 2023-01-13:
Have you checked so you get values from your MQTT integration? If you dont get any values there i can fully understand why home assistant say as it does in the log.

mkaiser, 2023-01-13:
hi, 

apparently, the SH6.0RT lacks the sensors for mppt1/2_voltage and phase_b/b/c_voltage.

I added some more error handling in the corresponding template sensor section. Can you test the fix in [a73c426] ?

florian-hild, 2023-01-14:
@lassej01 do I need the MQTT integration? I just followed the installation and configurations steps at the README.md of this repo.

Do I have to do more than just copy the integration file, include them in the configuration.yaml and add the three variables to the secret.yaml?

florian-hild, 2023-01-14:
@mkaiser I tried the last version  [[7b71f97](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/commit/7b71f9741f13aff5123e7009299a88beecf2d17b)] but got the same error:
```
2023-01-14 16:51:41.362 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{{ (states('sensor.export_power') | float * -1 | float) |int }
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1710, in forgiving_float_filter
    return float(value)
ValueError: could not convert string to float: 'unavailable'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 423, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1950, in _render_with_context
    return template.render(**kwargs)
  File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1713, in forgiving_float_filter
    raise_no_default("float", value)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1411, in raise_no_default
    raise ValueError(
ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ (states('sensor.export_power') | float * -1 | float) |int }}' but no default was spe

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 540, in async_render_to_info
    render_info._result = self.async_render(variables, strict=strict, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 425, in async_render
    raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ (states('sensor.export_power') | float * -1
2023-01-14 16:51:41.374 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering
2023-01-14 16:51:41.436 ERROR (MainThread) [homeassistant.components.netatmo] Error during webhook registration - 400 - Bad request - Allowed ports for webhooks are 80, 88, 443 a
2023-01-14 16:51:50.159 ERROR (SyncWorker_3) [pymodbus.factory] index out of range
2023-01-14 16:51:50.194 ERROR (SyncWorker_3) [pymodbus.transaction] Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2023-01-14 16:51:50.200 ERROR (SyncWorker_3) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Modbus Error: [Input/Output] Unable to decode request
2023-01-14 16:51:50.214 ERROR (SyncWorker_3) [pymodbus.factory] index out of range
2023-01-14 16:51:50.218 ERROR (SyncWorker_3) [pymodbus.transaction] Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2023-01-14 16:51:50.234 ERROR (SyncWorker_1) [pymodbus.factory] index out of range
2023-01-14 16:51:50.238 ERROR (SyncWorker_1) [pymodbus.transaction] Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2023-01-14 16:51:50.253 ERROR (SyncWorker_2) [pymodbus.factory] index out of range
2023-01-14 16:51:50.257 ERROR (SyncWorker_2) [pymodbus.transaction] Modbus Error: [Input/Output] Unable to decode request
[... cut ...]
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2023-01-14 16:51:50.285 ERROR (SyncWorker_5) [pymodbus.factory] index out of range
2023-01-14 16:51:50.287 ERROR (SyncWorker_5) [pymodbus.transaction] Modbus Error: [Input/Output] Unable to decode request
...
```

mkaiser, 2023-01-14:
> Have you checked so you get values from your MQTT integration? If you dont get any values there i can fully understand why home assistant say as it does in the log.

hmm I am afraif this integration has nothing to do with MQTT. 

[florian-hild](https://github.com/florian-hild), can you check the latest commit?  It should fix the side effects : 

```
ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ (states('sensor.export_power') | float * -1 | float) |int }}' but no default was spe
The above exception was the direct cause of the following exception:
```

of this problem:

```
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ (states('sensor.export_power') | float * -1
2023-01-14 17:16:50.257 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering
2023-01-14 17:17:00.121 ERROR (SyncWorker_4) [pymodbus.factory] index out of range
2023-01-14 17:17:00.130 ERROR (SyncWorker_4) [pymodbus.transaction] Modbus Error: [Input/Output] Unable to decode request
```

... but I still don't know thy the "index it out of range"

In the modbus register description from sungrow, there are no hints that the "import power" register should not be available for the SH6RT...

> Do I have to do more than just copy the integration file, include them in the configuration.yaml and add the three variables to the secret.yaml?

these three steps should be enough

If the current fix is not working: Can you check if the modbus tcp connection is kind of working? See #4 for some links. The guys there apparently did not succeed with the SH6.RS (not RT)

florian-hild, 2023-01-14:
Unfortunately, the same error with commit 97cfb1d.

Am I doing something wrong? What can I try?
Do I have to change maybe some settings at the inverter interface?

florian-hild, 2023-01-14:
I enabled now debug logging for modbus and got these messages:
The 172.20.0.2 is the ip of my hass Docker instance, is that right? Or should it be the ip of my inverter!

```python
2023-01-14 22:17:55.593 DEBUG (SyncWorker_3) [pymodbus.client.sync] Connection to Modbus server established. Socket ('172.20.0.2', 59639)
2023-01-14 22:17:55.594 INFO (SyncWorker_3) [homeassistant.components.modbus.modbus] modbus SungrowSHx communication open
2023-01-14 22:18:05.838 DEBUG (SyncWorker_6) [pymodbus.client.sync] New Transaction state 'SENDING'
2023-01-14 22:18:05.842 ERROR (SyncWorker_6) [pymodbus.factory] index out of range
2023-01-14 22:18:05.850 ERROR (SyncWorker_6) [pymodbus.transaction] Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2023-01-14 22:18:05.854 ERROR (SyncWorker_6) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Modbus Error: [Input/Output] Unable to decode request
```

mkaiser, 2023-01-14:
Hmm I suspect connection issues here.

Inverter <> docker <> Hass does apparently not directly work for you.

Please check the general modbus connectivity of the inverter using  a computer in the same network as the inverter. (See links above)

Next step is to learn more about docker to other devices (I am not very familiar with that)

florian-hild, 2023-01-14:
I got now more details from the logs:
As I can see, the connection should work and I get data.
```python
2023-01-14 23:06:30.785 DEBUG (SyncWorker_1) [pymodbus.transaction] Current transaction state - IDLE
2023-01-14 23:06:30.785 DEBUG (SyncWorker_1) [pymodbus.transaction] Running transaction 1
2023-01-14 23:06:30.785 DEBUG (SyncWorker_1) [pymodbus.transaction] SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x8f 0x0 0x1
2023-01-14 23:06:30.786 DEBUG (SyncWorker_1) [pymodbus.client.sync] New Transaction state 'SENDING'
2023-01-14 23:06:30.792 DEBUG (SyncWorker_1) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2023-01-14 23:06:30.795 DEBUG (SyncWorker_1) [pymodbus.transaction] Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2023-01-14 23:06:30.796 DEBUG (SyncWorker_1) [pymodbus.transaction] RECV: 0x0 0x1 0x0 0x0 0x0 0x2 0x1 0x84 0x2
2023-01-14 23:06:30.797 DEBUG (SyncWorker_1) [pymodbus.framer.socket_framer] Processing: 0x0 0x1 0x0 0x0 0x0 0x2 0x1 0x84 0x2
2023-01-14 23:06:30.797 DEBUG (SyncWorker_1) [pymodbus.factory] Factory Response[132]
2023-01-14 23:06:30.798 ERROR (SyncWorker_1) [pymodbus.factory] index out of range
2023-01-14 23:06:30.838 ERROR (SyncWorker_1) [pymodbus.transaction] Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2023-01-14 23:06:30.843 ERROR (SyncWorker_1) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Modbus Error: [Input/Output] Unable to decode request
2023-01-14 23:06:30.849 DEBUG (SyncWorker_2) [pymodbus.client.sync] Connection to Modbus server established. Socket ('172.20.0.2', 40849)
2023-01-14 23:06:30.850 DEBUG (SyncWorker_2) [pymodbus.transaction] Current transaction state - TRANSACTION_COMPLETE
2023-01-14 23:06:30.850 DEBUG (SyncWorker_2) [pymodbus.transaction] Running transaction 2
```

lassej01, 2023-01-15:
> hmm I am afraif this integration has nothing to do with MQTT.
>
> [florian-hild](https://github.com/florian-hild), can you check the latest commit? It should fix the side effects :
>
> ```
> ValueError: [...]

Yes, saw that. I had the ModbusTCP2MQTT integration installed aswell so im used to see modbus things in the mqtt log. But as you write. This has nothing tod do with your integration. Sorry for misleading you Florian-hild.

florian-hild, 2023-01-15:
Thank you guys for your great support!

After debugging, testing with the pymodbus.console and reading the manual, I found the problem.
The SH6.0RT got two Ethernet sockets and modbus only is working at the build in socket at the inverter. Not at the socket at the Wi-Fi connector.

So that was not a problem of that integration.
The integration and dashboards are working now.

Thank you for your help. I will close this issue.

Maybe you can mention this problem on the README.md, so that other people don't fall for the problem.

closed 2023-01-15 by florian-hild