mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #42: Making sense of data - SH10RT

elektrinis, 2023-01-13:
The inverter spits out a lot of data and some of it is difficult to understand. I hope we can collaborate on this to figure out what's what. Example below.

Q#1
Phase A/B/C current/voltage/power are zeros when there is no sun, and the sum of these power are equal to "Total DC power" (when battery is not charging), so I'm guessing this is an internal thing, the output of DC/AC inverter, and is not an actual measurement done on any of external ports?

Q#2
`Load power`, `meter active power` and `Export/import power` don't match. Do we know what these numbers represent?
`Meter power`, I'm guessing comes from one of registers of inverter, and inverter is getting this data from meter itself? Can you confirm?
`Export/import power`. Where does this come from?
`Load power` - ?

Q#3
`Backup power` - ? In my case I have my whole house connected to Backup port of the inverter and I expect this to be equal to `Load power`, and for sure there should be no negative values in the phases, right? Unless is some kind of internal thing again.

[image]

elektrinis, 2023-01-13:
When battery is charging, the sum of phase powers does not match battery power-DC power. I'm bit lost what's what.
[image]

Louis712, 2023-01-23:
Q#1
The Phase A/B/C outputs should are calculated using the dc power I think. I believe it was in photovoltaikforum where a user figured out that only the DC side of the inverter gets measured correctly The AC side is either calculated or measured using the smart meter. 
Q#2 
Load power comes from the meter and shows the current power consumption of the house. No matter if we're using grid power, PV Power or Battery Power.
I don't know what meter active power means, but I think it is how much power is currently flowing through the meter. So, when you're exporting 5kW the meter active power should also be 5kW. When you're importing 5kW, it should also be 5kW. If I understand correctly, you're importing energy in the figure above and are charging your battery with it? 
Export Power is a register from the inverter. Import Power is calculated using Export Power and the binary sensors. It's basically just what you prefer.
Q#3
I don't understand Backup Power. We switched to Backup Power and our whole house was running on Backup. It didn't show anything on the Backup Power registers. 
It is possible that you get negative numbers for Phase A/B/C power. The data type for Phase A/B/C voltage is uint16, so no negative numbers. The data type for Phase A/B/C current is int16, so it is possible that you get negative numbers, as Phase A/B/C power is calculated using the corresponding registers.

As I explained in Q#2, Meter Active Power is the power that is flowing through the meter. In certain situations it should be equal to the things you calculated, but now always. You also have to factor in the delay. You don't know if the data was captured at the same time.

dylan09, 2023-02-16:
I have also tried to figure out how this numbers are related to each other. Because lack of time I have given up for now.

My observations: Except some big devices (Dishwasher, Washing machine, dryer, ofen, ...) most parts of the house running on backup line. Total Power and Backup power never relates to each other.
So I think load power couldn't come from the meter. Because my backup didn't go through any meter.

The meter phase data is sometimes negativ, when there is no export?

If there is some spare time, I will dig a little bit deeper. And I will see, what the numbers will be, when I am on backup (off-grid).

elektrinis, 2023-02-17:
I don't think that `Load power` and `Backup power` are somehow related to the meter. These look completely separated and they update asyncronically as well.
I am looking at my numbers and I just can't work it out.

Some of meter phases can get negative when inverter is trying to hold the sum at zero in self-consumption mode (when operating from battery. For example, If I have a single 100W appliance on one phase, inverter will export 100W on two remaining phases (50W each).

closed 2025-02-12 by github-actions[bot]