mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #43: SH10RT Master & Slave

Matze5593, 2023-01-14:
Gibt es die Möglichkeit 2 Wechselrichter einzubinden wenn man die Konstellation im Master und Slave hat?

elektrinis, 2023-01-14:
Good question. I will be installing a second one soon.

Can we keep this in English please.

mkaiser, 2023-01-14:
hey, 

that should work. See https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/16

Recently, one guy in the home assistant forum confirmed that he got it working:
https://community.home-assistant.io/t/sungrow-sh10-rt-modbus-integration/361356/35

Matze5593, 2023-01-16:
Unfortunately, the method described in #16 does not work. i tried to avoid duplicate names by using the designation 2 or slave. but dying didn't help.

`# Home Assistant Sungrow inverter integration
# Master
modbus:
  - name: SungrowSHx
    type: tcp
    host: !secret sungrow_modbus_host_ip
    port: !secret sungrow_modbus_port
    retry_on_empty: true
    # retries: 10
    # close_comm_on_error: true
    delay: 5
    #timeout: 5
    sensors:
      - name: Sungrow Device type code
        unique_id: sg_dev_code
        slave: !secret sungrow_modbus_slave
        address: 4999 # 5000
        input_type: input
        count: 1
        data_type: uint16
        swap: word
        scan_interval: 600

      - name: Daily PV generation & battery discharge
        unique_id: sg_daily_pv_gen_battery_discharge
        slave: !secret sungrow_modbus_slave
        address: 5002 # 5003
        input_type: input
        count: 1
        data_type: uint16
        swap: word
        precision: 1
        unit_of_measurement: kWh
        device_class: energy
        state_class: total_increasing
        scale: 0.1
        scan_interval: 600

      - name: Total PV generation & battery discharge
        unique_id: sg_total_pv_gen_battery_discharge
        slave: !secret sungrow_modbus_slave
        address: 5003
        input_type: input
        count: 2
        data_type: uint32
        swap: word
        unit_of_measurement: kWh
        precision: 1
        device_class: energy
        state_class: total_increasing
        scale: 0.1
        scan_interval: 600

      - name: Inverter temperature
        slave: !secret sungrow_modbus_slave
        address: 5007
        input_type: input
        count: 1
        data_type: int16
        precision: 1
        unit_of_measurement: °C
        device_class: Temperature
        scale: 0.1
        scan_interval: 10

      - name: MPPT1 voltage
        slave: !secret sungrow_modbus_slave
        address: 5010
[... 1363 lines cut ...]
    action:
      - service: modbus.write_register
        data_template:
          address: 33046 # 33047
          slave: !secret sungrow_modbus_slave
          value: "{{ states('input_number.set_sg_battery_max_charge_power') | int /10}}"
          hub: SungrowSHx
    mode: single

  - id: "24321203543522" # random number
    alias: "sungrow inverter update battery max discharge power"
    description: "Sets battery max discharge power"
    trigger:
      - platform: state
        entity_id:
          - input_number.set_sg_battery_max_discharge_power
    condition: []
    action:
      - service: modbus.write_register
        data_template:
          address: 33047 # 33048
          slave: !secret sungrow_modbus_slave
          value: "{{ states('input_number.set_sg_battery_max_discharge_power') | int /10}}"
          hub: SungrowSHx
    mode: single`

leo4417, 2023-01-26:
This works with my two SH10RT! I've completed the 2nd yaml, see below.

[modbus_sungrow2.yaml.txt](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/files/10511782/modbus_sungrow2.yaml.txt)

I have an additional question: when I try to write a modbus register like this, it's not working. Do you have any idea? Actually the "Set battery SoC reserve" doesn't work either. Perhaps the same problem?

```
alias: Battery-SoC 17
description: ""
trigger:
  - platform: numeric_state
    entity_id: sensor.battery_level
    above: 90
condition: []
action:
  - service: modbus.write_register
    data:
      address: 13100
      hub: SungrowSHx
      slave: 2
      value: 17
mode: single
```

mkaiser, 2023-01-27:
great to know that this is working.

I will put this on the "ToDo" list, but final integration and documentation will take some time.

 
If you want to write to modbus register 13100 (reserved soc for backup), you need to decrement the address to 13099

copied from modbus_sungrow.yaml (arount line 1470)

```  - id: "automation_sungrow_inverter_update_reserved_soc_for_backup"
    alias: "sungrow inverter update reserved soc for backup"
    description: "Updates reserved SoC for backup register"
    trigger:
      - platform: state
        entity_id:
          - input_number.set_sg_reserved_soc_for_backup
    condition: []
    action:
      - service: modbus.write_register
        data_template:
          hub: SungrowSHx
          slave: !secret sungrow_modbus_slave
          address: 13099 # reg 13100
          value: "{{ states('input_number.set_sg_reserved_soc_for_backup') | int}}"
    mode: single```

Does this help?

leo4417, 2023-01-27:
Yes, it works! Thanks!!
But for example, if I enter value: "70" the Battery SoC reserve (sensor.battery_soc_reserve) displays 7%, with "90" this changes to 9%, values above 100 have no effect. Where is my mistake?

leo4417, 2023-01-27:
This is known, I haven't seen it..

```
      - name: Battery SoC reserve 2
        unique_id: sg_battery_soc_reserve_2
        slave: !secret sungrow_modbus_slave_inv2
        address: 13099 # reg 13100
        input_type: holding
        count: 1
        data_type: uint16
        swap: word
        unit_of_measurement: "%"
        device_class: battery
        state_class: measurement
        scale: 1 **# NOT 0.1**
        scan_interval: 10
```

leo4417, 2023-01-28:
I tried to adjust the max_charge_power because battery system SunGrow SBR265 supports up to 17.520 W
Unfortunately this doesn't work... Perhaps the register isn't correct?

```
  set_sg_battery_max_charge_power:
    name: Set max battery charge power in W
    #initial: 40
    min: 0
    max: 17520
    step: 100

  set_sg_battery_max_discharge_power:
    name: Set max battery discharge power in W
    #initial: 40
    min: 0
    max: 17520
    step: 100
```

Matze5593, 2023-02-07:
In the secret.yaml, the slave inverter must be set as a Modbus slave from 1 to 2 in order to be able to query data via the sh10rt's Modbus port

bleughb, 2023-05-26:
> This works with my two SH10RT! I've completed the 2nd yaml, see below. [...]

Thankyou.
Your YAML file worked perfectly for my SG5.0RS
I'd spent a couple of hours trying to create a second modbus_sungrow.yaml file to handle two inverters!
Now I can plot my 4 MPPT string currents

bleughb, 2023-05-28:
Quick heads up RE the dual YAML files - reduce increase the  scan_interval time ...the default will flood the Winet-s devices and knock them offline.
i've taken mine to 60s at the moment and will see how it goes

Matze5593, 2023-08-24:
hi, i have problems with the power mppt 1/2. I don't understand how the code works and my mistake lies. I just don't get any values ​​for the power. Could you please help me

`
- name: MPPT1 voltage inv 2
unique_id: sg_mppt1_voltage_inv2
slave: !secret sungrow_modbus_slave_inv2
address: 5010 # reg 5011
input_type: input
count: 1
data_type: uint16
swap: word
precision: 1
unit_of_measurement: V
device_class: Voltage
scale: 0.1
scan_interval: 10

  - name: MPPT1 current inv 2
    unique_id: sg_mppt1_current_inv2
    slave: !secret sungrow_modbus_slave_inv2
    address: 5011 # reg 5012
    input_type: input
    count: 1
    data_type: uint16
    swap: word
    precision: 2
    unit_of_measurement: A
    device_class: Current
    scale: 0.1
    scan_interval: 10

  - name: MPPT2 voltage inv 2
    unique_id: sg_mppt2_voltage_inv2
    slave: !secret sungrow_modbus_slave_inv2
    address: 5012 # reg 5013
    input_type: input
    count: 1
    data_type: uint16
    swap: word
    precision: 1
    unit_of_measurement: V
    device_class: Voltage
    scale: 0.1
    scan_interval: 10

  - name: MPPT2 current inv 2
    unique_id: sg_mppt2_current_inv2
    slave: !secret sungrow_modbus_slave_inv2
    address: 5013 # reg 5014
    input_type: input
    count: 1
    data_type: uint16
    swap: word
    precision: 2
    unit_of_measurement: A
    device_class: Current
    scale: 0.1
    scan_interval: 10

  - name: Total DC power inv 2
    unique_id: sg_total_dc_power_inv2
    slave: !secret sungrow_modbus_slave_inv2
    address: 5016 # reg 5017
    input_type: input
    count: 2
    data_type: uint32
    swap: word
    precision: 0
    unit_of_measurement: W
    device_class: power
    state_class: measurement
    scale: 1
    scan_interval: 10
sensor:
name: MPPT1_inv2 power
unique_id: sg_mppt1_power_inv2
unit_of_measurement: W
device_class: power
availability: "{{states('sensor.mppt1_voltage_inv2')|is_number and states('sensor.mppt1_current_inv2')|is_number }}"
state: "{{ (states('sensor.mppt1_voltage_inv2') | float * states('sensor.mppt1_current_inv2') | float) |int }}"

name: MPPT2_inv2 power
unique_id: sg_mppt2_power_inv2
unit_of_measurement: W
device_class: power
availability: "{{states('sensor.mppt2_voltage_inv2')|is_number and states('sensor.mppt2_current_inv2')|is_number }}"
state: "{{ (states('sensor.mppt2_voltage_inv2') | float * states('sensor.mppt2_current_inv2') | float) |int }}"

leo4417, 2023-09-08:
Since the last update i get no data from my second inverter... Does anybody have an idea?

Wolkenstuermer, 2023-09-10:
Last HA Update broke my complete Sungrow INtegration for SH and SG... Think there are some issues between Modbus in HA Core and the Code (there is something with regards to uint/int16 needs swap: none ... but this did not help...

dylan09, 2023-09-10:
Are you using the latest version of modbus.yaml file? There the error with `swap` is fixed.

Since HA 2023.9.x there are some new, weird ModBus issues. See [158](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/158)

There you will find a link to the HA issue regarding the problems with ModBus.

roennieinator, 2023-10-09:
I created a .yaml config file for my second inverter that works with Home Assistant 10.1.
I only included the sensors I really need, so of you like, feel free to complete the file - I was too lazy :)

If you want to use this, please follow the instructions given earlier in this thread. 
Thanks to @bleughb for the "inspiration" (I basically stole your way of doing the config). 

[second_inverter_version_10_1.yml.txt](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/files/12849331/second_inverter_version_10_1.yml.txt)

howi303, 2024-05-01:
Has anyone two SH10RT-V112 as Master/Slave working and both combined shown in a flow chart?

ulfben, 2024-08-10:
While waiting for [the YAML-generator](https://github.com/mkaiser/modbus_sungrow_yaml_generator/tree/main) to be available,  I attempted to automate the process of editing the [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml) to add a second inverter, using regular expressions. This was a quick and partial solution, but it might help someone avoid some manual work.

The process is as follows:

1. Copy the original YAML file.
2. Replace the secrets.
3. Append ` 2` to every name: and `_2`  to every unique_id.

Since I'm on Windows, I tested this only in Notepad++, but it should still be helpful for others. :)

### Step 1: copy the current [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml) and rename it to `modbus_sungrow_2.yaml`

### Step 2: Add the IP, port, and Modbus address of inverter 2 to secrets.yaml:
`sungrow_modbus_host_ip: 192.168.4.230 # master inverter
sungrow_modbus_port: 502 # Modbus port of your inverter. Default is '502'
sungrow_modbus_slave: 1 # device address of your inverter. Default is '1'

sungrow_modbus_host_ip_inv2: 192.168.4.238 # slave inverter
sungrow_modbus_port_inv2: 502
sungrow_modbus_slave_inv2: 2 #device address for the slave`

### Step 3: Open your newly created modbus_sungrow_inv2.yaml and perform a couple of  search-and-replace:

- `sungrow_modbus_host_ip` -> `sungrow_modbus_host_ip_inv2`
- `sungrow_modbus_port` -> `sungrow_modbus_port_inv2`
- `sungrow_modbus_slave` -> `sungrow_modbus_slave_inv2`

### Step 4: Use Regex to find and append _2 to all unique_ids:

**Find What:** `^\s*unique_id:.*[^_2\s]$`
**Replace with:** `$&_2`

[image]
(You can use the "Mark" function to preview which lines will be affected.)

### Step 5: Use Regex to find and append 2 to all names (this will handle comments correctly, but not quotes):

**Find What:** `(^\s*- name:\s*)(\"[^\"]*\"|[^\r\n#]*)(\s*#.*)?$`
**Replace with:** `$1$2 2 $3`

[image]

### Step 6: Use search-and-replace to fix all quoted names with trailing 2s:

**Find what:** `" 2`
**Replace with:** `2 "`

Save the file and check your configuration under Developer Tools. You may see warnings like:

> Configuration warnings
> Setup of package 'modbus_sungrow_2' failed: integration 'input_number' has duplicate key 'name'
> Setup of package 'modbus_sungrow_2' failed: integration 'input_select' has duplicate key 'name'

At this point, I’m not sure how to proceed. Deleting the inputs ([lines 2316-2415](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/b29e852d7a564a127eee0649fd78a0f479fd6ba7/modbus_sungrow.yaml#L2316)) removes the error, and after a reboot, everything seems to work.

@mkaiser - when setting up the 2nd inverter, how should we handle the input, script and the automation sections? I assume that they need to be "personalized" too?

ulfben, 2024-08-11:
I believe these are all the sensors referenced throughout the [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml). To finish off our modbus_sungrow_2.yaml properly, we should append "_2" to all of these terms throughout;

```
binary_sensor.battery_charging
binary_sensor.battery_discharging
binary_sensor.exporting_power
binary_sensor.importing_power
binary_sensor.pv_generating
sensor.battery_capacity
sensor.battery_discharging_start_power
sensor.battery_forced_charge_discharge_cmd
sensor.battery_forced_charge_discharge_cmd_raw
sensor.battery_forced_charge_discharge_power
sensor.battery_level
sensor.battery_level_nominal
sensor.battery_max_charge_power
sensor.battery_power_raw
sensor.daily_battery_charge
sensor.daily_battery_discharge
sensor.daily_exported_energy
sensor.daily_imported_energy
sensor.daily_pv_generation
sensor.ems_mode_selection
sensor.ems_mode_selection_raw
sensor.export_power_limit
sensor.export_power_limit_mode
sensor.export_power_limit_mode_raw
sensor.export_power_raw
sensor.global_mpp_scan_manual_raw 
sensor.load_power
sensor.max_soc
sensor.meter_active_power_raw
sensor.meter_phase_a_active_power_raw
sensor.meter_phase_b_active_power_raw
sensor.meter_phase_c_active_power_raw
sensor.min_soc
sensor.mppt1_current
sensor.mppt1_voltage
sensor.mppt2_current
sensor.mppt2_voltage
sensor.phase_a_voltage
sensor.phase_b_voltage
sensor.phase_c_voltage
sensor.reserved_soc_for_backup
sensor.running_state
sensor.sungrow_device_type_code
sensor.sungrow_inverter_state
sensor.system_state
sensor.total_battery_charge
sensor.total_battery_discharge
sensor.total_dc_power
sensor.total_exported_energy
sensor.total_imported_energy
sensor.total_pv_generation
```

ulfben, 2024-09-11:
Absotively. [I've created a gist for you](https://gist.github.com/ulfben/882142074fc031b07d500e7a95c9e466). Hopefully it's allowed to stay up. :P

mkaiser, 2024-09-11:
> Absotively. [I've created a gist for you](https://gist.github.com/ulfben/882142074fc031b07d500e7a95c9e466). Hopefully it's allowed to stay up. :P

Could you maybe make a github action out of it and auto-create the file whenever the modbus_sungrow.yaml changes?

I really like having the second inverter stuff in the git, but I am afraid of the maintenance (have barely time to maintain the one file...)

ulfben, 2024-09-12:
@mkaiser I am a complete child when it comes to git and github, but I do try to be helpful. hahaha :D 

I forked your repo and tried setting up an action. [Here's what I got so far](https://github.com/ulfben/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/.github/workflows/auto-generate-second-config.yml).

It runs whenever a commit touches modbus_sungrow.yaml, but I've got a syntax error in one my sed-commands that I can't figure out. [Here's the output from my test-run](https://github.com/ulfben/Sungrow-SHx-Inverter-Modbus-Home-Assistant/actions/runs/10828269747/job/30043238549):

```
Run cp modbus_sungrow.yaml modbus_sungrow_2.yaml
  cp modbus_sungrow.yaml modbus_sungrow_2.yaml
  sed -i 's/sungrow_modbus_host_ip/sungrow_modbus_host_ip_inv2/g; \
  s/sungrow_modbus_port/sungrow_modbus_port_inv2/g; \
  s/sungrow_modbus_slave/sungrow_modbus_slave_inv2/g; \
  s/set_sg_/set_sg_2_/g; \
  s/sg_set_/sg_2_set_/g; \
  s/- id: "automation_sungrow_global_/- id: "automation_sungrow_global_2_/g; \
  s/- id: "automation_sungrow_inverter_/- id: "automation_sungrow_inverter_2_/g; \
  s/alias: "sungrow inverter/alias: "sungrow inverter 2/g' modbus_sungrow_2.yaml
  shell: /usr/bin/bash -e {0}
sed: -e expression #1, char 427: unterminated address regex
Error: Process completed with exit code 1.
```
However, when I run this block on my own terminal it works as expected. Can anyone see what I'm missing?

ulfben, 2024-09-12:
> Could you maybe make a github action out of it and auto-create the file whenever the modbus_sungrow.yaml changes

Yes! I've set it up [in a fork over here](https://github.com/ulfben/Sungrow-SHx-Inverter-Modbus-Home-Assistant). I know *nothing* about git so... just do what you need to copy .github/workflows/ back to your repo and you should be off to the races.

**Remaining work** this action does not yet adjust [the sensors for monthly and yearly export](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/b29e852d7a564a127eee0649fd78a0f479fd6ba7/modbus_sungrow.yaml#L2165), as their names are concatenated in the yaml ([currently at lines 2165-2230](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/b29e852d7a564a127eee0649fd78a0f479fd6ba7/modbus_sungrow.yaml#L2165)) and I don't know the yaml (or python?) syntax well enough to touch any of that.

If somebody could come up with a good line of sed to touch those up, that would be great. :)

mkaiser, 2024-09-21:
> I am a complete child when it comes to git and github,

well, you have grown fast :)

Thanks a lot I just merged it (but the file has an "experimental" in its name. I do not have any time to look at this:

> Remaining work: this action does not yet adjust [the sensors for monthly and yearly [...]

... maybe someone will fix it :)

Thanks again!

lefjomp, 2025-01-11:
i also have 2 sh10rt v112 and a sbr25,6kw and a 9,6kw battery in parallel, the charging works great and the batteries are both charging with isolarcloud, but when i start a forced charging with mkaiser i just get the host battery to charge, the slave battery stays empty, any idea why it does that?

ulfben, 2025-01-12:
Your widget to enable force charging is probably only triggering the script for the first inverter. 

Make sure that your automation or widget triggers the respective scripts correctly. You have (for example) 'sg_set_forced_discharge_battery_mode' for the first inverter, and 'sg_2_set_forced_discharge_battery_mode' for the second inverter. The other scripts are mirrored in the same way.

lefjomp, 2025-01-12:
hej!

jag ser att du heter Ulf så du borde vara svensk 🙂

jag har lagt in modbus_sungrow_inverter_experimental.yaml, jag har dock inte gjort några ändringar i den.
i homeassistant har jag till en daschboard till där jag får information från den andra invertern, jag ändra i daschboard koden och lade till _2 bakom varje sensor.
jag kan dock inte starta tvångs laddning/urladdning där, men det går inte heller i isolarcloud att tvångsladda slave invertern.
jag vet inte om det har att göra med att den ligger i just slave läget.

mvh/Conny Andersson

Make sure that your automation or widget triggers the respective scripts correctly. You have (for example) 'sg_set_forced_discharge_battery_mode' for the first inverter, and 'sg_2_set_forced_discharge_battery_mode' for the second inverter (see: modbus_sungrow_second_inverter_experimental.yaml). The other scripts are mirrored in the same way.

lefjomp, 2025-01-13:
i have copied the modbus_sungrow_second_inverter_experimental.yaml ad i have made all the sensors like _2.
i have made another dashboard named sungrow 2, and there i can see the other inverter and most of the sensors work.
but i cant start or stop forced recharging.
but when i start/stop forced charging on the sungrow 1 dashboard i see that sungrow 2 changes state according to sungrow 1,s dashboard, so there is someting wrong there.

closed 2025-05-13 by github-actions[bot]