mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #45: Battery SoC Reserve is send in %

dylan09, 2023-01-23:
The value of reg 13100 `Reserved SOC for backup` is send as x %. You have defined it with `scale: 0.1`. 
Should be `scale: 1`. Maybe a typo or  related to copy & paste.

And the automation for setting the value should be modified. 

Should also fix the value range error reported for this sensor in #35

ViktorReinhold, 2023-01-23:
I can confirm it. Setting `scale: 1` on `Battery SoC Reserve` and changing from `int*10` to `ìnt` on the corresponding automation.

By the way, do somebody know how to enable or disable the backup mode at all? In the iSolarApp settings menu I can Enable or Disable it but don't see a way to do it with HA.

mkaiser, 2023-01-23:
incorporated your changes in [c2fd053](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/commit/c2fd05352331ac73f3556ea5b6cd19756f527aed) 

thanks!

I also would like to know, how to enable the backup stuff (still have not rewired anything to have something useful at the backup ports)

elektrinis, 2023-01-24:
If "backup mode" means a mode when inverter is supplying loads from battery and tries to keep grid at 0W, I also want how to do this.
Also, has anyone figured out whatt all those operatiom modes are? I just noticed that forced charge/dischantge does not work in "self consumption" mode, I have to stitch to "forced" mode for this to work.

mkaiser, 2023-01-24:
> If "backup mode" means a mode when inverter is supplying loads from battery and tries to keep grid at 0W, I also want how to do this. Also, has anyone figured out whatt all those operatiom modes [...]

"backup mode" refers to the case, when there is a blackout (the grid is not connected/delivering power) and the inverter uses the battery to power selected loads, which are connected to the dedicated backup power port of the inverter. 

The EMS mode is the switch to select the general energy management system (EMS). In most cases you want the default self consumption mode (PV power is distributed to loads, excess power is used to charge the battery. When the battery is full (or the maximum charge power is reached) the rest will be exported. 

When you change EMS to forced mode, the other selection mode "forced charge / discharge" is evaluated. 

So to charge your battery you need to adapt both selections (EMS and force charge/discharge)

dylan09, 2023-01-24:
For me I have changed the 'EMS mode' dropdown and automation a little bit. I have removed `External EMS, VPP and MicroGrid` from the drop-down selection and automation.
VPP and Microgrid are not applicable to my installation. And for `External EMS` I haven't enough background? Maybe someone could help and describe the meaning of this setting and what it changes in inverter behaviour.

dylan09, 2023-01-24:
> When you change EMS to forced mode, the other selection mode "forced charge / discharge" is evaluated.

But be careful. If you select `Foreced charge/discharge` and set `force charge power`. If there is not enough PV power, the rest is taken from the grid.
E.g. `force charge power` is 1000W. PV supplies only 600W. Then 400W will be taken from the grid to charge the battery.

dylan09, 2023-01-24:
> I also would like to know, how to enable the backup stuff

First you have to connect your backup loads to the inverter. After that change Common Parameter Settings -> Power Control -> Backup Mode (No. 3) to Enable in iSolarCloud (maybe could also be changed in the App with local access). 

To test if everything works you could switch off the AC circuit breaker or RCD in "front of" the inverter.

closed 2023-01-24 by mkaiser