mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #46: None of the configuration works

Louis712, 2023-01-24:
For me, none of the configuration stuff works. I‘ve also got a SH10RT inverter, but it just won‘t work. I tried to set the device to force charge, but none of the options seem to work. I‘ve tried it in the normal self-consumption mode and also the forced mode. The automation gets triggered, so I don‘t think that‘s the problem. It must be somewhere between the modbus config and the inverter. I‘m also running the code version from the 20th january 2023.

dylan09, 2023-01-24:
You have to do 3 steps for this to get it working:
1. Set the desired charging power `set_sg_forced_charge_discharge_power`
2. Select `Forced charge` from `set_sg_battery_forced_charge_discharge_cmd`
3. Select `Forced mode` from `set_sg_ems_mode`

And then, after a short while, the inverter should begin charging the battery. Could take a little while before it starts.

If this doesn't work try to force charge inside the iSolarCloud app.

Louis712, 2023-01-24:
I did try that, but it just doesn't seem to register correctly. I just want it working inside Home Assistant. When I change the power, the register battery_forced_charge_discharge_power still reports 3900W. Even if I change it to something completely different.

mkaiser, 2023-01-24:
What does the GUI say, if you change the value and wait for one minute or so?

The modbus register behind is not updated very frequently. Unfortunately the automation can not retrigger an immediate modbus register update. So you select a new value and need to wait for a certain amoiunt of time, until the next refresh of the modbus values

Louis712, 2023-01-24:
If I change the forced charge discharge power, it takes a few seconds and then it updates sensor.battery_forced_charge_discharge_power. The same thing applies for the other two registers. I change it and the corresponding sensor also changes. The problem is, that it doesn't do anything. The battery doesn't draw any power from the grid. If I change it in the ISolarCloud app, it works. I just want it working in Home Assistant, so that I can disconnect it from the cloud and only use the WiFi Dongle for Firmware-Updates.

Louis712, 2023-01-24:
Whoops, restarted my inverter and forgot to turn on the breaker to the grid. Just turned it on and will try again in a few minutes.

Louis712, 2023-01-24:
Takes a few minutes to start, but it works now. I don't know what was wrong this morning, but I just updated to the newest code and it somehow works now.

dylan09, 2023-01-24:
I noticed today that battery_forced_charge_discharge_power is also respected when discharging in self-consumption mode. 

Maybe someone could check if that is true. In that case it would be better to rename the sensor.

Louis712, 2023-01-24:
Would love to try it, but I don't exactly unterstand what you mean. Could you elaborate a bit more on what exactly happens?

mkaiser, 2023-01-24:
just checked this and it works for me. I can limit the battery discharging power with "set_sg_battery_max_discharge_power"

Also I could successfully limit the forced charge discharge power

Can you double check? It is easy to mix them up, because they look pretty much alike: 

- input_number.set_sg_battery_max_discharge_power
--> set_sg_battery_**max_discharge**_power   is providing the value for address: 33047 # reg 33048

- input_number.set_sg_battery_max_charge_power
--> set_sg_battery_**max_charge**_power   is providing the value for address: 33046# reg 33047

- input_number.set_sg_forced_charge_discharge_power
-->set_sg_**forced_charge_discharge**_power is providing the value for address: 13051 # reg 13052

Louis712, 2023-01-24:
Yup. Checked it. I can limit the forced charge and discharge power using set_sg_forced_charge_discharge_power. That is very accurate. With set_sg_battery_max_discharge_power it's always a bit under. If I set it to 200W it only discharges with about 112W. I think that's normal, because I've seen it in the Local Access of the WiNet-S Module. If the BMS limits the battery current to 6A it only takes about 5.4A. set_sg_battery_max_charge_power it's pretty accurate. Just 4-6 watts over what I've set. I set 200W max charge power and it takes about 204-206W.

dylan09, 2023-01-24:
Checked registers:
- sg_battery_max_charge_power has value 10.000 W
- sg_battery_max_discharge_power has value 10.000 W
- sg_forced_charge_discharge_power has value 1.000 W

Load on inverter is > 5.000 W. Nearly exactly 1.000 W were taken from the battery and the rest is coming from grid. But I will double check again if my battery is charged again. Actually no sun here in germany.

Blue is battery discharging power, green is power taken from grid. EMS setting is Self-Consumption mode (Default).
[image]

Louis712, 2023-01-24:
For me, if I set the EMS-Mode to Self-Consumption the forced charge discharge doesn't do anything. Only the max charge/discharge do something in self-consumption mode. What I can see is that your battery discharge is limited by the forced charge discharge power? I set mine to zero and I am still using my battery.

mkaiser, 2023-01-25:
> Checked registers:
> 
> * sg_battery_max_charge_power has value 10.000 W
> * sg_battery_max_discharge_power has value 10.000 W
> * sg_forced_charge_discharge_power has value 1.000 W
> 
> Load on inverter is > 5.000 W. Nearly exactly 1.000 W were taken from the battery and the rest is coming from grid. But I will double check again if my battery is charged again. Actually no sun here in germany.
> 
> Blue is battery discharging power, green is power taken from grid. EMS setting is Self-Consumption mode (Default). [image]
> 
> Sorry. Y-Axis is missing in image. But the maximum for blue line is at 1.000 W.

Could you try other values for discharge power? 

In your screenshot it looks like there is 1/10 of the 10.000 W discharge power being used. 

In case you have a inverter model that interprets the discharge power slightly different (factor 10), that could explain that behaviour

Whats your inverter model?

dylan09, 2023-01-25:
Yes. I will try some other values set the charge and discharge values to some other value than the default 10.000 W.
But for the next day the forecast says no sun for the place where I live 😫

mkaiser, 2023-01-25:
No sun here in Bielefeld, germany , too :/

I used the forced charge feature for an hour to have 10% SoC test the charge / discharge power settings.

Louis712, 2023-01-25:
> Maybe it could also be an EMS function. Limit the discharging power due to battery SoC or other conditions. 

I can say, that the BMS of my Sungrow SBR128 limits the current in certain conditions, for example when the battery is almost empty, it's too hot or too cold. 
When it's at about 95% it also limits the charging current.

GrandAmmor, 2023-01-26:
> You have to do 3 steps for this to get it working:
>
> 1. Set the desired charging power `set_sg_forced_charge_discharge_power`
> 2. Select `Forced charge` from [...]

This works perfectly for me (Home Assistant 2023.1.7; inverter integration last update: 2023-01-20; SH10RT-v112). I like your work a lot and I enjoy reading your modbus_sungrow.yaml ;) Thanks for that!

Although I like to change the values of `set_sg_ems_mode` and `set_sg_battery_forced_charge_discharge_cmd` manually in the GUI I would love an automation for that action.

My first usecase for that action will be the 'sensor.electricity_price_home' of my tibber integration (dynamic electricity tarif) as a trigger to charge my battery (if price <x € then charge) . Later then there should be conditions like pv-production forecast because usually the electricity price is cheaper at night than at day. But in the long run I don't want to force charging if the pv-production will be more than my consumption incl. saved electricity.

Can you help me with that? How should I define the action in that automation?

mkaiser, 2023-01-27:
hi [GrandAmmor](https://github.com/GrandAmmor), 

this would be a separate issue here... sorry, but I am loosing the overview of open, closable issues and discussion. 

You can use the automations at the end of the yaml file as a blueprint. If you are not familiar with YAML, you can also use the settings --> automation GUI. 

As Action, you need to select "Call a service", then "Modbus:Write register" and use the values from the yaml.

If you need more specific help, please open a separate issue. There are some friendly guys around here who are glad to help

GrandAmmor, 2023-01-27:
Thx for your fast reply! I'll keep on trying, but have to open a separate issue also, I guess ;)

dylan09, 2023-01-28:
> Load on inverter is > 5.000 W. Nearly exactly 1.000 W were taken from the battery and the rest is coming from grid. But I will double check again if my battery is charged again. Actually no sun here in germany.

Found the reason for this behavior. While I am experimenting with my inverter and iSolarCloud a played with `Active Power Limit`. Have set it to a value of 10%. And it seems, this is also respected when discharging the battery 😅. So has nothing to do with the modus register settings.

closed 2023-02-07 by mkaiser