mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #50: Automation for switching EMS Mode and battery forced charge // maybe tibber as trigger

GrandAmmor, 2023-01-27:
Hey guys, I learned a lot about this great integration of my SH10RT-v112 in HA. Thanks for your constructive discussion : )

Now I'm trying to write a automation which can
switch
the ems mode from sensor.ems_mode_selection_raw = 0 (Self-consumption mode) to 2 (Forced mode)
AND
the battery forced charge/dischrage cmd sensor.battery_forced_charge_discharge_cmd_raw = 0xCC (Stop) to 0xAA (Forced charge)
if
my tibber-integration-sensor sensor.electricity_price_home states a value like 0,30 EUR/kWh.

After one hour the switch should change back to self consumption mode and forced mode stop. Can you help me with the yaml for that "Modbus:write register"-action?

Maybe there will be smarter versions but I think this would be a good start...?!

GrandAmmor, 2023-01-27:
`- id: '1674839435197'
  alias: sungrow inverter_custom EMS Mode Change
  description: ''
  trigger:
    - platform: numeric_state
      entity_id: sensor.electricity_price_home
      attribute: max_price
      value_template: ""
      below: 0.4
  condition: []
  variables:
      ems_mode_self_consume: 0
      ems_mode_forced: 2
      ems_mode_external: 3
      ems_mode_vpp: 4
      ems_mode_microgrid: 8
  action:
  - service: modbus.write_register
    data:
      address: 13049
      slave: !secret sungrow_modbus_slave
      hub: SungrowSHx
      value: >-
        {% if is_state('input_select.set_sg_ems_mode', 'Self-consumption mode (default)') %}
        {{ems_mode_forced}}
        {% else %}
        0
        {% endif %}'
    alias: Change ems_mode
  mode: single`

GrandAmmor, 2023-01-27:
- id: '1674839435197'
  alias: sungrow inverter_custom EMS Mode Change
  description: ''
  trigger:
    - platform: numeric_state
      entity_id: sensor.electricity_price_home
      attribute: max_price
      value_template: ""
      below: 0.4
  condition: []
  variables:
      ems_mode_self_consume: 0
      ems_mode_forced: 2
      ems_mode_external: 3
      ems_mode_vpp: 4
      ems_mode_microgrid: 8
  action:
  - service: modbus.write_register
    data:
      address: 13049
      slave: !secret sungrow_modbus_slave
      hub: SungrowSHx
      value: >-
        {% if is_state('input_select.set_sg_ems_mode', 'Self-consumption mode (default)') %}
        {{ems_mode_forced}}
        {% else %}
        0
        {% endif %}'
    alias: Change ems_mode
  mode: single

GrandAmmor, 2023-01-27:
What I#ve done so far looks like:

`- id: '1674839435197'
  alias: sungrow inverter_custom EMS Mode Change
  description: ''
  trigger:
    - platform: numeric_state
      entity_id: sensor.electricity_price_home
      attribute: max_price
      value_template: ""
      below: 0.4
  condition: []
  variables:
      ems_mode_self_consume: 0
      ems_mode_forced: 2
      ems_mode_external: 3
      ems_mode_vpp: 4
      ems_mode_microgrid: 8
  action:
  - service: modbus.write_register
    data:
      address: 13049
      slave: !secret sungrow_modbus_slave
      hub: SungrowSHx
      value: >-
        {% if is_state('input_select.set_sg_ems_mode', 'Self-consumption mode (default)') %}
        {{ems_mode_forced}}
        {% else %}
        0
        {% endif %}'
    alias: Change ems_mode
  mode: single`

mkaiser, 2023-01-27:
This is a quiet good start . There was not much for me to add :)

in the visual editor the aliases are very helpful. Please check thoroughly before testing 

``` alias: Testing EMS mode Change
description: ""
trigger:
  - platform: numeric_state
    entity_id: sensor.electricity_price_home
    attribute: max_price
    below: 0.4
condition: []
action:
  - service: modbus.write_register
    data:
      address: 13049
      slave: !secret sungrow_modbus_slave
      value: 2
      hub: SungrowSHx
    alias: Set EMS mode to "Forced mode"
  - service: modbus.write_register
    data:
      address: 13050
      slave: !secret sungrow_modbus_slave
      value: 170
      hub: SungrowSHx
    alias: Set forced mode to "Forced Charge"
  - delay:
      hours: 1
      minutes: 0
      seconds: 0
      milliseconds: 0
  - service: modbus.write_register
    data:
      address: 13049
      slave: !secret sungrow_modbus_slave
      value: 0
      hub: SungrowSHx
    alias: Set EMS mode back to "Self-consumption mode"
mode: single
 ```

mkaiser, 2023-01-27:
Unfortunately, if you switch to visual mode all the !secret s and nice hexadecimal values are getting lost :/

GrandAmmor, 2023-01-28:
Wow, thanks a lot again! I configured the automation in my automation.yaml and try to avoid the visual mode in this case. I had to change the " to ' but since that it works like a charm : )
I should have tested it at the time of your reply, because the price was much better at night ;)

Does it matter if there ist one more service: modbus.write_register after the delay like
`  - service: modbus.write_register
    data:
      address: 13050
      slave: !secret sungrow_modbus_slave
      value: 240 #0xCC
      hub: SungrowSHx
    alias: Set forced mode to 'Stop (default)'` ?

My next steps will be to figure out how the trigger can be smarter and pays attention to the tibber-price forecast to choose the cheapest spot to start the "power-hour". Including the pv-forecast would be lovely.

I checked out EMHASS for that, but I failed to configure the tibber-part with that...

If you are interested in tibber or need somebody to test with it, just give me a call. Wiesbaden isn't that far away : )

mkaiser, 2023-01-28:
> Does it matter if there ist one more service: modbus.write_register after the delay like

I would recommend that to be double safe that you do not accidently force charge/discharge the battery, when EMS mode reset to self-consumption mode did not work (theoretically)

I had a quick peek into Tibber. Where can I see the momentarily prices even if I am not registered?

dylan09, 2023-01-28:
I'm not that experienced in HA automations. Mostly use Node-Red.
I would try to move the query `{% if is_state('input_select.set_sg_ems_mode', 'Self-consumption mode (default)') %}` to the `condition:` section.

GrandAmmor, 2023-01-28:
That makes sence...especially if you want to change the mode: single to mode: queued or for more than just one automation. In every case you have to check the state berfore the action, I guess.

GrandAmmor, 2023-01-30:
> I had a quick peek into Tibber. Where can I see the momentarily prices even if I am not registered?

This https://github.com/custom-components/nordpool could be an easy way to display the spot market price instead of the tibber API. Have you seen this before?

GrandAmmor, 2023-02-09:
Your EMS mode selection should be fine wirth 'Self-consumption mode (default)' to feed the load out of the battery. Changing the  Battery forced charge discharge cmd to Stop (default) also is recommeded by mkaiser "to be double save that you do not accidently force charge/discharge the battery".

elektrinis, 2023-02-11:
I just figured out my battery is failing, error 833. Raised separate issue #60

mkaiser, 2023-02-13:
> > I had a quick peek into Tibber. Where can I see the momentarily prices even if I am not registered?
> 
> This https://github.com/custom-components/nordpool could be an easy way to display the spot market price instead of the tibber API. Have you seen this before?

thank you for the links. I will consider switching to some of these new "energy players", as soon as I can remotely control my heat pump :)

Will close the issue now. Documentation to be done is marked in #38 (will be happy if someone would write some sentences about that, so i can just merge a PR :)

closed 2023-02-13 by mkaiser

GrandAmmor, 2023-02-13:
At the moment I'm on a seperat tariff with my heatpump because the prices for that are 'ok'. But I'll step back to just one electricity meter for the total load when I'm experienced enough in HA :D

I'm not sure if this is the right place to post, but here is my first 'semi'-atomation:

`- alias: sungrow AC-Laden EMS Mode Change  
  description: ''
  trigger:
    - platform: numeric_state
      entity_id: input_number.einkaufspreis
      #entity_id: sensor.electricity_price_home
      #attribute: today
      below: sensor.electricity_price_home
  condition: []
  action:
    - service: modbus.write_register
      data:
        address: 13049
        slave: !secret sungrow_modbus_slave
        value: 2
        hub: SungrowSHx
      alias: Set EMS mode to 'Forced mode'
    - service: modbus.write_register
      data:
        address: 13050
        slave: !secret sungrow_modbus_slave
        value: 170
        hub: SungrowSHx
      alias: Set forced mode to 'Forced Charge'
    - delay:
        hours: 1
        minutes: 0
        seconds: 0
        milliseconds: 0
    - service: modbus.write_register
      data:
        address: 13050
        slave: !secret sungrow_modbus_slave
        value: 0xCC
        hub: SungrowSHx
      alias: Set forced mode back to 'Stop (default)'
    - service: modbus.write_register
      data:
        address: 13049
        slave: !secret sungrow_modbus_slave
        value: 0
        hub: SungrowSHx
      alias: Set EMS mode back to 'Self-consumption mode (default)'
    - delay:
        hours: 7
        minutes: 0
        seconds: 0
        milliseconds: 
  mode: queued`

I need a little helper for that, which looks like that:
[image]

At the moment it feels more comfortable for me to "push the trigger" manually, but you can use the sensor.electricity_price_home and it's attributes instead.

morph027, 2023-08-29:
I'm struggeling to get this working on my installation. The commands are being sent successfully and the sensor readings do change. Battery power goes to zero and all power is drawn from grid, but the battery does not charge. Any hints?

GrandAmmor, 2023-08-29:
Hey,
I would start with a quick check if
- the inverter is running
- EMS mode = forced mode
- battery forced charge discharge = forced charge
- forced charge discharge power is set to > 0
- max battery charge power is set to > 0

morph027, 2023-08-29:
Ah, thanks, (4) forced charge discharge power was missing, set that to 5000 for my SH80RT and it immediately swallows power from grid ;) Thank you. Now for some clever automation based on tibber price and pv forecast, but that's not n scope of this project.

GrandAmmor, 2023-08-29:
Glad I could help :)

At the beginning I also tried to build a füll Automation. Right know I'm fine with my helper based Semi-Automation, because there are to many changing conditions day by day (IMO).

[image]

'Einkaufsdauer' (duration of the purchase) und 'Einkaufspreis' (purchase price) are to be set manually and are part of my automation:

```
- alias: sungrow AC-Laden EMS Mode Change  
  description: ''
  trigger:
    - platform: numeric_state
      entity_id: sensor.electricity_price_home
      below: input_number.einkaufspreis
  condition: []
  action:
    - service: modbus.write_register
      data:
        address: 13049
        slave: !secret sungrow_modbus_slave
        value: 2
        hub: SungrowSHx
      alias: Set EMS mode to 'Forced mode'
    - service: modbus.write_register
      data:
        address: 13050
        slave: !secret sungrow_modbus_slave
        value: 170
        hub: SungrowSHx
      alias: Set forced mode to 'Forced Charge'
    - delay:
        minutes: '{{ states(''input_number.einkaufsdauer'') | int  }}'
    - service: modbus.write_register
      data:
        address: 13050
        slave: !secret sungrow_modbus_slave
        value: 0xCC
        hub: SungrowSHx
      alias: Set forced mode back to 'Stop (default)'
    - service: modbus.write_register
      data:
        address: 13049
        slave: !secret sungrow_modbus_slave
        value: 0
        hub: SungrowSHx
      alias: Set EMS mode back to 'Self-consumption mode (default)'
    - delay:
        hours: 0
        minutes: 5
        seconds: 0
        milliseconds: 0
  mode: queued```

```
```

jayjay982, 2024-03-17:
Hi all, I came across this topic as I was searching for an automation to load my battery from the grid. I have a Sungrow inverter and battery. And I also use the mkaiser integration in HA. Since last week I have Tibber as my energy provider. Unfortunately I am not good in YAML. I tried rebuilding the automation with the visual editor but I am not able to get the same results.

Can someone help me out here? I need a helper which is pulling the current price from tibber and that helper than triggers the automation (setting EMS mode, etc.), correct?

Thanks, JJ.

jove01, 2024-07-28:
Hey
I am starting with this theme.  And now I found this discussion.

My solution-sequence is like yours, but between all modbus-steps I have a delay of 15 sec. Is this necassery or not?

Than another question. You set the "duration of the purchase".  I thaught it ist meaningful to have also  a contition for max SoC for the battery an stop than. What is your experience with the time?

Knows somebody a solution to stop the delay  to reset the Sungrow?

martinknechtel, 2024-12-21:
@GrandAmmor Nice idea controlling this with purchase price and purchase duration. I recently came up with this approach in case you want to have it a bit more automated: I charge in the cheapest hours, consume from battery when it's expensive, e.g. +25% above purchase price. In between, it would not pay off to consume battery since there is always some loss in a charge cycle so I just purchase from grid. In case it helps, here it comes: https://github.com/martinknechtel/sungrow-tibber-home-assistant