mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #53: Large `Meter power` value during blackout (meter inaccessible)

elektrinis, 2023-01-31:
Had a blackout today, home was running on battery.

Got `meter active power` (and also all meter phases) values equal to 2^31.

Meter was powered off and was inaccessible via RS485.
Perhaps a workaround is needed?

[image]

Louis712, 2023-01-31:
It could be done to set up a template to set the Meter active Power to Zero if it's got an unreasonably high value (for example if it's over 30kw.)

mkaiser, 2023-02-01:
for me it looks like expected behaviour:

The meter is connected between grid and house wires and is powered by one  'incoming' phase from the grid. 

If the grid is off, the meter does not have power. All RS485 read/write requests will return 0xFFFF FFFF. This raw value is interpreted as a signed integer, so 0x7FFF FFFF (2 147 483 647 decimal)

We could use a template sensor to check the raw value for 0x 7FFF FFFF and set the sensor to "unavailable".  
It is somehow easily testable: Just switch off your circuit breakers and have a look . 

It should be just 4 sensors, right? Total power and phase A-C power.

elektrinis, 2023-02-01:
Yes I think template for this value would be the correct way to go.
Yes I think it's needed for 4 sensors.

I'm just not sure if it should be "unavailable" of just zero.
In case of blackout, zero would be closer to reality.

dylan09, 2023-02-01:
I do not use the package in the form as here in the repository. I have split it into modbus, template and helper. So I don't have to restart HA with every change.

And unfortunately not so fit with GitHub and pull requests yet. But here is my change to the sensors:

1. I renamed the 4 modbus sensors and added a `_raw` to the name.
2. as a sample for one of the sensors, here is my template sensor definition:
```
  - name: sg_meter_active_power
    unique_id: 20737f1c-544e-4daf-8fcb-282462c4e28a
    device_class: power
    state_class: measurement
    unit_of_measurement: W
    state: "{{ states('sensor.sg_meter_active_power_raw') | float(0) }}"
    availability: "{{ states('sensor.sg_meter_active_power_raw') | int(0) != 0x7FFFFF }}"
```

Sensor is `unavailable` in case of black-out. I think that is the best state for that case. 

Translated with www.DeepL.com/Translator (free version)

mkaiser, 2023-02-06:
done in dabc6ec27d4b34e00458998232f03732f5523786

thank you for the hint with the "availability" parameter. I adapted all templates to this and readability got improved :)

`And unfortunately not so fit with GitHub and pull requests yet. But here is my change to the sensors:`

I am still learning all this, too (svn is enough at work for my things...). Just give it a try next time - I won't bite, if something goes wrong! ;)

closed 2023-02-06 by mkaiser