mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #57: Exporting Power not working

andi-blafasl, 2023-02-07:
After a few weeks with very little sun, today my battery was fully charged and the Inverter (SH6.0RT connected directly via eth, not through winet-s) started exporting power. But in HA there is no info about that. The Binary Sensor did not show and the exporting value is always zero.
Here's a screenshot from the day:
[image]
as you can see, binary sensor for importing is working as expected. Any Ideas what's going wrong?

mkaiser, 2023-02-07:
Are you using the latest version? I recently made some changes, when the running_state is not available.

Can you check the version and also the value of running_state?

GrandAmmor, 2023-02-08:
I had the same problem and fixed it with the update (230206).

At the moment sensor.meter_active_power shows unavailable for me. sensor.meter_active_power_raw works well in my evcc config.

andi-blafasl, 2023-02-08:
I'm on version 2023-01-26, which was the latest at the time of posting 😄
Will do an update on the weekend.

andi-blafasl, 2023-02-13:
Did the update to 2023-02-12 yesterday night and today it's very sunny. Export Power is now working for me:
[image]

[image]

Importing is not shown because the battery was never empty the last days.

mkaiser, 2023-02-13:
> Did the update to 2023-02-12 yesterday night and today it's very sunny. Export Power is now working for me: ![image]
nice - thank you for the feedback :)

[AlexHG2](https://github.com/AlexHG2)
> Updated yesterday to version "2023-02-12". The result for export unfortunately still remains at "0". Export_power_raw have results.

Can you give me more information? What is your inverter and what are the other values showing? Maybe post me screenshots of the PV tabs

mkaiser, 2023-02-14:
> Not sure if this is related, but I had some battery issues, and now it won't discharge. I can set it to charge in forced mode, and it will charge the battery from sun in self-consumption mode. But it will not discharge in forced mode, and will not run on battery in self-consumption mode. After a full charge to 100% it did hold my house on battery for an hour or so, and then it just moved to the grid. So the battery is sitting at 98% now and I'm waiting for sungrow support to answer...
> 
> But it does export fine when it's sunny and battery is full.

I had that once, when I started playing with the modbus interface. After a factory reset of the inverter everything was smooth again.

Can you check the two "start power" values?

[image]

Another possibility could be that the battery is going to a calibration mode. Just read about that, never experienced myself

elektrinis, 2023-02-14:
These two values are at zero. Not sure what this means, I thought it will not start charging/discharging until surplus/load is above this value?
How did you factory-reset it? I can't find this in manual...

AlexHG2, 2023-02-14:
> nice - thank you for the feedback :)
>
> [AlexHG2](https://github.com/AlexHG2)
>
>
> Can you give me more information? What is your inverter and what are the other values showing? Maybe post me screenshots [...]

Good news, it s working now :-)  I deleted the old config which i moved to a subfolder in the integrations dir as backup. 
Therefore the version fixed the issues also on my side. Thanks a lot!!

mkaiser, 2023-02-14:
glad that it worked ! :)

mkaiser, 2023-02-14:
before resetting, try to set the battery dis/charging start power values

The sungrow app annoyed me so much .... I spent 15 minutes searching for it. Then I implemented it in yaml. Can you check the last commits?
[image]

I don't recall exactly, but I think I triggered a factory reset using the isolarcloud webinterface, logged in as installer and after some time I found a button for factory reset

mkaiser, 2023-02-14:
sorry [elektrinis](https://github.com/elektrinis), did not mean to close the issue, yet :)

dylan09, 2023-02-14:
If I remember correctly Sungrow recommends to set start charging and discharging power to a value > 0.
My setting is 100W for both.

And Max SoC should be set to 100%. Set that the balancer could do their work. But I think thats not the reason why it is not working for you.

Have solved my last problem through complete inverter shutdown. Shut off AC, DC and battery (sequence depends on your setup). Than wait 10-15 minutes and restart.

mkaiser, 2023-02-14:
yep, 100W for start values is reasonable. 

Will increase the input for max SoC to 100 in the next commit

elektrinis, 2023-02-14:
Changed this to 100W, no effect.
Also could not find the reset function... will keep looking.

mkaiser, 2023-07-05:
there was no activity here for a long time. 

Do the issues still exist? if yes, please check if your inverter has the current firmware first and report if something changed, please