mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #60: Error handling

elektrinis, 2023-02-11:
I just figured out my battery is failing, error 833 and some others. I can force-charge it, but can't discharge, it will just ignore the command. Same in self-consumption mode. Inverter status is unchanged and I don't get any feedback that something is wrong, it just ignores me.
I am discussing this with sungrow support.

Would it be possible to read errors/status from the inverter somehow?
Anyone has info what running/system state codes mean?

mkaiser, 2023-02-11:
A quick peek in the modbus datasheet shows that the running state seems fine:_

[image]

In your case bits 1, 2 and 5 are toggling, bit3 is always 1.

bit1: battery charging
bit2: battery discharging
bit5: Import power

bit3: type of load (active / reactive)

Where did you get the error 833 from?

in the modbus datasheet, in appendix 1.3 there are fault code instructions. I never cared to implement them until now, thinking that in such a case you just have a look into the sungrow app. 

maybe some of the registers between (Inverter alarm 13050) and (min cell temp 13118 ) might be of help, if you don't want to use the app. 

What kind of battery do you have? My pylontech has an own webinterface for fault messages

elektrinis, 2023-02-17:
It's sungrow SBR battery. The error happened again today, inverter went in to "standby" state.
I'm seeing this on iSolarCloud portar.
Sadly support is ignoring me. 
I think I will just open it up and measure every cell myself. Current voltage levels are not what one would expect from LiFePO4, as it's 3.45V at full charge, while the normal level is 3.65V.

Louis712, 2023-02-17:
I also contacted Support a while ago and they didn't answer for about a month. Hopefully they'll Tell you what you can try, because it could just be the BMS misbehaving.

elektrinis, 2023-02-20:
My battery errored out today again. Inverter running state is 40 (0010 1000).
Bit 3 set to "1" means "load is active"
Bit 5 set to "1" means "importing from grid"

I also gave "system state" set to 8, which only means it is in Standby state.
So "Standby" state and the bits are probably the only way to check if it's doing what it's supposed to, and otherwise treat it as error.

Can we implement reading of registers 88-101, as per appendix 1.3? Looks like any non-zero value in any of these registers points to a fault?

Louis712, 2023-02-20:
Did you see any error in the isolarcloud? The cable connecting the BMS to the inverter could also be a Part of the problem.

elektrinis, 2023-02-21:
I get "Error 833" in iSolarCloud, the log is filled with it every 20 seconds. But strangely there is no "active fault" even when this error is being reported in log history at that moment.
The most interesting thing is that I get these errors mostly only when sun is down. Here's what I discovered:

2023-02-20 18:36 ... 2023-02-21 7:20
2023-02-16 23:00 ... 2023-02-17 8:20
2023-02-09 21:30 ... 2023-02-10 8:19
2023-02-06 18:29 ... 2023-02-07 8:17

If it's errored out, I can't get it running in the evening/night/morning, tried to start/stop, change modes etc, nothing helps. That means I can't force charge/discharge the battery, I can't get my house running off battery in self-consumption mode and backup function would probably fail in case of blackout.
But the moment sun rises, even if there is no actual sun due to clouds and snow on roof, it just fixes on it's own. Looks like it has a hardcoded "sunrise" timestamp when it changes modes or resets itself. Go figure.

elektrinis, 2023-02-21:
By the way, we had a blackout just few minutes ago due to our electrical company doing some maintenance work.
The backup failed to work.
Sun was shining and battery was charging, SOC at 30%. The inverter had a "running" state (not in error) and set to self-consumption mode, but battery mode dropdown was left in "forced charge".
No errors or warnings, it just came back to life once power was up again, as if nothing had happened.
Couple weeks ago we had the same thing, but then backup worked just fine. Perhaps it's due to that "forced charge/discharge" dropdown that I thought has no effect in self-consumption mode. Will try to figure this out by selecting "Stop" and switching power off.

dylan09, 2023-02-21:
You have disabled Backup Mode?

elektrinis, 2023-02-22:
I have not, yet I found it disabled.
BTW it rent in to error state last evening and recovered this morning, as usual.
Investigation continues....

dylan09, 2023-02-22:
Would only give a short hint. Because in your Snapshot it is disabled. And I didn't understand the markers you have painted into the snapshot.

As if I am right, `Backup Mode` must be enabled to switch to Off-Grid mode.

elektrinis, 2023-02-25:
I have a new problem. My modbus if failing and I can't understand why.
The data is just randomly jumping from valid to unknown, as a christmas tree.
Any ideas?

[image]
[image]

Louis712, 2023-02-25:
Sungrow inverters sometimes have problems with modbus. What you can Check is if you've got the newest Firmware on your inverter. If not, update it, for me it fixed the issue, if you've already got the newest Version, you can add message_wait_milliseconds: 50 to the configuration, right under timeout. You can try other delays, but for me 50 ms worked fine.

elektrinis, 2023-02-25:
It's latest. Somehow the issue only appeared now.
Will try the delay. Thanks.

elektrinis, 2023-02-28:
I had this today. Was in self-consumption mode, plenty of charge remaining. It simply stopped discharging the battery.
I can't find anything that could have changed in settings/states.
No change when switching modes, also won't listen to forced mode.
Tried to restart, then it stuck in standby.

Anyone have any ideas if this inverter goes in to standby for night etc?
Nothing in iSolar fault logs this time.

closed 2025-02-12 by github-actions[bot]