mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #65: Deactivate/ hide battery features

i-peter, 2023-02-16:
Hi,
first of all: Thanks for your great work!

I have a SH8.0RT and it's working great for me. Althought it was hard to find out, that I have to use the second LAN port next to the WiNet-S dongle to make it work. (BTW: Is there anywhere a guideline how to upgrade firmware for Sungrow? I only found outdated descriptions, which are not working for me.)

The question/ feature request I really have is the following:
I don't have a battery and I would like to deactivate or hide all battery functionallities. 

Would be great to have a central parameter instead of going through all lines of code to deactivate.

Best regards,
Iven

mkaiser, 2023-02-16:
hi

> I have a SH8.0RT and it's working great for me. Althought it was hard to find out, that I have to use the second LAN port next to the WiNet-S dongle to make it work.

I have the first revision of the SH10.RT without WiNet-S, so I can't really say something about all this.  I just know, that putting the cable in the WiNetS port often causes problems :/

Now, that you figured it out, could you give me some hints to update the documentation for this? I am pretty sure that would help others, too :)

> I don't have a battery and I would like to deactivate or hide all battery functionallities.
hmm the only way I see at the moment is to split the modbus_sungrow.yaml in multiple files, lets say "basic", "battery" and "history". 
I am afraid, that a more dynamic way is not possible in the yaml (there is no yaml-wide if/else possible).

Will give the splitting some thoughts in the future :)

With a  more versatile Python integration for Home Assistant this would be much easier to achieve and more dynamic (just include feature XY, if checkbox is marked). Unfortunately, I have never used Python, yet and quiet frankly I don't have the time at the moment to learn it.

If someone knows how to do this, I would try to help :)

mkaiser, 2023-09-11:
had many thoughts about this. 

I only have the capacity to deal with one integration variant (so the one with battery). 

If you do not want them, delete the YAML entries. 

I am sorry, but there is no other way with a pure YAML-integration. Only python could fix this

closed 2023-09-11 by mkaiser