mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #69: Featur request: `Total PV Power`

elektrinis, 2023-02-20:
Currently `Total DC Power` is being calculated by the inverter incorrectly - this value is somehow influenced by other inverter, connected after smart meter.
In screenshot below, when my other inverter is idling, it should be around 712W, but only 563W is reported. There's no way  150W are lost to heat in inverter?!

So I suggest to add a sensor called `Total PV Power`, which is a sum of `MPPT1 Power` and `MPPT2 Power`.

Louis712, 2023-02-21:
The problem with this solution would be, that the MPPT currents aren‘t really accurate. They don‘t get reported by the inverter correctly. Iit also outputs as accurate as 100mA. For example, in an extreme case the current of MPPT2 could be 0.55A, that would be converted to 0.60A. That‘s an extremely big difference. That would bring your „Total PV Power“ down to 631W. It‘s still a big loss, but not as vig. Maybe there is some undocumented sensor that puts out the real total Power, because MPPT1 Power and MPPT2 Power are calculated using the corresponding voltage and current. A friend of mine has got a SMA inverter and that actually outputs power of each MPPT. And as far as I know Sungrow and SMA are somewhat similar.

Louis712, 2023-02-21:
For example, my PV is starting to generate. Total DC Power is about 500W, the voltage of the two MPPTs is there, but the current isn‘t getting reported. It‘s always about half an hour late.

elektrinis, 2023-02-21:
Just did a quick comparison of `total DC power` with my helper that I named `solar roof power`.
I wonder why it differs so much. Any ideas?

[image]

elektrinis, 2023-02-21:
To make my case stronger, just adding a screenshot from my another issue. In this case my other inverter is generating power (connected between smart meter and SH10RT). In this case my solar roof is covered with snow and therefore not generating any power, however another inverter (solar wall) is generating something. And this somehow ends up in SH10RT's DC power....
[image] [image]

mkaiser, 2023-02-28:
this is all very strange, indeed. 

I am still very confused, which "power value" would be the correct power value. 

Measuring DC is more accurate than AC. So The "Total DC" power could be in fact "AC generated" power.

dylan09, 2023-02-28:
I think you have connected your HA instance to the master inverter? And you have configured master / slave information and settings correctly in iSolarCloud?

Probably  your problem is related to how the two inverters are configured in iSolarcloud. There are different ways how to setup how the values for master and slave will be calculated correctly.
But if I remember correctly, there is no configuration which will be give you valid values in all cases.

elektrinis, 2023-02-28:
No, I only have a single Sungrow SH10RT and, to be precise, 14 pcs of per-PV_module micro inverters. Those are in no way integrated with the Sungrow and they output AC directly, just as any other household device.

So SH10RT is seeing a mismatch between smart meter readings and what the inverter measures itself, and it panics.
It's an obvious bug, as even Sungrow marketing material shows one can connect 3rd party inverters in this way:

[image]

dylan09, 2023-02-28:
I understand. Thought you have two Sungrow inverter.
Maybe a question for photovoltaikforum.de. Maybe there is someone with a similar configuration.

closed 2025-01-28 by github-actions[bot]