mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #71: GUI to read/set any register

elektrinis, 2023-02-21:
I would love to have a way to debug my system by setting or reading any register.
For example I want to know min/max/avg battery cell voltages (registers 13112-13114), min/max cell temperatures etc.
Also would be interesting to play with Load timing registers 13009 etc.
I understand I am probably the only one here with sungrow SBR battery?? So It's obvious not everyone needs this. Some even want no battery data at all.
I also understand I can make a script with `modbus.write_register` service call myself, but perhaps this could be a good idea to include a simple GUI in the dashboard?
One number input for `register number`, one input for `data` and button `Write`, one `data` with button `Read`.
Or just a script that does this?

ViktorReinhold, 2023-02-21:
Hi @elektrinis,
are you aware of the WiNet-S UI? 
It is available under the IP of you WiNet-S-Dongle. If you login with Username 'admin' and Password 'pw8888' you can access a tab called 'General Parameters' under the menu 'Device Monitoring'. There is a form where you can issue some commands to read and write registers you want. I have not tried to write anything here. The same form is available for the inverter and for the battery.

[image]

I'm not familar with the hex schema, but in this example you can see, that I can read out the device type code of my SH10RT-V112 inverter. When you ignore the two pairs of zeros at the beginning, the rest of "0E 0F" matches the code which is checked in the modbus_sungrow.yaml configuration file.

{% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E0F) %}
            SH10RT-V112

If you want to have a decimal number, you also need to convert the last four digits from Hex to Dec.

I'm also not sure what the exact meaning of 'Register Number' and 'Address Type' is. But with this settings I could get something meaningful.
And you are not alone with a Sungrow SBR battery ;-) I also have one, as you can see.

Louis712, 2023-02-21:
I also have a Sungrow SBR battery, but it doesn't show up in the web interface. Did you change anything?

ViktorReinhold, 2023-02-21:
Hi @Louis712, no, I did nothing for that.

lassej01, 2023-02-24:
I read in a forum that it depends on the version of the SBR battery if they show up as own entities or not. There is a standard model and two version of a premium model. The 2nd version of the premium model shows up if i understodd it right, and the first version of the premium may show up after the latest battery upgrade. I am probably on the latest described version. But have not seen any working updates for me yet. Some zip file though, but i cannot install that one.

Louis712, 2023-02-24:
@lassej01 I know, that there are two versions and I have the older one (orange label on the packaging). How did you do your update? My battery doesn't show up in iSolarCloud or anything. I wanted to do it locally, downloaded SBR128 firmware, but the app said, that there wasn't any file found. Do you have any information on that?

mkaiser, 2023-02-28:
> I also understand I can make a script with modbus.write_register service call myself, but perhaps this could be a good idea to include a simple GUI in the dashboard?

I don't think this is a good idea, because people will fuck up when they have a chance to randomly access registers :)
If you want to test registers, I can also recommend QModBus Master for this (Windows Software).

elektrinis, 2023-02-28:
I have automated this a little with service that's already provided, and it's good enough. I think you are right, people can mess up things and current way is good enough for advanced user.
Thanks and I'm closing this for now.

closed 2023-02-28 by elektrinis

elektrinis, 2023-02-28:
Sorry, I just noticed there is service to write, but not to read data.
Not sure if those registers are a part of this project though.
Can we implement a service `modbus read register`?