mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #72: Battery capacity scale or unit wrong?

dylan09, 2023-02-28:
Hi,

I am testing a little bit around with my battery and backup settings.
I have found, that I must have a really huge battery. Entity `sg_battery_capacity` tells me a capacity of 12.000 kWh. 🙃
Maybe the scale factor or the unit is wrong. Connected is a HVS 12,8 kWh battery. Solutions could be
- change the scale to 0.01
- change unit to Wh
I could not find the source for this register. It's not in the Sungrow Modus description. So I am unable to verify which one is the correct solution.
Maybe someone could help how to fix the entity definition

mkaiser, 2023-02-28:
oh, you are right. In my case the battery capacity returns "14,400 kWh"

... damn ',' vs. '. '       ... mixed them up...., again ;)

I saw the register definition in a german forum. I guess it was iobroker or https://www.photovoltaikforum.com. 

Have no time to test at the moment. If you could do a PR, this bug will be fixed sooner :)

dylan09, 2023-02-28:
Found the source for the register definitions: [photovoltaikforum.de](https://www.photovoltaikforum.com/thread/166134-daten-lesen-vom-sungrow-wechselrichtern-modbus/?postID=2837513&highlight=33049#post2837513) 
Scale should be 0.01. 

Now capacity has correct value.

closed 2023-02-28 by mkaiser