mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #76: Flapping Binary Sensors Importing / Exporting Power

andi-blafasl, 2023-03-13:
I often see the two binary sensors flapping. Especially when the Batterie is charging the import/export power is flapping around +/-50W. I'm trying with an added `delay_on` and `delay_off` no.

```
      - name: Exporting power
        unique_id: sg_exporting_power
        availability: "{{states('sensor.running_state')|is_number or states('sensor.export_power_raw')|is_number }}"
        state: >-
          {% if states('sensor.running_state')|is_number %}
            {# use available sensor running_state #}
            {{ states('sensor.running_state')|int(default=0)|bitwise_and(0x10) > 0 }}
          {% else %} 
            {# workaround for SH*RS inverters without working running_state #}
            {% if states('sensor.export_power_raw')|int > 0 %}
                1
            {% else %} 
                0 
            {% endif %}
          {% endif %}
        delay_on:
          seconds: 30
        delay_off:
          seconds: 30

      - name: Importing power
        unique_id: sg_importing_power
        availability: "{{states('sensor.running_state')|is_number or states('sensor.export_power_raw')|is_number }}"
        state: >-
          {% if states('sensor.running_state')|is_number %}
            {# use available sensor running_state #}
            {{ states('sensor.running_state')|int(default=0)|bitwise_and(0x20) > 0 }}
          {% else %} 
            {# workaround for SH*RS inverters without working running_state #}
            {% if states('sensor.export_power_raw')|int < 0 %}
              1
            {% else %}
              0 
            {% endif %}
          {% endif %}
        delay_on:
          seconds: 30
        delay_off:
          seconds: 30
```

Not sure, but maybe a longer delay_on and a shorter delay_off is needed to prevent the two sensors from showing up in parallel? will test the behavior in the next days.

Any thoughts about this approach?

dylan09, 2023-03-13:
I also have tried this. Seen the same behaviour with the charge/discharge binary sensors. Adding `delay_on`and `delay_off` didn't work as I expected.
For testing I have changed the `scan_interval` for modbus sensor `running_state` to 30 sec. That helped a little bit because when discharging the binary sensors changed state at the 10 sec scan interval. These changes significantly reduced. 
But in the near future I will trey to change the interval to 60 sec.
I think its not so important to get every change for the inverter running state. So maybe 60 sec could be a compromise. 
But I don't want to set it longer than 60 sec. After this time I would like to know that the inverter is in off-grid (backup) mode,

andi-blafasl, 2023-03-14:
It looks promising for me:
Bevor the change:
[image]
After adding delay_on/off
[image]
It's no longer flappen every ten seconds between import and export. I will increase the on time a bit and keep watching.

mkaiser, 2023-03-21:
interesting HA feature (https://www.home-assistant.io/integrations/template/#delay_on), they have nearly thought of everything :)

what about just adding the delay_on to Exporting power 
        delay_on:
          seconds: 60

and not use the delay_off. 

so going to "exporting" will take at least 60 seconds before the value is set. But as soon as power is imported, the sensor goes to "not exporting"

mkaiser, 2023-07-07:
Adressesed in a1a5688

thank you! :)

closed 2023-07-07 by mkaiser