mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #77: Entities are not receiving data

lampi84, 2023-03-14:
I'm havin a Sungrow SH6.0RT inverter. I followed the instructions. An entered the IP adress, but unfortunatly the entities are not receiving any data from the modbus (See attached screenshot).
Is there a way to troubleshoot what is the problem there.

[image]

MrMontesa, 2023-03-15:
Hey @lampi84,
an error messages from home-assistant.log? Also try to "telnet IP 502" from the host homeassistant is running on towards the inverter. Does it connect?
And lastly make sure to use the native LAN adapter. I had issues with wifi dongle or the LAN on the wifi dongle.
Hope this helps

lampi84, 2023-03-16:
Hi,
many thanks for your quick reply. i just checked the logs. From the error messages it seems like it can connect but it is not able to read the data. I'm getting a lot of these kind of error messages:

2023-03-14 21:21:56.911 ERROR (SyncWorker_1) [pymodbus.factory] index out of range
2023-03-14 21:21:56.912 ERROR (SyncWorker_1) [pymodbus.transaction] Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2023-03-14 21:21:56.912 ERROR (SyncWorker_1) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Modbus Error: [Input/Output] Unable to decode request
2023-03-14 21:21:56.923 ERROR (SyncWorker_4) [pymodbus.factory] index out of range

Do you have an idea how to fix it or to find out how to change configuration so that it can read the data.

lassej01, 2023-03-16:
Hi! Are you using the real Ethernetport (Rj45) thats behind (closer to wall) than the wifi-module?

lampi84, 2023-03-16:
This is the users manual for the wifi dongle.
[https://www.memodo.de/media/pdf/e0/78/1f/zubehoer-sungrow-winet-s-communication-module-quick-installation-guide-en_2c92296d00358ad18a84a5.pdf](https://www.memodo.de/media/pdf/e0/78/1f/zubehoer-sungrow-winet-s-communication-module-quick-installation-guide-en_2c92296d00358ad18a84a5.pdf)

lampi84, 2023-03-16:
[image]
 This view is showing some values i'm getting.

lampi84, 2023-03-16:
[image]
I got it working :-) if i have both connected the wifi dongle and the ethernet jack it is working fine. I think the rest of the values will be filled once it has build up some statistics. Many thanks for all your help ;-)

Louis712, 2023-03-16:
Did you change the IP for modbus to the RJ45 plug on the inverter? The winet s dongle doesn't give out all of the information. It's somewhat limited.