mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #78: Energy dashboard suddenly all wrong values

Cliff-R-K, 2023-03-24:
I have been using this addon for many months without any problems but yesterday my energy dashboard suddenly shows wrong values. Can anyone tell me what can cause this ?

[image]
[image]

Louis712, 2023-03-24:
Maybe a corrupted database? What is the true value of Total PV generation?

Cliff-R-K, 2023-03-24:
How do I check that value ?
[image]

Cliff-R-K, 2023-03-24:
[image]

Louis712, 2023-03-24:
What the hell? That thing should only increase. Yours is just fluctuating between 0 and about 12.000kWh… Which version of this yaml integration have you got? If you’ve got one older than about 1 month you can try using the newest one. If not generating, it should stay the same value.

Cliff-R-K, 2023-03-24:
Ok will try to update, I was using the version from 2023-01-20. But I can see that the WiNet-S module was updated the same day as my problems started so I am guessing the new firmware messed up something

Louis712, 2023-03-24:
That’s your problem. Not particularly the version of this yaml file, although you could update it, you’re using the WiNet-S module. You should use the internal RJ45 jack if possible. The WiNet-S doesn’t give you all the values.

Cliff-R-K, 2023-03-24:
Ok, but I think that it is a little strange that it have worked "well" for many months using the WiNet-S. But can I still use the isolarcloud app/webpage as usual if I plug in a ethernet cable directly to the inverter ?
And do I need to change any settings in the inverter before/after for the ethernet to be activated ?

elektrinis, 2023-03-24:
Maaaybe related to this: after restart my values were updating like 10 times per second, even in main dashboard. I thought cool, finally!
And then it slowly went back to slow updates as usual.
Not sure about the version, it like 3-5 months old. Updating is a pain and I don't have time for it.

mkaiser, 2023-03-24:
It seems that this is (again) some Sungrow issue:

Abtract description: Sungrow uses two Modbus implementations on separate microcontrollers: 
- the internal SH*RT port 
- The WiNet S

The implementation of the internal port is very stable - if possible, use this
Sungrow is actively working on the WiNetS implementation for over a year now. Every update changes things (like you experienced). It looks like they make two steps forward, one back (sometimes the other way round...)

[Cliff-R-K](https://github.com/Cliff-R-K): do you have an internal Etherner port, in addition to the WiNetS?

Cliff-R-K, 2023-03-27:
I took the advice and connected an ethernet cable directly to the inverter and it looks like everything works again now.

Cliff-R-K, 2023-04-22:
After switching to the LAN port of the inverter all worked as expected again. But now I am forced to change back to the WiNet-S again as I have started using a service where I get payed for letting a power-company controling my battery. (Long story...)

Is there some easy way to manualy change the code so that the 0 values does not register ?

closed 2023-07-07 by mkaiser