mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #79: binary sensors battery charging and discharging are off

Falkoh150, 2023-03-27:
What can be the reason for the sensors mentioned switching off again and again and no longer supplying any values? I'm new to home assistant and don't really understand the connections yet. Many thanks for the help.

dylan09, 2023-03-27:
They are constantly `off`? Or do they toggling between `on` and `off` in short intervals?

Falkoh150, 2023-03-27:
They were on at the beginning and then they went off, so I didn't get any more values ​​for charging and discharging. If I turn them on manually, they turn off again after a short time. Maybe it has to be like this, I'm not that familiar with it. But I would like the values.

Falkoh150, 2023-03-27:
WiNet

dylan09, 2023-03-27:
That should be the reason for your trouble. Please read the 3rd paragraph [binary sensors battery charging and discharging are off · Issue #79 · mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant).
ModbusTCP isn`t working without errors over WiNet Adapter. Use the native LAN port.

dylan09, 2023-03-27:
Maybe you could use the cable connected to the WiNet for testing. I think for a few hours it will be ok to be not connected to iSolarCloud.
And be aware that your DHCP server will assign a new IP address to the Sungrow LAN adapter. Must be changed in your secrets.yaml

Falkoh150, 2023-03-27:
Thank you for the tips and help.

mkaiser, 2023-04-09:
@Falkoh150 

do you have updates? Could you solve the issues?

Falkoh150, 2023-04-11:
With a LAN cable on the separate LAN connecter, all data is now delivered stably. Thank you very much for your help

closed 2023-04-11 by mkaiser