mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #80: Turn on Load Relay (DO relay)

Kawobei, 2023-03-28:
Hi, I do not understand something from the documentation.
Both addresses for Total PV generation 13002 and Load Adjustment 13002
10 Load 1 adjustment mode
13002 U16
0: Timing mode;
1: ON/OFF mode;
2: Power optimized mode

How can I add this as a switch?
I have my hot water hooked up but I would like more complex automation rules than the inverter allows.
Regards,
Brenton

Louis712, 2023-03-28:
Should still work. Those are different types. If you look at "Inverter start stop" and "System state" it's the same thing.

Kawobei, 2023-03-28:
You are correct, I got it working last night (very happy) my guess was the holding ones are different even though same numbers.
Not sure if you want the code? I dont know how do pull/push stuff.

Louis712, 2023-03-29:
You can Post it here. Maybe someone will also find it useful. I might Do a pullrequest.

Kawobei, 2023-03-29:
# Home Assistant Sungrow inverter integration
# https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant
# by Martin Kaiser
# last update: 2023-03-24

modbus:
  - name: SungrowSHx
    type: tcp
    host: !secret sungrow_modbus_host_ip
    port: !secret sungrow_modbus_port
    retry_on_empty: true
    retries: 10
    close_comm_on_error: true
    delay: 5
    # timeout: 5
    sensors:
      - name: Sungrow device type code
        unique_id: sg_dev_code
        slave: !secret sungrow_modbus_slave
        address: 4999 # reg 5000
        input_type: input
        count: 1
        data_type: uint16
        swap: word
        scan_interval: 600

      - name: Daily PV generation & battery discharge
        unique_id: sg_daily_pv_gen_battery_discharge
        slave: !secret sungrow_modbus_slave
        address: 5002 # reg 5003
        input_type: input
        count: 1
        data_type: uint16
        swap: word
        precision: 1
        unit_of_measurement: kWh
        device_class: energy
        state_class: total_increasing
        scale: 0.1
        scan_interval: 600

      - name: Total PV generation & battery discharge
        unique_id: sg_total_pv_gen_battery_discharge
        slave: !secret sungrow_modbus_slave
        address: 5003 # reg 5004
        input_type: input
        count: 2
        data_type: uint32
        swap: word
        unit_of_measurement: kWh
        precision: 1
        device_class: energy
        state_class: total_increasing
        scale: 0.1
        scan_interval: 600

      - name: Inverter temperature
        unique_id: sg_inverter_temperature
        slave: !secret sungrow_modbus_slave
        address: 5007 # reg 5008
        input_type: input
        count: 1
        data_type: int16
        precision: 1
        unit_of_measurement: °C
        device_class: Temperature
        scale: 0.1
        scan_interval: 10

      - name: MPPT1 voltage
[... 2537 lines cut ...]
          {% else %} 0 {% endif %}"
          hub: SungrowSHx
    mode: single
    
  - id: "44321413553544" # random number
    alias: update DO relay ONOFF cmd
    description: "update DO relay ONOFF cmd"
    trigger:
      - platform: state
        entity_id:
          - input_select.set_sg_do_relay_onoff_cmd
    condition: []
    variables:
      DO_relay_ON: 0xAA
      DO_relay_OFF: 0x55
    action:
      - service: modbus.write_register
        data_template:
          address: 13010 # 13011
          slave: !secret sungrow_modbus_slave
          value: "{% if is_state('input_select.set_sg_do_relay_onoff_cmd' , 'ON') %} {{DO_relay_ON}}
          {% elif is_state('input_select.set_sg_do_relay_onoff_cmd' , 'OFF') %} {{DO_relay_OFF}}
          {% else %} 0 {% endif %}"
          hub: SungrowSHx
    mode: single

Louis712, 2023-03-30:
Would you mind checking this for me, since I don't have anything connected to my DO relay?
[just copy the content of modbus_sungrow.yaml](https://github.com/Louis712/Sungrow-SHx-Inverter-Modbus-Home-Assistant-fork)

Kawobei, 2023-03-31:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

OFFICIAL

I did try today and it didn’t work. I am in the process of trying to identify which line is the issue.
Looking at the changes in a compare I don’t really see what is the problem. I’ll spend some more time over the weekend.

OFFICIAL

RTO Code: 4693 CRICOS no. 01985A
This email and any attachments may contain privileged and confidential information. If you are not the intended recipient, you may not distribute reproduce this e-mail or the attachments. If you have received this message in error, please notify us by return e-mail.

Please consider the environment before printing this email.

closed 2025-01-21 by github-actions[bot]