mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #81: Index out of range

Timon321, 2023-03-28:
I have some errors in my logs of my SH8.0RT:

Logger: pymodbus.logging
Source: components/modbus/modbus.py:398
First occurred: 13:24:57 (826 occurrences)
Last logged: 13:40:05

General exception: index out of range
Modbus IO exception Modbus Error: [Input/Output] Unable to decode request

`2023-03-28 13:38:35.351 ERROR (SyncWorker_7) [pymodbus.logging] General exception: index out of range
2023-03-28 13:38:35.352 ERROR (SyncWorker_7) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
2023-03-28 13:38:35.353 ERROR (SyncWorker_7) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Modbus Error: [Input/Output] Unable to decode request
2023-03-28 13:38:35.363 ERROR (SyncWorker_2) [pymodbus.logging] General exception: index out of range
2023-03-28 13:38:35.364 ERROR (SyncWorker_2) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
2023-03-28 13:38:35.374 ERROR (SyncWorker_3) [pymodbus.logging] General exception: index out of range
2023-03-28 13:38:35.375 ERROR (SyncWorker_3) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
2023-03-28 13:38:35.386 ERROR (SyncWorker_8) [pymodbus.logging] General exception: index out of range
2023-03-28 13:38:35.387 ERROR (SyncWorker_8) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
2023-03-28 13:38:45.334 ERROR (SyncWorker_5) [pymodbus.logging] General exception: index out of range
2023-03-28 13:38:45.336 ERROR (SyncWorker_5) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
2023-03-28 13:38:45.337 ERROR (SyncWorker_5) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Modbus Error: [Input/Output] Unable to decode request
2023-03-28 13:38:45.347 ERROR (SyncWorker_7) [pymodbus.logging] General exception: index out of range
2023-03-28 13:38:45.349 ERROR (SyncWorker_7) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
2023-03-28 13:38:45.360 ERROR (SyncWorker_2) [pymodbus.logging] General exception: index out of range
2023-03-28 13:38:45.362 ERROR (SyncWorker_2) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
2023-03-28 13:38:45.373 ERROR (SyncWorker_3) [pymodbus.logging] General exception: index out of range
2023-03-28 13:38:45.375 ERROR (SyncWorker_3) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
2023-03-28 13:38:55.338 ERROR (SyncWorker_5) [pymodbus.logging] General exception: index out of range
2023-03-28 13:38:55.338 ERROR (SyncWorker_5) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
2023-03-28 13:38:55.340 ERROR (SyncWorker_5) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Modbus Error: [Input/Output] Unable to decode request
2023-03-28 13:38:55.350 ERROR (SyncWorker_7) [pymodbus.logging] General exception: index out of range
2023-03-28 13:38:55.351 ERROR (SyncWorker_7) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
2023-03-28 13:38:55.362 ERROR (SyncWorker_2) [pymodbus.logging] General exception: index out of range
2023-03-28 13:38:55.363 ERROR (SyncWorker_2) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
2023-03-28 13:38:55.373 ERROR (SyncWorker_3) [pymodbus.logging] General exception: index out of range
2023-03-28 13:38:55.375 ERROR (SyncWorker_3) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request`

Timon321, 2023-03-28:
Yes, the most values are valid:
[image]

dylan09, 2023-03-28:
WiNet-S Lan Port is not working flawlessly with ModBus. Try using the inverter LAN interface.

Timon321, 2023-03-28:
Now with the inverter LAN-Interface it is working flawlessly! Thank you.

Louis712, 2023-03-28:
That's weird. For me the LAN port got a new IP address as soon as it was connected. No powering down and such. Maybe it's the version of the inverter? I've got the normal one, perhaps that only applies to the V112?
All I wanted to say with this was, that you should try just plugging it in and see if it works. That would save you a bit of work.

dylan09, 2023-03-29:
I think adding the picture to the documentation will clarify things heavily. It's an eye catcher. Really nice idea.

My SH10RTv112 gets a new IP address every time I disconnect / connect the LAN cable. Maybe you have to wait a few seconds before reconnecting the cable. So the DHCP client inside the inverter has a chance to detect the disconnect.

dylan09, 2023-03-29:
> @dylan09 Have you tried to give the inverter a fixed IP in your router? (put the IP as static in the DHCP part of your settings in the router)

For me everything is working. I am happy. And in my network servers and most IoT devices get a reserved IP address from the DHCP server. Simplifies firewall configuration and routing between different networks a little bit 😀
 
And regardless of reserving an IP address in your DHCP server or you let the server decide which address he gives to the client, the client musst start the DHCP discovery. So the Sungrow inverter must detect that LAN cable is disconnected. 

Don't know how long that takes. I am waiting 5-10 seconds after pulling the cable before reconnecting.