mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #83: Imported Energy Cost

mschneider82, 2023-03-30:
Hi, In the Tesla Dashboard there are two unknown entities

                    entity: sensor.daily_imported_energy_cost_3
                    entity: sensor.daily_exported_energy_compensation_3

How to create these entities with my custom kwh costs?

mschneider82, 2023-03-30:
cc @Louis712

Louis712, 2023-03-30:
I created those with the values given from the energy dashboard built in to home assistant. I found, that those aren't reliable. I will probably Do a pullrequest to add that stuff. You can copy the templates and add them to your configuration. Unfortunately I probably won't get to it until this evening.

Louis712, 2023-03-31:
I'll create a pull request as soon as the other feature is tested. Might take a couple of days. If you want to have it now, here I have it with the templates. [here](https://github.com/Louis712/Sungrow-SHx-Inverter-Modbus-Home-Assistant-fork/blob/1d1ff707c0e22cde053523edbcfc0a508f97ea35/modbus_sungrow.yaml)

hAMMERlized, 2023-04-15:
Hi,

anything happened here so far? i have the same issue. i copied the content of the modbus_sungrow.yaml which younhave linked above, but it's still the sam issue.

Louis712, 2023-04-15:
You should change the entities in the dashboard.

hAMMERlized, 2023-04-15:
> You should change the entities in the dashboard.

Not sure if right but the only entities which could fit in there that i could finde are **"sensor.daily_exported_energy_compensation_2"** and **"sensor.daily_imported_energy_cost_2"**. Is this right?

Thanks for the quick response btw.

Louis712, 2023-04-15:
That's right.

hAMMERlized, 2023-04-15:
ok but it satay at 0,0€ for both, exported and imported. Where does the sensors take the amount (€) for the imported kWh and exported kWh from? Did i miss some option where to edit the exact amount?

Louis712, 2023-04-16:
Right, I created input numbers for that. They're called "Set energy cost/compensation per kWh".

closed 2023-09-11 by mkaiser