mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #84: Grid Frequency Scale

ehajo, 2023-03-30:
Got 500Hz on SH8.0RT, don't know if others are affected... Changed parameter
      - name: Grid frequency
          scale: 0.01
from 0.1 for correct values...

Louis712, 2023-03-30:
Are you using the RJ45 port on the inverter? If you are using the one on the WiFi module, that's the problem.

ehajo, 2023-03-30:
hi louis, ah ok. i am using the connector on the wifi-module... i am not able to get the other one working....

Louis712, 2023-03-30:
That's the thing. The other one is probably working, you just need the WiFi module to send the data to iSolarCloud. Your ideal setup would be to run two ethernet cables, one to the internal lan port and a second one for the WiFi module. 
Huge fan of your BNO055 breakout, btw.

ehajo, 2023-03-30:
oh, did not need the isolarcloud since your great implementation into home assistant :-)
but tried a lot but native connector is not working :-(

thank you :-)

Louis712, 2023-03-30:
Maybe restart your inverter to get the internal LAN port to get a new ip address? You can look at the procedure in this issue [here](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/81#issuecomment-1486989690).

ehajo, 2023-03-30:
thank you, i will give it a try!

dylan09, 2023-03-30:
> but tried a lot but native connector is not working :-(

Did you take a look at your router which IP address was assigned when you connect the LAN port? It gets a different IP address. So you have to change the IP address in secrects.yaml.

mkaiser, 2023-04-09:
@ehajo 

could you solve the LAN connection issue?

ehajo, 2023-04-09:
i tried it today! had to buy a poe-switch and so on to get a 2nd lan-port outdoor....
frequency is working as attended with the native lan-port instead of the cloud-dongle.
so just as you said:
- native lan-port for communication, nothing else...
- dong-lan-port just for isolar-cloud...

mkaiser, 2023-04-09:
great!

thank you for the feedback :)

closed 2023-04-09 by mkaiser