mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #85: Switching grid off and running off battery when I want to?

elektrinis, 2023-03-31:
I want to be able to control my grid connection, I mean, to disconnect from grid and operate in Backup mode only, when conditions are right.
I'm aware I can simply add a contactor and control that as any device, but is this doable with inverter itself?

I want this because my energy company is charging me for phase imbalance - even when overall sum of phase power is zero, some phases are negative and some are positive. And they are making me "buy" the energy from myself. This is also a standard practice in many countries.

dylan09, 2023-03-31:
Probably depends on how often you want to switch!
But as soon as the inverter is in off-grid mode, it no longer feeds energy into the house grid. All loads that are not connected to the inverter backup draw their energy from the public grid.

How much energy are we talking about? For us it's about 0.1-0.3 kWh a day. 
This is because the inverter cannot respond quickly enough to load changes. And because of regulating inaccuracies.

elektrinis, 2023-03-31:
Well it not only does not respond quickly, but it will not balance the phases as well. Even though the hardware is probably separate for all three channels and is capable of it. For example, if everything is off and I only have my oven running at 2kW on one phase, inverter will feed 2kW back on all phases distributed equally. So my meter will see 1.33kW draw on one phase and 0.66kW feed-in on two other phases. Overall power will be zero.
And then my electricity company takes the feedin from two phases, cuts some "tax" from it and sell it back on the remaining phase.
This illustrates how they treat us:
[image]
Here we can see almost perfectly symmetrical graphs of "taken from grid" vs. "feed in to grid". While actually overall power was kept at zero.

My inverter/house is already wired in a way that everything is connected to Backup port of the inverter, except for EV chargers. The beauty of this inverter, compared to others, is that I offers full 10kW Backup, which is plenty of power for basic home needs, even for heating.
I would like to automate the switching to occur roughly 2-3 times per day (when EVs are charging, when I have too much surplus, etc). I understand mechanical contactors have limited life, especially when switching under high load. In this case however I can make sure the power is close to zero before switching.

The schematic of inverter's internals is bit lacking:
[image]
They say there is a bypass switch between Grid and Backup, which is fine. But there should also be a switch between Grid and actual inverter to isolate Grid when running in Backup mode (for anti-islanding protection and soft frequency synchronization when grid comes back on).

elektrinis, 2023-03-31:
Based on information online:

> Microgrids are self-contained systems (i.e. islanded from the main power grid) while VPPs are a combination of resources dependent on grid infrastructure. When the grid is down, VPPs can't deliver power to consumers. Microgrids functionally require some capacity for local storage such as battery systems.

I wonder what exactly MicroGrid mode does.

Louis712, 2023-04-01:
I would say that MicroGrid does exactly what it says. It's a small grid. You're independent from the outside. The generation, storage, distribution and consumption all take place in that small grid. For example, if you have a house in a rural area with no electricity, you would use it.

elektrinis, 2023-04-02:
I have switched it to MicroGrid mode to test. Can't see any difference. While I don't know if it has physically disconnected from grid, it is still reporting some power flowing.
[image]

I was able to see much higher Grid power being reported when which on the microwave.
So I don't see any difference from self-consumption mode.

Any other ideas?

mkaiser, 2023-04-09:
I did not fully understand this, yet and have several general questions: 

Will the relay between grid and inverter switch off, if you go to microgrid / some backup mode? 

Lets assume it switches off (you could also simulate this by manually turn off the circuit breaker or RCD to the inverter):

After some time (>1 hour) your local micro grid and the "cities" power grid won't be phase-aligned any more due to minor timing issues.

How could you safely switch back to the city-power grid? Are there some synchronisation mechanisms? I have not found this in sungrows documentation

mkaiser, 2023-04-09:
btw. how did you realise the surplus loading? So charging your cars, when with spare power from PV without draining the batteries?

I thought about going to EVCC, but I am not very convinced, that this is a good software for this, as you could also do it all manually in HA

closed 2025-01-21 by github-actions[bot]