mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #86: Dropdown selection is not updated

elektrinis, 2023-03-31:
This is probably issue with home assistant itself, as I am getting this with other integrations as well.
When I select EMS Mode to, for example, Forced mode, it selects fine. However, if the mode is changed from somewhere else, like a script, the dropdown does not update and either stays at old value, or becomes blank.  For example now I see this:

[image] [image]

dl1ekm, 2023-04-01:
Hi, I extended this great work with some automations (triggered by sensor change) to update the UI in HA if I change something in iSolar.

mkaiser, 2023-04-01:
Had a quick peek at your work - awesome?

Will do some general documentation first and splitting up sensors and automations. 

Would love to merge your ideas, afterwards:)

mkaiser, 2023-05-29:
[elektrinis](https://github.com/elektrinis), I made some more automations to update the sliders. Could you test it? My local isolarcloud app is so buggy and laggy, I did not wan't to test any more ;)

Just did a quick test using the HA service for modbus in the devtools....

dl1ekm, 2023-05-31:
Hi, I just tested the new sensor template - fine - no breaking changes as seen very often on others (just crashed my Weishaupt influx data collection after update.... :-)). 
Also the new default dashboard is fine - BUT: I now have 2 error messages with entities "Not available" in the dashboard:
- sensor.sg_battery_level_nom
- sensor.sg_battery_charge_nom
-

mkaiser, 2023-06-07:
Damn... forgot this issue before merging the testing branch into main....

> Also the new default dashboard is fine - BUT: I now have 2 error messages with entities "Not available" in the dashboard:
> 
> sensor.sg_battery_level_nom
> sensor.sg_battery_charge_nom

can't reproduce this. Is this still an issue? Can you give me more information?

the battery * nom sensors are based on following sensors. Are these valid in your installation?
sensor.battery_level
sensor.min_soc
sensor.max_soc

BenniG82, 2023-06-28:
@mkaiser @dl1ekm 
I just installed the latest yaml today and had the same issues with sensor.sg_battery_level_nom and sensor.sg_battery_charge_nom.
In my case, the entities were available in home assistant, but with different names.
*sensor.sg_battery_level_nom* was *sensor.battery_level_nominal*
and
*sensor.sg_battery_charge_nom* was *sensor.battery_charge_nominal*

The warning was gone after changing the entity_id via the UI (Settings -> Devices&Services -> Entities -> Select Entity -> Click on the cog -> Edit "Entity Id") 
[image]

There are some threads in the homeassistant community where people have the same issue. There seems to be no way to directly set the entity_id of a sensor. It is derived from the _name_ of the entity.
In my opinion this is a weak concept, because the entity id is used in many places (e.g. in dashboards or automations).

But I found the following workaround

modbus_sungrow.yaml
```
modbus:
  sensors:
      - name: "sg_battery_charge_nom"
        unique_id: sg_battery_charge_nom
        unit_of_measurement: kWh
        device_class: energy
```

modbus_sungrow_customize.yaml
```
homeassistant:
  customize:
    sensor.sg_battery_charge_nom:
      friendly_name: "Battery charge (nominal)"
```

Should also work if everything is defined one file

mkaiser, 2023-07-07:
hey,

this should be fixed now. The bad thing with the unique_id changes I made, that I cannot undo that for all users. 
So you need to do some manual stuff. Described it at the top of the repositories readme.

Dropdown selection update should work now, so I will close this thread ;)

closed 2023-07-07 by mkaiser