mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #88: entity not available: sensor.sg_battery_charge_nom and sg_battery_level_nom

dl1ekm, 2023-04-01:
In the new "default" dashboard these fields are shown as errors in the first page.

From my view the problem is that these are relicts from earlier versions. The correct entity (sensor.battery_charge_nominal) are already in the dashboard.

mkaiser, 2023-04-12:
sorry, I missed this issue. Will take care of it, soon! :)

Thank you for reporting!

mkaiser, 2023-07-07:
Finally i had the time to fix this: 

I suppose many users (since end of march) have this issue now. :/

Folloe EXATLY these steps to fix this (takes < 5min)

0) Update to current versions (modbus_sungrow.yaml) and dashboard(s) from the git

1) modbus_sungrow.yaml:
Comment the two sensors "Battery level (nominal)" and "Battery charge (nominal)"

2) check config & restart HA

3) GoTo "Developers --> States" and type "nominal" into the field "entity filters" (screenshot)
[image]

4) In my case, I had 4 entities matching "nominal". We want to (termporarily delete all of them to add them again with the correct unique_id.  Repeat for each one (Screenshot):
-- (1) Click on the information logo 
-- (2) Click on the gears (not shown)
-- (3) Delete and confirm
[image]

Now all sensor history connecting unique_id and sensor name are gone 

5) Revert commenting the two sensors in yaml 

6) check config & restart HA

mkaiser, 2023-07-07:
@All: 

Can you give me some feedback, if this fix is working and if the instructions are "human understandable"? ;)

mastameista, 2023-07-10:
@mkaiser - thx a lot for this fix but also for this really great integration. I am a proud owner of a solar plant since one week. :)

### Feedback

**Instructions**: yes, they are easily understandable

**Results in my case**

Besides some Powerwalker sensors containing "nominal" in their name I only had two of your sensors listed in DevTools:

[image]

I deleted those and after restarting HA, "Battery charge (nominal)" is "unavailable"
[image]

I've checked in 
`/config/.storage/core.entity_registry`

before and after the update -> no change

      {
        "aliases": [],
        "area_id": null,
        "capabilities": {
          "state_class": "measurement"
        },
        "config_entry_id": null,
        "device_class": null,
        "device_id": null,
        "disabled_by": null,
        "entity_category": null,
        "entity_id": "sensor.battery_level_nominal",
        "hidden_by": null,
        "icon": null,
        "id": "14ded1e923e219d3b814699e87b1e596",
        "has_entity_name": false,
        "name": null,
        "options": {
          "conversation": {
            "should_expose": false
          }
        },
        "original_device_class": "battery",
        "original_icon": null,
        "original_name": "Battery level (nominal)",
        "platform": "template",
        "supported_features": 0,
        "translation_key": null,
        "unique_id": "sg_battery_level_nom",
        "unit_of_measurement": "%"
      },
      {
        "aliases": [],
        "area_id": null,
        "capabilities": null,
        "config_entry_id": null,
        "device_class": null,
        "device_id": null,
        "disabled_by": null,
        "entity_category": null,
        "entity_id": "sensor.battery_charge_nominal",
        "hidden_by": null,
        "icon": null,
        "id": "3e5f3410186001d4f3cbdbf402b939d7",
        "has_entity_name": false,
        "name": null,
        "options": {
          "conversation": {
            "should_expose": false
          }
        },
        "original_device_class": "energy",
        "original_icon": null,
        "original_name": "Battery charge (nominal)",
        "platform": "template",
        "supported_features": 0,
        "translation_key": null,
        "unique_id": "sg_battery_charge_nom",
        "unit_of_measurement": "kWh"
      }

-> I've still an entity called "**sensor.battery_charge_nominal**" in the registry. In your .yaml file such an entity doesn't exist anymore.

**Any idea how to fix this?**

FYI: I prepared this integration in May and updated it last week on Monday (copy/paste modbus_sungrow.yaml and dashboard.yaml).

mkaiser, 2023-07-10:
hi, thanks for the feedback! :)

> FYI: I prepared this integration in May and updated it last week on Monday (copy/paste modbus_sungrow.yaml and dashboard.yaml)

last Friday I made the fixes, so after you upgraded: Can you upgrade to the newest git-revisions?

> I've still an entity called "sensor.battery_charge_nominal" in the registry. In your .yaml file such an entity doesn't exist anymore.

This seems to be correct: 
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/c163aa638d2c65565d4a199ce19ba563139377be/modbus_sungrow.yaml#L2188C1-L2188C1

>       - name: "Battery charge (nominal)"
>         unique_id: sg_battery_charge_nom

HA will use "battery_charge_nominal" as internal name

mkaiser, 2023-07-10:
> PS: I have a Sungrow SBR128 battery and sensor.battery_capacity always shows 0. But this is another topic and I don't want to hijack this thread ;)

I am allowed to hijack ;)
Your  installation is quiet new, so I assume that these values have not been set, yet. Try setting the battery capacity via isolarcloud (maybe app). AFAIK the is no "Auto negotiation" between battery and inverter regarding the capacity

mastameista, 2023-07-10:
Great :)

As my Solarteur is new in Sungrow business, I've setup the cloud initially. Under "Plant Configuration > Plan" the battery capacity is automatically read. 

[image]

Clicking on "Settings", no changes are possible:
[image]

But in "Advanced Settings - Hybrid Inverter - Advanced Settings - Battery Parameters" the Parameter "Battery Capacity(kWh) is 0. 

[image]

Due to my little knowhow I didn't change this parameter yet. Maybe it would fix this issue. 
And because I will be away from home for the next 2 months.

Boklov, 2023-07-17:
Hello Very good stuff. I am using this a lot. But i wounder if it is also possible to add the Feed-in limatiation trough modbus and Homeassistant

In solarcloud you can do it lie this
Advansed Otions =>  Powercontrol =>12 Feed in limitation (Enable/closed) 
=> if enabled 
12.1 Feed-in limation value in kw (0.0 - to max) 
12.2 Feed-in limitation ratio (0-100%)

This can bee usefull when prises fore the exported energy are below 0 and you dont what to export anything.

mkaiser, 2023-07-18:
already threre :)

see "export power limit"

[image]

Boklov, 2023-07-19:
> <mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/88/1641029210@
> github.com>

Hello

Ok thanks. I totally missed that option.

Thanks fore a very good work.

closed 2023-07-19 by mkaiser

btota, 2023-07-22:
@mkaiser input_select.set_sg_start_stop_mode in modbus_sungrow.yaml doesn´t exist anymore right?

btota, 2023-07-22:
> Great :)
>
> As my Solarteur is new in Sungrow business, I've setup the cloud initially. Under "Plant Configuration > Plan" the battery capacity is automatically read.
>
> [image]
>
> Clicking on "Settings", [...]

@mastameista I can affirm your suggestion. It works fine now, battery capacity is shown up the right way - **important**: you need to login in as fitter in isolarcloud.eu to get to advanced settings as @mastameista showed in his thread.

mkaiser, 2023-07-22:
> @mkaiser input_select.set_sg_start_stop_mode in modbus_sungrow.yaml doesn´t exist anymore right?

renamed for more clarity to:

> input_select:
>   set_sg_inverter_run_mode:
>     name: Inverter mode
>     options:
>       - "Enabled"
>       - "Shutdown"

btota, 2023-07-22:
then you need to update your dashboard.yaml I think :).

mkaiser, 2023-07-22:
ooops, forgot to change that - thank you :)

mheloy, 2023-07-27:
> Great :)
>
> As my Solarteur is new in Sungrow business, I've setup the cloud initially. Under "Plant Configuration > Plan" the battery capacity is automatically read.
>
> [image]
>
> Clicking on "Settings", [...]

How did you get this Battery Parameters from ISolar Cloud.. I don't have this option in mine but I have battery SBR096

btota, 2023-07-27:
As I said: "important: you need to login in as fitter in isolarcloud.eu to get to advanced settings"

from germany? then use this guidethread, howto describes how to get an fitteraccount (in case you do not have one)   -->
[https://www.photovoltaikforum.com/thread/158136-sungrow-sammelthread-produktmanagement/?postID=2495656#post2495656](url)

dvogt23, 2023-08-22:
works great, thanks for update 💯  @mkaiser

mastameista, 2023-09-13:
> @mastameista I can affirm your suggestion. It works fine now, battery capacity is shown up the right way - **important**: you need to login in as **_fitter_** in isolarcloud.eu to get to advanced [...]

@btota thanks for your verification. Will also add this parameter :)

leo4417, 2023-09-25:
Unfortunately, nothing changes for me even with the current yaml and multiple procedures as in the instructions - still "unavailable". What else can I do?

mkaiser, 2023-09-27:
> Unfortunately, nothing changes for me even with the current yaml and multiple procedures as in the instructions - still "unavailable". What else can I do?

Please open a new issue (this is already closed) and show us, what entities are shown in your installation, when you go to "Developer Tools" and filter entities for "battery_charge"

[image]

commit c46fadee (2023-07-07): Fix sensor.sg_battery_charge_nom and sg_battery_level_nom

fixes #88, reported by dl1ekm

removed references to sensors with prefix sg_ (these were relicts from development-testing)

sg_battery_level_nom  --> battery_level_nominal
sg_battery_charge_nom --> battery_charge_nominal