mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #90: Modbus connection to SG8.0RT doesn't work

dukicn, 2023-04-03:
I configured this integration with my home assistant but it doesn't work. My inverter is SG8.0RT and is connected over WiFi to my home network (currently I can't connect the device to ethernet, and I am not sure the inverter has an internal ethernet port). Device is accessible on my LAN and I can telnet to port 502 which indicates modbus service is alive. However, when I integrate it with Home Assistant, I get the following errors:

2023-04-03 12:04:33.482 ERROR (SyncWorker_1) [pymodbus.logging] General exception: index out of range
2023-04-03 12:04:33.511 ERROR (SyncWorker_1) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
2023-04-03 12:04:33.530 ERROR (SyncWorker_1) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Modbus Error: [Input/Output] Unable to decode request

Any idea what could be causing this?

Thanks

Louis712, 2023-04-03:
The WiFi module doesn't give out all registers. You would have to connect it via Ethernet.

mkaiser, 2023-04-03:
I updated the Documentation yesterday, see the drawing on the first page of the README

[image]

I am afraid, you cannot access the modbus registers at the moment. Only via LAN cable. Maybe sungrow will fix the modbus implementation on the WiNet-S stick in the future. 

Will close this issue, sorry, that we cannot help :/

closed 2023-04-03 by mkaiser

dukicn, 2023-04-03:
@mkaiser, yes, all clear now :-)

Thank you.

dylan09, 2023-04-03:
> ah, thanks for the quick feedback. I hope, I removed the ambiguity now, did I?
> 
> [image]

Great work. Think this will clarify things for new users.

ThomDietrich, 2023-05-08:
Hey friends,
for quick reinsurance :) @dukicn you own a non-hybrid SG model and you can confirm that it can be fully read by this integration? Does the integration offer all datapoints you wished for? 

https://ger.sungrowpower.com/upload/file/20220609/DE%20DS%20SG5.0_6.0_7.0_8.0_10_12RT%20Datenblatt.pdf

I intend to buy an inverter and excellent Home Assistant compatibility is one of the important decision criteria. Many thanks!

dukicn, 2023-05-08:
@ThomDietrich, per statement above, this inverter doesn't work with this integration unless you use an internal ethernet port. It doesn't work with WiFi or ethernet on WiNet-S device. Unfortunately, I can't connect internal port, but I can confirm that it doesn't work with wifi. As expected. However, there are alternatives when using WiNet-S module by using SungrowModbusWebClient. I played with several tools and settled at the end on https://sungather.net/. This works perfectly for now. With one note, I still don't have energy meter installed but should get it in the next couple of weeks. All the meters that are currently present on inverter are read by Sungather integration.

ThomDietrich, 2023-05-08:
I see! May I ask why you can't access the internal LAN? Are you not able to reach with a physical cable or is it technically challenging to access this "internal LAN port"?

dukicn, 2023-05-08:
@ThomDietrich, I didn't even try as I don't have physical cable anywhere close to my inverter.

ThomDietrich, 2023-05-08:
@mkaiser hope it's not a silly question but could you clarify where the "internal LAN port" can be found? Beware that I do not own the device yet and can't find any reference to a LAN port. Maybe something worth clarifying in text or photo :) Thanks!

lassej01, 2023-05-08:
When i look at the pictures from the manual for an SGx.0 Unit, i cannot see an MBUS-Lan port in the same way i can see it for an SHx.0 device. Seems to be a port where you can put in Mbus communication cables (a pair) but not a real RJ45 port.

ThomDietrich, 2023-05-09:
Manual: https://uk.sungrowpower.com/upload/file/20230324/EN_UM_SG3.0_4.0_5.0_6.0_7.0_8.0_10_12_15_17_20RT_User_Manual.pdf

According to the manual, The SG series features the WiNet-S connector and a two-wire RS485 connector on COM2.

[image]

@mkaiser sounds like I need a Modbus to TCP converter for the RS485 bus, or a WiNet-S Dongle, however I understand that your integration isn't compatible with it? I am confused whether or not to buy this inverter model :-/

umartiensen-dafo, 2023-05-09:
Yes. Seems the SG have only the WiNet port available. Maybe you could try ModBus communication via WiNet for your SG inverter. Maybe not all register will be available.

mkaiser, 2023-05-11:
> @mkaiser hope it's not a silly question but could you clarify where the "internal LAN port" can be found? Beware that I do not own the device yet and can't find any reference to a LAN port in the manual. Maybe something worth clarifying in text or photo :) Thanks!

Have a look here 
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/doc/help.md

May I ask, why you intend to buy a non hybrid inverter? Batteries are getting quiet cheep at the moment :)

btw. I am following your HA-community thread on EVCC integration. But I never had the time (and still no electic vehicle, but a Wallbox ;) ) to make the connection between the Sungrow inverter from HA to EVCC :) 

But it is on the long long Todo List :)

ThomDietrich, 2023-05-12:
Thanks, didn't see that page before. Okay the image confirms what the others said. The SG series doesn't offer a LAN port. So I have to either try the WiNet port (which according to what is known doesn't provide all sensors) or I need to tap into the Modbus bus on COM2 (for which I would need to buy a RS485 to Ethernet adapter and success is still not guaranteed).

Any other thoughts? I would certainly do my experiments but as I am yet to select and buy an inverter model, I might shy away from the Sungrow/SG...

> May I ask, why you intend to buy a non hybrid inverter?

Good question. Modularity and my specific situation. I intend to add more solar panel arrays to the property and eventually I will end up with multiple AC-coupled inverters. Because of that I am looking to invest into a centralized BMS like offered by Victron (still tbd). I can let you know whether that plan played out in a few years :D

> I am following your HA-community thread on EVCC integration

Nice :) We need a EVCC integration for HA. My provided yaml sensors are working just fine, but an integration would make everything a lot easier. Regarding the setup: You just need to install EVCC as a HA OS addon, then configure your inverter like this: https://docs.evcc.io/docs/devices/meters/#sungrow

dukicn, 2023-05-12:
@ThomDietrich, even though Modbus TCP doesn't work with WiNet-S, SungrowModbusWebClient does. It's not ideal as you will require a middle man between inverter and Home Assistant, but so far this worked nicely for me. To my knowledge, Sungrow doesn't support attaching 3rd party products to RS485 port and I would stay away from that.

ThomDietrich, 2023-05-12:
SungrowModbusWebClient?

I was hoping to avoid buying the WiNet adaptor. With it the price of the inverter quickly reaches other products which are good alternatives.

> Sungrow doesn't support attaching 3rd party products to RS485 port and I would stay away from that.

It's a standard two wire Modbus connection. Of course they do not "support" that but let's see how they are going to stop me

mkaiser, 2023-05-12:
> Thanks, didn't see that page before. Okay the image confirms what the others said. The SG series doesn't offer a LAN port. So I have to either try the WiNet port (which according to what is known [...]

I believe (not sure!) that the COM Terminals pin for modbus are only for interconnecting the power meter via RS485. So sniffing around won't get you very far. 

There are additional ports for CAN bus attached batteries (My Pylontech battery is attached by this). If you sniff this bus you could only see some battery related status information. 

I guess changing the Inverters internal settings (EMS mode etc.) is just possible by TCP Modbus or the Web-based WiNeT stuff

> Good question. Modularity and my specific situation. I intend to add more solar panel arrays to the property and eventually I will end up with multiple AC-coupled inverters. Because of that I am [...]

I was (and still am) skeptical, when it comes to some vendor-specific BMS solutions. There is no guaranteed interoperability with other vendors. Real /practical standards are still missing in this field. 

For me, everything will come down to using a Software-BMS like EVCC (there are others like the EMHASS Add-on), where you have to manually interconnect the components. This will most likely work very good, when sensors are collected in HA and then passed by MQTT (like your HA <> EVCC)

> Nice :) We need a EVCC integration for HA. My provided yaml sensors are working just fine, but an integration would make everything a lot easier. Regarding the setup: You just need to install EVCC as a HA OS addon, then configure your inverter like this: https://docs.evcc.io/docs/devices/meters/#sungrow - but yeah I get you. Not on top of the todo list :)

Won't work that good, because you will have two TCP connections to the Modbus TCP port in parallel. They will most probably interfere. I strive to the architecture: Inverter <--Modbus--> HA <-- MQTT -- > EVCC 

After 1.5 years of experience with the SGH10.RT I can say that the hardware is pretty good, but the software (still!) sucks. But with this modbus stuff, I don't have to care about the Sungrow software any more :)
At the time of buying the performance / cost ration was better than the competitors ratios (Victron, SMA, ...) :)

ThomDietrich, 2023-05-23:
Hey guys,
thanks for the great discussion! This was very helpful.

I just wanted to quickly come back and let you know that because of various reasons, I decided against the Sungrow inverter.

Thanks again. Cheers!