mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #94: LAN connection with Winet-S

scraja69, 2023-04-12:
I got Modbus LAN connection details from Sungrow. I was able to get Modbus data from following instructions given below from my SG15RT inverter. I used LAN port in Winet-S.

[The Procedure for Setting up Winet .
[TD_202102_iSolarCloud_WiNet Dongle Quick Guide_V1.0.pdf](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/files/11209230/TD_202102_iSolarCloud_WiNet.Dongle.Quick.Guide_V1.0.pdf)
Modbus Connection.pdf](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/files/11209216/The.Procedure.for.Setting.up.Winet.Modbus.Connection.pdf)
[image]

lassej01, 2023-04-12:
Hi!. good for you. Still there are many users here who have had problems with some or all values from Modbus using the method with connection to modbus through the dongle.

mkaiser, 2023-04-12:
Good to know. Are all sensors read and working well? 

Most issues were battery-related (not applicable in your case) and some sensors like grid frequency (off by factor 10)

Can you tell us the installed firmware version?

I don't want to believe yet, that sungrow is finally fixing the winet stuff ;)

scraja69, 2023-04-12:
> <mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/94/1505126877@
> github.com>

Hi,
Thank you for your great project. I have not tested fully. I will test and
let you know. Here is the log from ModbusTCP2MQTT

Regards

Seethaka

pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output]
Unable to decode request
Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/pymodbus/transaction.py",
line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py",
line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py",
line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output]
Unable to decode request
2023-04-12 22:26:51 WARNING  Modbus connection failed
Modbus connection failed
2023-04-12 22:26:51 DEBUG    Modbus Error: [Input/Output] Unable to
decode request
2023-04-12 22:26:51 DEBUG    Scraping: hold, 4999:10
2023-04-12 22:26:51 DEBUG    load_registers: hold, 4999:10
2023-04-12 22:26:51 DEBUG    Connection to Modbus server established.
Socket ('172.30.33.19', 48687)
2023-04-12 22:26:51 DEBUG    Current transaction state - TRANSACTION_COMPLETE
2023-04-12 22:26:51 DEBUG    Running transaction 51
2023-04-12 22:26:51 DEBUG    Clearing current Frame : - 0x0 0x32 0x0
0x0 0x0 0x2 0x1 0x84 0x2
2023-04-12 22:26:51 DEBUG    SEND: 0x0 0x33 0x0 0x0 0x0 0x6 0x1 0x3
0x13 0x87 0x0 0xa
2023-04-12 22:26:51 DEBUG    New Transaction state 'SENDING'
2023-04-12 22:26:51 DEBUG    Changing transaction state from 'SENDING'
to 'WAITING FOR REPLY'
2023-04-12 22:26:51 DEBUG    Changing transaction state from 'WAITING
FOR REPLY' to 'PROCESSING REPLY'
2023-04-12 22:26:51 DEBUG    RECV: 0x0 0x33 0x0 0x0 0x0 0x17 0x1 0x3
0x14 0x7 0xe7 0x0 0x4 0x0 0xc 0x0 0x16 0x0 0x1a 0x0 0x29 0x0 0xcf 0x0
0xaa 0x3 0xe8 0xff 0xff
2023-04-12 22:26:51 DEBUG    Processing: 0x0 0x33 0x0 0x0 0x0 0x17 0x1
0x3 0x14 0x7 0xe7 0x0 0x4 0x0 0xc 0x0 0x16 0x0 0x1a 0x0 0x29 0x0 0xcf
0x0 0xaa 0x3 0xe8 0xff 0xff
2023-04-12 22:26:51 DEBUG    Factory Response[ReadHoldingRegistersResponse: 3]
2023-04-12 22:26:51 DEBUG    Adding transaction 51
2023-04-12 22:26:51 DEBUG    Getting transaction 51
2023-04-12 22:26:51 DEBUG    Changing transaction state from
'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2023-04-12 22:26:51 DEBUG    Scraping: hold, 5009:10
2023-04-12 22:26:51 DEBUG    load_registers: hold, 5009:10
2023-04-12 22:26:51 DEBUG    Current transaction state - TRANSACTION_COMPLETE
2023-04-12 22:26:51 DEBUG    Running transaction 52
2023-04-12 22:26:51 DEBUG    SEND: 0x0 0x34 0x0 0x0 0x0 0x6 0x1 0x3
0x13 0x91 0x0 0xa
2023-04-12 22:26:51 DEBUG    New Transaction state 'SENDING'
2023-04-12 22:26:51 DEBUG    Changing transaction state from 'SENDING'
to 'WAITING FOR REPLY'
2023-04-12 22:26:51 DEBUG    Changing transaction state from 'WAITING
FOR REPLY' to 'PROCESSING REPLY'
2023-04-12 22:26:51 DEBUG    RECV: 0x0 0x34 0x0 0x0 0x0 0x17 0x1 0x3
0x14 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-04-12 22:26:51 DEBUG    Processing: 0x0 0x34 0x0 0x0 0x0 0x17 0x1
0x3 0x14 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
[... 115 lines cut ...]
0.0, "string_10_current": 0.0, "string_11_current": 0.0,
"string_12_current": 0.0, "string_13_current": 0.0,
"string_14_current": 0.0, "string_15_current": 0.0,
"string_16_current": 0.0, "string_17_current": 0.0,
"string_18_current": 0.0, "string_19_current": 0.0,
"string_20_current": 0.0, "string_21_current": 0.0,
"string_22_current": 0.0, "string_23_current": 0.0,
"string_24_current": 0.0, "start_stop": "Start",
"power_limitation_switch": "Enable", "power_limitation_setting":
100.0, "export_power_limitation": 0, "export_power_limitation_value":
0, "current_transformer_output_current": 0,
"current_transformer_range": 0, "current_transformer": 0,
"export_power_limitation_percentage": 0.0, "installed_pv_power": 0.0,
"power_factor_setting": 0.0, "night_svg_switch": "Disable",
"reactive_power_adjustment_mode": "Enable Q(U)",
"reactive_power_percentage_setting": 0.0,
"power_limitation_adjustment": 15.0, "reactive_power_adjustment": 0.0,
"pid_recovery": "Disable", "anti_pid": 0, "fullday_pid_suppression":
0, "export_to_grid": 0, "import_from_grid": 0, "timestamp": "2023-4-12
22:26:41"}
2023-04-12 22:26:52 DEBUG    MQTT: Message 24 Published
2023-04-12 22:26:52 INFO     MQTT: Published
2023-04-12 22:26:52 INFO     Closing Session: ModbusTcpClient(192.168.1.145:502)
2023-04-12 22:26:52 DEBUG    Processing Time: 0.74 secs
2023-04-12 22:26:52 INFO     Next scrape in 89 secs

mkaiser, 2023-04-12:
I am wondering why there is some MQTT involved as this integration has nothing do with MQTT *confused*

scraja69, 2023-04-13:
Hi,
I have provided you the log of addon- ModbusTCP2MQTT by https://github.com/MatterVN/ModbusTCP2MQTT to show modbus registers  received. 
I have given below screenshots of your dashboard. Strangely Some of registers are not visible  from your dash board .Ex  Inverter phase current A, Inverter phase current B, Inverter phase current C but they are visible in ModbusTCP2MQTT ?
[screenshots.docx](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/files/11217839/screenshots.docx)
[Modbustcp2mqtt.docx](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/files/11217840/Modbustcp2mqtt.docx)

dylan09, 2023-04-13:
Please change to the internal LAN Port. And check if all registers have valid values.
It seems that the WiNet S doesn`t provide all modbus registers. Even in the latest firmware version. 
There are some posts and information in photovoltaikforum.de

KLucky13, 2023-04-18:
I'm also struggling to find out the best way forward at this point. I also have 2 inverters using the Winet-S dongle. I'm currently using this integration: https://github.com/MickMake/GoSungrow

But all data has a 5 minute update interval which makes it useless for most of the automations I had in mind..

tjhowse, 2023-06-29:
I am running a Sungrow SH5k-20 and I had issues with the modbus connection dropping regularly until I added a firewall rule in my router that blocks internet access to the inverter. This *improved* the connection, but it still drops occasionally. I suspect the internal modbusRTU to modbusTCP interface inside the inverter/winet dongle can only gracefully handle a single concurrent TCP connection. If Sungrow's cloud stuff is talking to the inverter this might cause issues.

closed 2023-09-11 by mkaiser