mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #98: Total DC Power wrong

svenschedlbauer, 2023-05-05:
Hello,
There seems to be something wrong with my Total DC Power. Does anybody know what could be wrong and how can I solve it?
Normally it should be now "0".
[image]
Thanks,
Sven

svenschedlbauer, 2023-05-05:
Yes, of course. See following Screenshot.
[image]

mkaiser, 2023-05-05:
If all other sensor values are in reasonable ranges, I suppose this is a sungrow firmware bug. 

Did you already restart the inverter?

svenschedlbauer, 2023-05-05:
The other values seems to be ok. I have switched the Sungrow 6 now off and after 20 seconds on. But the value is still at 158.728.192 W +/-.

mkaiser, 2023-05-06:
How is the value now, when the sun is shining?

svenschedlbauer, 2023-05-07:
It is still the same... :-(
[image]

Louis712, 2023-05-07:
Something like that also happened to me, just not with Total DC Power, but with the yearly pv generation. I checked it with qmodmaster and it isn't a home assistant bug. Sungrow is known for their pretty bad implementation of MODBUS.

svenschedlbauer, 2023-05-07:
But how have you solved the problem? With which sensors I can make the helper to get the right value?

Louis712, 2023-05-07:
I unfortunately wasn't able to solve it. I'm probably just going to live with it. In your case, it's pretty bad though. As a quick workaround, you could look at how much it jumped and then create a helper, that subtracts that value from the entity. It seems to have spiked a bit, so you are getting the right data, just with an offset.

svenschedlbauer, 2023-05-07:
Problem is, that iDC Power shows also different very high values in the night, when there is no sun. :-(

Louis712, 2023-05-07:
Then you can probably just remove the value shown at night.

svenschedlbauer, 2023-05-07:
Yes, but unfortunately this value also differs at night.

Louis712, 2023-05-07:
Hmmm... That's pretty bad. Maybe contact sungrow. Perhaps they have a solution or will forward this problem to the software developers.

elektrinis, 2023-05-07:
Let us know if you manage to get a response, and from who.
They have started to completely ignore me, even after having potential hardware problem with battery.

svenschedlbauer, 2023-05-07:
I will inform you, when I "hopefully" get an Feedback from them.

svenschedlbauer, 2023-05-07:
Is following correct?

[image]

Louis712, 2023-05-07:
That's correct.

dylan09, 2023-05-07:
Which port did you use to connect to your inverter? WiNet S or direct LAN port?

svenschedlbauer, 2023-05-07:
Direct LAN Port

svenschedlbauer, 2023-05-07:
Sorry Guys, It seems that I found now the failure in my system. I have used the wrong IP address. I have changed it and now the value looks correct. I still don't from where the other address is. May from the WiFi dongle of the Sungrow.
But thank you all for your try to help.

closed 2023-05-07 by svenschedlbauer