mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #99: Fully local Battery charge state prediction

HAuser1234, 2023-05-07:
Hello,
first this isn‘t a issue. I made a github repository for a homeassistant template code that tries to predict the state oft the battery during das and night. sinne I use this Sungrow-SHx-Inverter integration, this code should work plug and play.

# Features
all features are tested and only approximate real circumstances!

- fully offline!
- predict battery state at evening after charging
- predict battery state at morning after decharging
- predict battery charging time
- predict minimum charge needed to get over night
- predict time left of solar
- predict start of charging
- example lovelace card (mushroom required)

If you are interested have a look here: https://github.com/HAuser1234/Homeassistant-solar-forecast-charge-prediction
I am glad for contributions of improvements :)

I hope I am at the right place here!

mkaiser, 2023-05-11:
hi, 

great ideas ;) Thank you for the hints

Will check it in a couple of weeks, I am very swamped this may with work ...

mkaiser, 2023-05-12:
Took me a while to get the dashboard configured with your config. The mushroom hacs stuff with some spacing drove me nuts..... Sharing it here :)

Will evaluate later, have to work now, again ;)

[lovelace_raw_battery_prediction_mushroom_hacs_config.yaml.txt](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/files/11464458/lovelace_raw_battery_prediction_mushroom_hacs_config.yaml.txt)

HAuser1234, 2023-05-12:
Oh great. I will probably add this to the repo soon. Or do you want to PR this?
I am looking forward to your experience with it!
Maybe you find something i missed or find a way to add more functionality :)

closed 2025-01-12 by github-actions[bot]