mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #110: Battery Charging / Discharging

Lempomat, 2023-06-02:
Hi,
first of all, thanks a lot for your modbus script. I have a problem with the "Battery charging Power" and "Battery discharging Power". But there are only 0KW... So the "Battery Row" value works.

[image]
[image]

Have you got an idea?

Lempomat, 2023-06-02:
Hi,
so i checked the sensors and I changed this:

- name: Battery charging
        unique_id: sg_battery_charging
        availability: >-
          {{states('sensor.battery_power_raw')|is_number and 
            states('sensor.total_dc_power')|is_number and 
            states('sensor.load_power')|is_number 
          }}
        state: >-
               {% if ( ( states('sensor.total_dc_power')|int > states('sensor.load_power')|int ) ) and states('sensor.battery_power_raw')|int > 0 %}
                {# more power generated than consumed. assuming battery charging #}
                On
               {% else %} 
                Off
               {% endif %}

      - name: Battery discharging
        unique_id: sg_battery_discharging
        availability: >-
          {{states('sensor.battery_power_raw')|is_number and 
            states('sensor.total_dc_power')|is_number and 
            states('sensor.load_power')|is_number 
          }}
        state: >-
            {% if ( ( states('sensor.total_dc_power')|int < states('sensor.load_power')|int ) ) and states('sensor.battery_power_raw')|int > 0 %}
                {# more power consumed than generated and some battery power -->  assuming battery discharging #}
              On
            {% else %} 
              Off
            {% endif %}

So this worked for me.

mkaiser, 2023-06-04:
Hi, 

I can't comprehend, what was your issue with the two template sensors and how your removal of the case decisions regarding the forced charge / discharge fixed the issue (without this the sensors are giving a false status in forced mode)

Can you give me more information? I guess you are using an RS inverter, that does not support the running_state, because you removed it from the template sensor.

Lempomat, 2023-06-04:
Hi,
so the two entities had no values, because some configured entities had not the values that you defined in your "if" statements. The above configuration that I changed works correctly for me. So everything is now fine 😄

closed 2023-06-04 by Lempomat

mkaiser, 2023-06-04:
But now I am curious - which sensors did not work in the first place?

Lempomat, 2023-06-04:
Hi,

"Battery charging Power" and "Battery discharging Power"

After I changed the if condition by this entities it worked.