mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #115: Input via slider "Forced Charge Power" will be resettet immediately

Matze-aaa, 2023-06-17:
I have inserted all patch-lines in the modbus_sungrow.yaml.
Now, when I set a value via slider "forced charging/discharging power", it will be resetted immediately.

mkaiser, 2023-07-21:
sorry, I completely missed this issue

is this still an issue in the current versions? Made quiet a few changes the last month...

Matze-aaa, 2023-07-23:
Hello, yes, the issue is still here... have implemented all of the changes

mkaiser, 2023-07-23:
What inverter model are you using?

Matze-aaa, 2023-07-23:
Sh10RT V11

mkaiser, 2023-09-11:
Some of the last commits should fix this issue. Can you update and double-check?

.... but maybe wait one or two weeks with upgrading. There are a lot of issues ongoing with pymodbus <> HA <> Sungrow :)

closed 2024-12-29 by github-actions[bot]