mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #117: Problems / No ModBus connection on regular LAN-Port

MIRR23, 2023-06-23:
I currently also use GoSungrow, but there are problems with 2 inverters (both SH10RT) and a long-awaited update is unfortunately waiting.
I have now found this really cool way to read the data, which has everything I need and more.
With ModBus I am not so knowledgeable, but that will be with time.

Big problem: I can not read the data reasonably, no matter what I do...
- Direct LAN port at the inverter --> ModBus connection error
- LAN port via WinNet-S --> not all data available, aborts again and again
- Wifi via WinNet-S --> this works surprisingly best, but not all data available and always aborts.

I have tried everything to use the normal LAN port:
- Reboot the inverter
- unplugged LAN cable for 10 min and for 2 hours
- Changing the ports on the switch
- Tried LAN connection with WinNet-S (as described in issue #94 ) and also WLAN connection 

Is there anything else I can do???

dylan09, 2023-06-24:
Did you use the correct IP address for the LAN port connection? LAN port has a  different address than the LAN port within Winet-S adapter.

MIRR23, 2023-06-24:
Yes I did. I actually tried all 3 IP addresses of each inverter (LAN, LAN over Winet, WLAN).

After quite many restarts I somehow managed to get Modbus data from the first inverter.
The second one still it is not working.
Could there maybe be an issue with master/slave configuration?

closed 2024-12-29 by github-actions[bot]